mavlink/MAVSDK-Python #263: battery.remaining_percentage returns negative value

ramgrandhi, 2020-10-05:
Hi there!

I'm running on SITL setup with latest releases : `PX4 (1.11.0) + mavsdk_server (0.30.1) + mavsdk-python (0.12.0)`. 
My pre-flight-checks are failing due to the negative value that is returned for `battery.remaining_percent`.

Just by running `mavsdk-python/examples/telemetry.py` you will notice the following.

```
Waiting for mavsdk_server to be ready...
Connected to mavsdk_server!
Position: [latitude_deg: 47.3977504, longitude_deg: 8.5456064, absolute_altitude_m: 488.0440368652344, relative_altitude_m: -0.08500000089406967]
**Battery: -0.009999999776482582**
```

Appreciate your attention!

cheers,
Ram

JonasVautherin, 2020-10-05:
Could it be that PX4 sends that value? Do you have a way to check that from the PX4 side? I guess we may have an error somewhere on the way (numbers get translated), but it would be nice to know what PX4 sends in that case :+1:.

julianoes, 2020-10-22:
Battery stuff has changed a bit, so yes we should check what PX4 is sending now.