mavlink/MAVSDK-Python #264: Indoor flight

hi6otb, 2020-10-08:
I want to use PX4 in indoor so I use Realsense T265 and VIO.
https://docs.px4.io/master/en/peripherals/camera_t265_vio.html

However "health.is_local_potision_ok" and "health.is_global_potision_ok" are "FALSE".

Anyone know how to use other FakeGPS method?

julianoes, 2020-10-22:
@hi6otb the important thing is that PX4 must have a position estimate. You can do that by feeding visual position estimates from VIO sensors or flow modules. However, all of that is more of a PX4 thing to figure out. Have a look here https://dev.px4.io/master/en/ros/external_position_estimation.html.

Then, once you have a position estimate you should be able to use offboard control but not mission because mission (at least for PX4) requires GPS, so a "global" position estimate. At this point you will just have to ignore some of the health data because it doesn't apply.

hi6otb, 2020-10-26:
@julianoes I confirmed PX4 stream back the received external pose as MAVLink ODOMETRY messages by using realsense T265. 
However, I can't set initial setpoint on python so I can't change mode to offboard.
I guess it needs "health.is_local_potision_ok" and "health.is_global_potision_ok" are "True".

julianoes, 2020-10-26:
> However, I can't set initial setpoint on python so I can't change mode to offboard.

Why not? What's missing?

hrnbot, 2021-01-04:
@hi6otb Why are you not able to setpoint in python.
 `await drone.offboard.set_position_ned(PositionNedYaw(0.0, 0.0, 0.0, 0.0))`

[Check this Example](https://github.com/mavlink/MAVSDK-Python/blob/main/examples/offboard_position_ned.py)

julianoes, 2021-03-26:
GPS/position data can now be provided using the [Mocap plugin](http://mavsdk-python-docs.s3-website.eu-central-1.amazonaws.com/plugins/mocap.html).

closed 2021-03-26 by julianoes