mavlink/MAVSDK-Python #266: accessing DEBUG mssg

Traveller30, 2020-10-13:
Hi 
Is  there a way to get to the "DEBUG" mavlink mssgs ? 
I mean any of the three that exist in PX4 - "DEBUG","DEBUG_VECT","NAMED_VALUE_FLOAT"
I can see the messages and data in QGC but can't  find how to get to the data by MAVSDK-python API

julianoes, 2020-10-22:
It would be nice to have an option to enable these messages similar to the C++ side.

For now, you should be able to use this: http://mavsdk-python-docs.s3-website.eu-central-1.amazonaws.com/plugins/telemetry.html#mavsdk.telemetry.StatusText

Traveller30, 2020-10-22:
thank you Julian
I am using the "StatusText" for now  ,but the mssg txt interface and the mssg's that popup in QGC driving me crazy :-(
I guess we can all  benefit from using the "debug mssg"  for monitoring and development.