mavlink/MAVSDK-Python #268: Telemetry subscriptions eating up Threads

GriffinMack, 2020-10-15:
I have a simple while loop that calls the code below (from the telemetry examples) until a change is detected. My problem is that after about 4000 runs through this loop, I get the **grpc error** below. Looking at my task manager, I can see THOUSANDS of threads and ports being opened by mavsdk_server while this loop is running. I have attached a screenshot of this as well. Is there a better way to do this that cleans up the thread perhaps? I may be missing something simple here.

I have tried all the other telemetry functions so it is not just position() that has this issue.

    async for position in pixhawkVehicle.telemetry.position():
        absolute_altitude = position.absolute_altitude_m
        relative_altitude = position.relative_altitude_m
        latitude = position.latitude_deg
        longitude = position.longitude_deg
        break

async for position in pixhawkVehicle.telemetry.position():
  File "/Users/griffinmack/.pyenv/versions/3.8.1/lib/python3.8/site-packages/mavsdk/telemetry.py", line 2883, in position
    async for response in position_stream:
  File "/Users/griffinmack/.pyenv/versions/3.8.1/lib/python3.8/site-packages/aiogrpc/utils.py", line 154, in __anext__
    return await asyncio.shield(self._next_future)
  File "/Users/griffinmack/.pyenv/versions/3.8.1/lib/python3.8/site-packages/aiogrpc/utils.py", line 145, in cb
    fut.result()
  File "/Users/griffinmack/.pyenv/versions/3.8.1/lib/python3.8/concurrent/futures/thread.py", line 57, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/Users/griffinmack/.pyenv/versions/3.8.1/lib/python3.8/site-packages/aiogrpc/utils.py", line 131, in _next
    return next(self._iterator)
  File "/Users/griffinmack/.pyenv/versions/3.8.1/lib/python3.8/site-packages/grpc/_channel.py", line 416, in __next__
    return self._next()
  File "/Users/griffinmack/.pyenv/versions/3.8.1/lib/python3.8/site-packages/grpc/_channel.py", line 689, in _next
    raise self
grpc._channel._MultiThreadedRendezvous: <_MultiThreadedRendezvous of RPC that terminated with:
        status = StatusCode.RESOURCE_EXHAUSTED
        details = ""
        debug_error_string = "{"created":"@1602747090.002901000","description":"Error received from peer ipv6:[::1]:50051","file":"src/core/lib/surface/call.cc","file_line":1062,"grpc_message":"","grpc_status":8}"

[image]

JonasVautherin, 2020-10-15:
Right, it means that there is a leak on the C++ side: when a stream client disconnects, it should release the thread. I thought I had fixed that in the past, but maybe it came back when we started auto-generating `mavsdk_server` :thinking:.

Just out of curiosity: since you `break` after one update, I am guessing that you run that `async for` in an external loop. Why do you do that? Why do you need the `break`?

GriffinMack, 2020-10-15:
@JonasVautherin I guess you are correct, I can probably remove that break in this case. I had it in because I needed to check multiple telemetry values, so I would **break** once one value came back.

GriffinMack, 2020-10-16:
@JonasVautherin I was able to rewrite this loop so it only makes one stream. However, using the telemetry generators and breaking after one request is pretty common in my code (and the mavsdk examples are filled with this as well). During a long flight period, these threads could really become a problem. 

Any short term solutions to this? Or any idea where to look to fix this on the C++ side?

JonasVautherin, 2020-10-16:
That could be [here](https://github.com/mavlink/MAVSDK/blob/develop/src/backend/src/plugins/telemetry/telemetry_service_impl.h#L1094-L1127). Every new subscription creates a new thread that calls `Subscribe*`. I don't get where the leak would come from, though, as it's supposed to stop when the client disconnects :thinking:. Happy to get insights, since you can reproduce that!

julianoes, 2021-03-26:
I'm going to close this, please re-open or comment if this is still an issue.

closed 2021-03-26 by julianoes