mavlink/MAVSDK-Python #280: Installation of MAVSDK on Windows failed

olafwessler, 2020-11-02:
Hi, 

when running
pip3 install mavsdk
 on a Windows PC, I get the below error message (shortened).

Thanks for any advise!

C:\data>pip3 install mavsdk
Collecting mavsdk
  Using cached mavsdk-0.13.4-py3-none-win_amd64.whl (2.8 MB)
Collecting grpcio>=1.11.0
  Using cached grpcio-1.33.2.tar.gz (20.9 MB)
Collecting aiogrpc>=1.5
  Using cached aiogrpc-1.8-py3-none-any.whl (10 kB)
Requirement already satisfied: protobuf>=3.13.0 in c:\users\user\appdata\local\programs\python\python39\lib\site-packages (from mavsdk) (3.13.0)
Requirement already satisfied: six>=1.5.2 in c:\users\user\appdata\local\programs\python\python39\lib\site-packages (from grpcio>=1.11.0->mavsdk) (1.15.0)
Requirement already satisfied: setuptools in c:\users\user\appdata\local\programs\python\python39\lib\site-packages (from protobuf>=3.13.0->mavsdk) (49.6.0)
Building wheels for collected packages: grpcio
  Building wheel for grpcio (setup.py) ... error
  ERROR: Command errored out with exit status 1:
   command: 'c:\users\user\appdata\local\programs\python\python39\python.exe' -u -c 'import sys, setuptools, tokenize; sys.argv[0] = 
'"'"'C:\\Users\\User\\AppData\\Local\\Temp\\pip-install-bxinc9lj\\grpcio\\setup.py'"'"'; __file__='"'"'C:\\Users\\User\\AppData\\Local\\
Temp\\pip-install-bxinc9lj\\grpcio\\setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', 
'"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d 'C:\Users\User\AppData\Local\Temp\pip-wheel-ty6jccwv'
       cwd: C:\Users\User\AppData\Local\Temp\pip-install-bxinc9lj\grpcio\
  Complete output (79 lines):
  ASM Builds for BoringSSL currently not supported on: win-amd64
  Found cython-generated files...
  running bdist_wheel
  running build
  running build_py
  running build_project_metadata
  creating python_build
  creating python_build\lib.win-amd64-3.9
  creating python_build\lib.win-amd64-3.9\grpc
  copying src\python\grpcio\grpc\_auth.py -> python_build\lib.win-amd64-3.9\grpc
 ...
 ...
  copying src\python\grpcio\grpc\_cython\_credentials\roots.pem -> python_build\lib.win-amd64-3.9\grpc\_cython\_credentials
  running build_ext
  error: [WinError 2] Das System kann die angegebene Datei nicht finden
  ----------------------------------------
  ERROR: Failed building wheel for grpcio
  Running setup.py clean for grpcio
Failed to build grpcio
Installing collected packages: grpcio, aiogrpc, mavsdk
    Running setup.py install for grpcio ... error
    ERROR: Command errored out with exit status 1:
     command: 'c:\users\user\appdata\local\programs\python\python39\python.exe' -u -c 'import sys, setuptools, tokenize; sys.argv[0] =
 '"'"'C:\\Users\\User\\AppData\\Local\\Temp\\pip-install-bxinc9lj\\grpcio\\setup.py'"'"'; __file__='"'"'C:\\Users\\User\\AppData\\Local\\
Temp\\pip-install-bxinc9lj\\grpcio\\setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', 
'"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record 'C:\Users\User\AppData\Local\Temp\pip-record-6u8jt7ja\
install-record.txt' --single-version-externally-managed --compile --install-headers 'c:\users\user\appdata\local\programs\python\python39\
Include\grpcio'
         cwd: C:\Users\User\AppData\Local\Temp\pip-install-bxinc9lj\grpcio\
    Complete output (80 lines):
    ASM Builds for BoringSSL currently not supported on: win-amd64
    Cython-generated files are missing...
    We could not find Cython. Setup may take 10-20 minutes.
    running install
    running build
    running build_py
    running build_project_metadata
    creating python_build
    creating python_build\lib.win-amd64-3.9
    creating python_build\lib.win-amd64-3.9\grpc
    copying src\python\grpcio\grpc\_auth.py -> python_build\lib.win-amd64-3.9\grpc
...
...
    copying src\python\grpcio\grpc\_cython\_credentials\roots.pem -> python_build\lib.win-amd64-3.9\grpc\_cython\_credentials
    running build_ext
    error: [WinError 2] Das System kann die angegebene Datei nicht finden
    ----------------------------------------
ERROR: Command errored out with exit status 1: 'c:\users\user\appdata\local\programs\python\python39\python.exe' -u -c 'import sys, 
setuptools, tokenize; sys.argv[0] = '"'"'C:\\Users\\User\\AppData\\Local\\Temp\\pip-install-bxinc9lj\\grpcio\\setup.py'"'"'; 
__file__='"'"'C:\\Users\\User\\AppData\\Local\\Temp\\pip-install-bxinc9lj\\grpcio\\setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)
(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record
 'C:\Users\User\AppData\Local\Temp\pip-record-6u8jt7ja\install-record.txt' --single-version-externally-managed --compile --install-headers 
'c:\users\user\appdata\local\programs\python\python39\Include\grpcio' Check the logs for full command output.

JonasVautherin, 2020-11-02:
MAVSDK-Python depends on grpcio (`pip3 install grpcio`), and it seems like it is not distributed as a binary and fails to compile on Windows. I don't think I could solve that myself, I would advise you try to find in the grpc project why `grpcio` cannot build on Windows :confused:.

olafwessler, 2020-11-02:
Thanks for quick Response.
Downgrading python to 3.7 solved the issue

closed 2020-11-02 by JonasVautherin