mavlink/MAVSDK-Python #285: Question: Recommended way to run MAVSDK-Python on Raspberry Pi?

GriffinMack, 2020-11-08:
I am trying to run MAVSDK-python on a raspberry pi 3b+ connected to a Pixhawk 4 through a serial converter attached to one of the USB ports. 

My issue is getting mavsdk_server built and running. I first tried pip installing which prompted me to build mavsdk_server from scratch. I then attempted to follow this [guide](https://discuss.px4.io/t/talking-to-a-px4-fmu-with-a-rpi-via-serial-nousb/14119), but ran into issues trying to compile the backend. I then tried to cross compile on a linux and macOS system using dockercross, and both of these attempts failed at different spots from missing CMakeLists files.

I'm wondering which method is currently recommended for getting MAVSDK-python up and running?

julianoes, 2020-11-08:
The mavsdk pip installation should now include the mavsdk_server binary and it should run on Raspberry Pis.

Try `pip3 install --user --upgrade mavsdk` again please.

GriffinMack, 2020-11-08:
@julianoes will the mavsdk_server not launch automatically like it does on Linux/macOS on the raspberry Pi? I am still stuck waiting for the server to be ready.

JonasVautherin, 2020-11-08:
It should, unless it shows you a message saying that you should start it manually (which I believe is not the case here). Why do you say it is waiting for the server to be ready? What code do you run and what message do you see?

By installing it through `pip`, you still get `mavsdk_server` installed somewhere on your RPi, and you can run it manually. This will give you more log output (e.g. you'll see if a drone is discovered or not).

GriffinMack, 2020-11-09:
Okay, it looks like I got ahead of myself trying to build mavsdk from scratch ASSUMING pip install still didn't work for raspberry Pi. After uninstalling mavsdk and reinstalling using pip3 install --user --upgrade mavsdk the mavsdk_server launched automatically (I am using the firmware_version example). When I said the server was waiting to be ready, I had forgotten to remove the 'localhost' from the System() connection. Thanks for the help!

The only change I had to make for communication through USB to work, if anyone is interested:
await drone.connect(system_address="serial:///dev/ttyUSB0:921600")

closed 2020-11-09 by GriffinMack

faris150, 2021-11-06:
@GriffinMack How you connect the Raspberry Pi to the drone? using USB telemetry or using TX & RX pins ?

julianoes, 2021-11-08:
@GriffinMack both should work.