mavlink/MAVSDK-Python #286: Offboard Attitude Example Error: COMMAND_DENIED

canberkdurmus, 2020-11-18:
Hello,

I am trying to use the python examples from the repository. 
Installation:  "pip3 install mavsdk"
OS: MacOS Catalina 10.15.7
Test Environment: PX4-Autopilot master branch SITL Gazebo

Python Script Output:
```
$ python3 offboard_attitude.py 

Waiting for mavsdk_server to be ready...
Connected to mavsdk_server!
Waiting for drone to connect...
Drone discovered with UUID: 0
-- Arming
-- Setting initial setpoint
-- Starting offboard
Starting offboard mode failed with error code:               COMMAND_DENIED
-- Disarming
```

PX4 SITL Gazebo Output:
```
$ make px4_sitl gazebo

[0/4] Performing build step for 'sitl_gazebo'
ninja: no work to do.
[3/4] cd /Users/canberk/PX4-Autopilot/...k/PX4-Autopilot/build/px4_sitl_default
SITL ARGS
sitl_bin: /Users/canberk/PX4-Autopilot/build/px4_sitl_default/bin/px4
debugger: none
program: gazebo
model: none
world: none
src_path: /Users/canberk/PX4-Autopilot
build_path: /Users/canberk/PX4-Autopilot/build/px4_sitl_default
empty model, setting iris as default
GAZEBO_PLUGIN_PATH :/Users/canberk/PX4-Autopilot/build/px4_sitl_default/build_gazebo
GAZEBO_MODEL_PATH :/Users/canberk/PX4-Autopilot/Tools/sitl_gazebo/models
LD_LIBRARY_PATH :/Users/canberk/PX4-Autopilot/build/px4_sitl_default/build_gazebo
empty world, setting empty.world as default
SITL COMMAND: "/Users/canberk/PX4-Autopilot/build/px4_sitl_default/bin/px4" "/Users/canberk/PX4-Autopilot/build/px4_sitl_default"/etc -s etc/init.d-posix/rcS -t "/Users/canberk/PX4-Autopilot"/test_data
INFO  [px4] Creating symlink /Users/canberk/PX4-Autopilot/build/px4_sitl_default/etc -> /Users/canberk/PX4-Autopilot/build/px4_sitl_default/tmp/rootfs/etc

______  __   __    ___ 
| ___ \ \ \ / /   /   |
| |_/ /  \ V /   / /| |
|  __/   /   \  / /_| |
| |     / /^\ \ \___  |
\_|     \/   \/     |_/

px4 starting.

INFO  [px4] Calling startup script: /bin/sh etc/init.d-posix/rcS 0
Info: found model autostart file as SYS_AUTOSTART=10016
INFO  [param] selected parameter default file eeprom/parameters_10016
[param] Loaded: eeprom/parameters_10016
INFO  [dataman] Unknown restart, data manager file './dataman' size is 11798680 bytes
INFO  [simulator] Waiting for simulator to accept connection on TCP port 4560
INFO  [simulator] Simulator connected on TCP port 4560.
INFO  [commander] LED: open /dev/led0 failed (22)
Gazebo multi-robot simulator, version 9.15.0
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[Msg] Waiting for master.
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 192.168.43.121
INFO  [init] Mixer: etc/mixers/quad_w.main.mix on /dev/pwm_output0
INFO  [ekf2] starting instance 0, IMU:0 (1310988), MAG:0 (197388)
INFO  [ekf2] starting instance 1, IMU:1 (1310996), MAG:0 (197388)
INFO  [ekf2] starting instance 2, IMU:2 (1311004), MAG:0 (197388)
INFO  [ekf2] starting instance 3, IMU:0 (1310988), MAG:1 (197644)
INFO  [ekf2] starting instance 4, IMU:1 (1310996), MAG:1 (197644)
INFO  [ekf2] starting instance 5, IMU:2 (1311004), MAG:1 (197644)
INFO  [mavlink] mode: Normal, data rate: 4000000 B/s on udp port 18570 remote port 14550
INFO  [mavlink] mode: Onboard, data rate: 4000000 B/s on udp port 14580 remote port 14540
INFO  [mavlink] mode: Onboard, data rate: 4000 B/s on udp port 14280 remote port 14030
INFO  [mavlink] MAVLink only on localhost (set param MAV_BROADCAST = 1 to enable network)
INFO  [px4] Startup script returned successfully
pxh> INFO  [ecl/EKF] 812000: GPS checks passed
INFO  [ecl/EKF] 812000: GPS checks passed
INFO  [ecl/EKF] 812000: GPS checks passed
INFO  [ecl/EKF] 812000: GPS checks passed
INFO  [ecl/EKF] 824000: GPS checks passed
INFO  [ecl/EKF] 824000: GPS checks passed
INFO  INFO  [ecl/EKF] 2032000: reset position to last known position
[ecl/EKF] 2032000: reset position to last known position
INFO  [ecl/EKF] 2032000: reset velocity to zero
INFO  [ecl/EKF] 2032000: reset position to last known positionINFO  
[ecl/EKF] INFO  [ecl/EKF] 2032000: reset velocity to zero
2032000: reset position to last known position
INFO  [ecl/EKF] 2032000: reset velocity to zero
INFO  [ecl/EKF] 2032000: reset velocity to zero
INFO  [ecl/EKF] 2032000: reset position to last known position
INFO  [ecl/EKF] 2032000: reset velocity to zero
INFO  [ecl/EKF] 2032000: reset position to last known position
INFO  [ecl/EKF] 2032000: reset velocity to zero
INFO  INFO  INFO  [ecl/EKF] 8340000: EKF aligned, (baro hgt, IMU buf: 18, OBS buf: 14)
[ecl/EKF] 8340000: EKF aligned, (baro hgt, IMU buf: 18, OBS buf: 14)
INFO  [ecl/EKF] 8340000: EKF aligned, (baro hgt, IMU buf: 18, OBS buf: 14)
[ecl/EKF] INFO  [ecl/EKF] 8340000: EKF aligned, (baro hgt, IMU buf: 18, OBS buf: 14)
8340000: EKF aligned, (baro hgt, IMU buf: 18, OBS buf: 14)
INFO  [ecl/EKF] 8340000: EKF aligned, (baro hgt, IMU buf: 18, OBS buf: 14)
INFO  [ecl/EKF] 8512000: reset position to GPS
INFO  [ecl/EKF] 8512000: reset position to GPS
INFO  INFO  [ecl/EKF] 8512000: reset velocity to GPS
[ecl/EKF] INFO  [ecl/EKF] 8512000: reset position to GPS
8512000: starting GPS fusion
INFO  [ecl/EKF] 8512000: reset velocity to GPS
INFO  [ecl/EKF] 8512000: starting GPS fusion
INFO  [ecl/EKF] 8512000: reset velocity to GPS
INFO  [ecl/EKF] 8512000: starting GPS fusion
INFO  [ecl/EKF] 8512000: reset position to GPS
INFO  [ecl/EKF] 8512000: reset position to GPS
INFO  [ecl/EKF] 8512000: reset velocity to GPS
INFO  [ecl/EKF] 8512000: reset velocity to GPS
INFO  [ecl/EKF] 8512000: starting GPS fusion
INFO  [ecl/EKF] 8512000: starting GPS fusion
INFO  [ecl/EKF] 8512000: reset position to GPS
INFO  [ecl/EKF] 8512000: reset velocity to GPS
INFO  [ecl/EKF] 8512000: starting GPS fusion
INFO  [tone_alarm] home set
ERROR [tone_alarm] notify negative
ERROR [tone_alarm] notify negative
[Wrn] [GuiIface.cc:119] Populating font family aliases took 244 ms. Replace uses of missing font family "Sans" with one that exists to avoid this cost. 
INFO  [mavlink] partner IP: 127.0.0.1
INFO  [commander] Armed by external command
ERROR [tone_alarm] notify negative
INFO  [commander] Disarmed by external command

```

julianoes, 2020-11-18:
Thanks for the issue. I will try to reproduce this in the next days.

julianoes, 2020-11-24:
This works for me, however, it doesn't actually take off very high. It mostly scrapes across the ground which is not great.

Could you upload your logfile to logs.px4.io?

canberkdurmus, 2020-11-24:
Hi Julian, I removed the PX4-Autopilot repository completely and cloned it again. It works right now, I think the continuous attitude message checking mechanism was not up to date in my local repository.

By the way, 0.8 for climbing and 0.78 for 30 degree roll is fine to be able to see the example.

We can close the issue I think, thank you.

julianoes, 2020-11-24:
Ok, thanks for the feedback!

closed 2020-11-24 by julianoes