mavlink/MAVSDK-Python #287: Mission upload fails when running PX4 SITL and mavsdk_server in 2 different docker containers

ykhedar, 2020-11-20:
I have encountered a problem while using the [mission.py](https://github.com/mavlink/MAVSDK-Python/blob/master/examples/mission.py) script. My setup is as follows

I have two separate docker containers each for [PX4 SITL](https://github.com/JonasVautherin/px4-gazebo-headless) and [mavsdk_server](https://github.com/mavlink/MAVSDK-Go/tree/master/docker) . I use the [docker-compose](https://github.com/mavlink/MAVSDK-Go/tree/master/docker) to run them together and link them. The mavsdk_server is connected to the SITL successfully and is seen also in the log messages.

`mavsdk_server_1         | [09:30:03|Info ] MAVSDK version: v0.34.0 (mavsdk_impl.cpp:20)`
`mavsdk_server_1         | [09:30:03|Debug] New: System ID: 0 Comp ID: 0 (mavsdk_impl.cpp:417)`
`mavsdk_server_1         | [09:30:03|Info ] Server started (grpc_server.cpp:44)`
`mavsdk_server_1         | [09:30:03|Info ] Server set to listen on 0.0.0.0:50051 (grpc_server.cpp:45)`
`mavsdk_server_1         | [09:30:03|Info ] Waiting to discover system on udp://:14540... (connection_initiator.h:21)`
`mavsdk_server_1         | [09:30:04|Info ] New system on: 172.18.0.1:57533 (with sysid: 1) (udp_connection.cpp:190)`
`mavsdk_server_1         | [09:30:04|Debug] Component Autopilot (1) added. (system_impl.cpp:314)`
`mavsdk_server_1         | [09:30:04|Debug] Discovered 1 component(s) (UUID: 5283920058631409231) (system_impl.cpp:471)`
`mavsdk_server_1         | [09:30:04|Info ] System discovered (connection_initiator.h:57)`
`mavsdk_server_1         | [09:30:04|Warn ] command denied (512). (mavlink_commands.cpp:166)`
`mavsdk_server_1         | [09:30:04|Warn ] command denied (512). (mavlink_commands.cpp:166)`
`mavsdk_server_1         | [09:30:05|Debug] Falling back to gimbal protocol v1 (mission_impl.cpp:130)`
`mavsdk_server_1         | [09:30:05|Debug] Falling back to Gimbal Version 1 (gimbal_impl.cpp:58)`
`mavsdk_server_1         | [09:30:05|Debug] MAVLink: info: [logger] ./log/2020-11-20/21_05_59.ulg (system_impl.cpp:229)`

I then use the  [mission.py](https://github.com/mavlink/MAVSDK-Python/blob/master/examples/mission.py) example with the IP address of mavsdk_server set correctly. This script is run on the host machine (also Ubuntu 18.04).  When I run this script, I get the following error:

`mavsdk_server_1         | [09:32:09|Debug] MAVLink: critical: Operation timeoutOperation timeout (system_impl.cpp:229)`

This happens only for the mission.py script. I am able to get the telemetry with the [telemetry.py](https://github.com/mavlink/MAVSDK-Python/blob/master/examples/telemetry.py) script. 

I also tried to upgrade the mavsdk_server to the version 0.34 and PX4 SITL image to the PX4 version 1.11.0 it still creates the same problem.

**What solves the problem**
If I keep the PX4 SITL in the docker but use the mavsdk_server on the host machine, the mission upload works without problem. 

If it is only a TIMEOUT problem, is there a way to increase the timeout limit so that this error does not occur. Or is there another way to solve the problem.

JonasVautherin, 2020-11-23:
Sounds a bit like messages can go from sitl to mavsk_server, but not the other way round. Telemetry is typically `sitl -> mavsdk_server`, while uploading a mission is `mavsdk_server -> sitl`.

Does it work for actions, e.g. arm and takeoff? Or setting a parameter? If my guess is correct, they should fail, too.

ykhedar, 2020-11-23:
I checked the same setup for arm and take-off also and it seems to work. Only thing is that it produces some warning about `landing denied`. Detailed output is as follows:

`mavsdk_server_1         | [07:41:55|Debug] MAVLink: critical: Landing denied! Please land manually (system_impl.cpp:257)`
`mavsdk_server_1         | [07:41:59|Debug] MAVLink: info: Landing detected (system_impl.cpp:257)`

Also the error produced by the mission.py script for mission upload failure is as follows:

`Waiting for mavsdk_server to be ready...`
`Connected to mavsdk_server!`
`Waiting for drone to connect...`
`Drone discovered with UUID: 5283920058631409231`
`-- Uploading mission`
`Traceback (most recent call last):`
`  File "mission.py", line 103, in <module>`
`    loop.run_until_complete(run())`
`  File "/usr/lib/python3.6/asyncio/base_events.py", line 484, in run_until_complete`
`    return future.result()`
`  File "mission.py", line 61, in run`
`   await drone.mission.upload_mission(mission_plan)`
`  File "/usr/local/lib/python3.6/dist-packages/mavsdk/mission.py", line 681, in upload_mission`
`    raise MissionError(result, "upload_mission()", mission_plan)`
`mavsdk.mission.MissionError: TIMEOUT: 'Timeout'; origin: upload_mission(); params: (<mavsdk.mission.MissionPlan object at 0x7efde80d74a8>,)`

After this stacktrace is produced, the mission.py script just hangs and I have to kill it with `ctrl+c`.

I also checked the date-time of both the containers and it seems to be no noticable difference between the two.

JonasVautherin, 2020-11-23:
The time out suggests that you don't receive the ACKs from the drone. How do you test that arm and takeoff work? Sending the "arm" command may arm the drone even if it times out (it means that the message reached the drone, but the response did not come back).

ykhedar, 2020-11-23:
I monitor the gazebo_sitl docker logs for this. The logs confirm that take off was detected after being armed by an external command. Here is a snippet:

`gazebo_sitl_headless_1  | INFO  [commander] Armed by external command`
`gazebo_sitl_headless_1  | ERROR [commander] vehicle_command lost, generation 1 -> 4`
`gazebo_sitl_headless_1  | ERROR [navigator] vehicle_command lost, generation 1 -> 4`
`gazebo_sitl_headless_1  | INFO  [navigator] Using minimum takeoff altitude: 2.50 m`
`gazebo_sitl_headless_1  | INFO  [navigator] Using minimum takeoff altitude: 2.50 m`
`gazebo_sitl_headless_1  | INFO  [navigator] Using minimum takeoff altitude: 2.50 m`
`gazebo_sitl_headless_1  | INFO  [commander] Takeoff detected`
`gazebo_sitl_headless_1  | INFO  [commander] Landing at current position`
`gazebo_sitl_headless_1  | WARN  [commander] Landing denied! Please land manually`
`gazebo_sitl_headless_1  | WARN  [commander] Landing denied! Please land manually`
`gazebo_sitl_headless_1  | INFO  [commander] Landing detected`
`gazebo_sitl_headless_1  | INFO  [commander] Disarmed by landing`
`gazebo_sitl_headless_1  | INFO  [logger] closed logfile, bytes written: 9993034`

Is there any other way to confirm this?

julianoes, 2020-11-24:
Can you try to also open the port `14580` on the MAVSDK docker side?

https://github.com/PX4/PX4-Autopilot/blob/5e855c67477fe08517297a38e718ed1102536cf6/ROMFS/px4fmu_common/init.d-posix/rcS#L108

JonasVautherin, 2020-11-24:
That's weird. So you get messages from SITL (like telemetry), and some messages reach SITL from `mavsdk_server` (arm, takeoff), but some don't (mission), or just time out for some reason. And it works if `mavsdk_server` is running on the host and not in a container, which suggests that it is a networking issue.

:thinking:

ykhedar, 2020-11-24:
> Can you try to also open the port 14580 on the MAVSDK docker side?

Makes no difference

> And it works if mavsdk_server is running on the host and not in a container, which suggests that it is a networking issue.

Yes i think so too. I will investigate further when I have more time. I have noticed one difference though between the logs of mavsdk_server:

**MAVSDK Server outside docker - working version**

`[09:52:51|Info ] MAVSDK version: v0.34.0 (mavsdk_impl.cpp:20)`
`[09:52:51|Debug] New: System ID: 0 Comp ID: 0 (mavsdk_impl.cpp:417)`
`[09:52:51|Info ] Server started (grpc_server.cpp:44)`
`[09:52:51|Info ] Server set to listen on 0.0.0.0:50051 (grpc_server.cpp:45)`
`[09:52:51|Info ] Waiting to discover system on udp://:14540... (connection_initiator.h:21)`
`[09:52:51|Info ] New system on: 172.18.0.2:14580 (with sysid: 1) (udp_connection.cpp:190)`
`[09:52:51|Debug] Component Autopilot (1) added. (system_impl.cpp:314)`
`[09:52:52|Debug] Discovered 1 component(s) (UUID: 5283920058631409231) (system_impl.cpp:471)`
`[09:52:52|Info ] System discovered (connection_initiator.h:57)`

Notice that it discovers only one system `New system on: 172.18.0.2:14580`

**MAVSDK Server inside docker - NOT working version**

`mavsdk_server_1         | [08:35:45|Info ] MAVSDK version: v0.34.0 (mavsdk_impl.cpp:20)`
`mavsdk_server_1         | [08:35:45|Debug] New: System ID: 0 Comp ID: 0 (mavsdk_impl.cpp:417)`
`mavsdk_server_1         | [08:35:45|Info ] Server started (grpc_server.cpp:44)`
`mavsdk_server_1         | [08:35:45|Info ] Server set to listen on 0.0.0.0:50051(grpc_server.cpp:45)`
`mavsdk_server_1         | [08:35:45|Info ] Waiting to discover system on udp://:14540... (connection_initiator.h:21)`
`mavsdk_server_1         | [08:35:51|Info ] New system on: 172.18.0.3:14580 (with sysid: 1) (udp_connection.cpp:190)`
`mavsdk_server_1         | [08:35:51|Debug] Component Autopilot (1) added. (system_impl.cpp:314)`
`mavsdk_server_1         | [08:35:51|Info ] New system on: 172.18.0.1:55514 (with sysid: 1) (udp_connection.cpp:190)`
`mavsdk_server_1         | [08:35:51|Debug] Discovered 1 component(s) (UUID: 5283920058631409231) (system_impl.cpp:471)`
`mavsdk_server_1         | [08:35:51|Info ] System discovered (connection_initiator.h:57)`

Notice that it detects two systems `172.18.0.3:14580 (with sysid: 1) ` and `172.18.0.1:55514 (with sysid: 1)` . Does this lead to a conflict for mavsdk server to resolve?

JonasVautherin, 2020-11-24:
aha! Yes that's weird, and one of those systems will just be ignored. I would try to add, in `gazebo_sitl_headless` (docker-compose), the command arguments. You know, when running px4-gazebo-headless, you run something like:

```
docker run --rm -it jonasvautherin/px4-gazebo-headless:1.11.0 192.168.0.12
```

And MAVLink is sent to `192.168.0.12:14540`. If you don't give an argument (which is your case right now), then it sends MAVLink on `172.17.0.255:14540` (or whatever broadcast address it is), which reaches the host, but I don't know what happens when `mavsdk_server` is also in a container. So I would want to try the equivalent of:

```
docker run --rm -it jonasvautherin/px4-gazebo-headless:1.11.0 mavsdk_server
```

`mavsdk_server` above is a hostname, it should work (i.e. you should be able to `ping mavsdk_server` when docker-compose starts it). So in order to do that from docker compose, I think it's with [`command`](https://docs.docker.com/compose/compose-file/#command) in the `docker-compose.yml`. So something like:

```
    gazebo_sitl_headless:
        image: jonasvautherin/px4-gazebo-headless:v1.10.1
        environment:
            - NO_PXH=1
        links:
            - mavsdk_server
        command: mavsdk_server
```

Can you try that?

ykhedar, 2020-12-19:
sorry for the long delay. I checked your suggestion today and it works when I add the IP address of the MAVSDK Server container to the SITL container as a command. 

So basically I run first the docker containing mavsdk server and get its IP address.

Then I run the sitl container as you suggested:
`docker run --rm -it jonasvautherin/px4-gazebo-headless:1.11.0 IP_ADDRESS_MAVSDK_CONTAINER` 

simply using the host name 

> mavsdk_server

somehow does not work, so you need to use the actual IP of the mavsdk container.
Due to this same problem, I was not able to make it work with the docker compose since you need to know the IP address of the containers before hand. I could do static IP for all containers using custom network but somehow it did not work on first try. This could be a different issue altogether. For now we can close this issue. Thanks for your help :)

closed 2020-12-19 by ykhedar