mavlink/MAVSDK-Python #289: mavsdk.info.InfoError: INFORMATION_NOT_RECEIVED_YET

hashimK, 2020-12-08:
Hi I'm using MVSDK-Python for quite some while and its helpful in many ways.
Recently I tried using MAVSDK-Python to work on raspberry pi which is a companion computer with Pixhawk 4.
The serial connection between Pi and Pixhawk4 is on /dev/ttyAMA0 port and the MAVLINK data is routed to udp port 13000 using mavlink-router.
While the `code_A.py ` works on the Pi, I get an error when I try to run `code_B.py`.
Also until yesterday `code_B.py` ran without errors on Pi, but today I'm getting this error `mavsdk.info.InfoError: INFORMATION_NOT_RECEIVED_YET` (full error below)

**code_A.py** (this works as expected)
```
!/usr/bin/env python3

import asyncio
from mavsdk import System

async def run():
    # Init the drone
    drone = System()
    await drone.connect(system_address="udp://:13000")

    # Start the tasks
    asyncio.ensure_future(print_battery(drone))
    asyncio.ensure_future(print_gps_info(drone))
    asyncio.ensure_future(print_in_air(drone))
    asyncio.ensure_future(print_position(drone))

async def print_battery(drone):
    async for battery in drone.telemetry.battery():
        print(f"Battery: {battery.remaining_percent}")

async def print_gps_info(drone):
    async for gps_info in drone.telemetry.gps_info():
        print(f"GPS info: {gps_info}")

async def print_in_air(drone):
    async for in_air in drone.telemetry.in_air():
        print(f"In air: {in_air}")

async def print_position(drone):
    async for position in drone.telemetry.position():
        print(position)

if __name__ == "__main__":
    # Start the main function
    asyncio.ensure_future(run())

    # Runs the event loop until the program is canceled with e.g. CTRL-C
    asyncio.get_event_loop().run_forever()
```

**code_B.py**
```
#!/usr/bin/env python

import asyncio
from mavsdk import System

async def run():

    # Init the drone
    vehicle = System()
    await vehicle.connect(system_address="udp://:13000")

    # vehicle = System(mavsdk_server_address="localhost", port=50040)
    # await vehicle.connect()

    async for state in vehicle.core.connection_state():
        if state.is_connected:
            print(f"Vehicle discovered with UUID: {state.uuid}")
            break

    id_info = await vehicle.info.get_identification()
    print(id_info)

if __name__ == '__main__':
    # Start the main function
    asyncio.ensure_future(run())

    # Runs the event loop until the program is canceled with e.g. CTRL-C
    asyncio.get_event_loop().run_forever()
```

**code_B.py output**
```
Waiting for mavsdk_server to be ready...
Connected to mavsdk_server!
Vehicle discovered with UUID: 0
Task exception was never retrieved
future: <Task finished coro=<run() done, defined at drone_id.py:6> exception=InfoError(<mavsdk.info.InfoResult object at 0x7528a2f0>, 'get_identification()')>
Traceback (most recent call last):
  File "drone_id.py", line 20, in run
    id_info = await vehicle.info.get_identification()
  File "/home/pi/.local/lib/python3.7/site-packages/mavsdk/info.py", line 666, in get_identification
    raise InfoError(result, "get_identification()")
mavsdk.info.InfoError: INFORMATION_NOT_RECEIVED_YET: 'Information Not Received Yet'; origin: get_identification(); params: ()
```

I'm not able to understand why this line (`id_info = await vehicle.info.get_identification()`) is throwing error.
Both the codes work on Ubuntu PC, only the `code_B.py` doesn't work on Pi. Also, to add to the weirdness `code_B.py `was working yesterday on the same Pi.

I'm using the latest Raspian Buster OS.

Thanks.

julianoes, 2020-12-09:
> ` Vehicle discovered with UUID: 0`

This doesn't look like a vehicle was actually discovered. Are you sure the connection is working properly? 

```
    # vehicle = System(mavsdk_server_address="localhost", port=50040)
    # await vehicle.connect()
```
Why is this commented out?

hashimK, 2020-12-09:
Hi @julianoes ,
The connection is working properly as the `code_A.py` runs perfectly fine on raspberry Pi and prints proper vehicle position and altitude.
If I put the below code in `code_A.py` it still prints the same `Vehicle discovered with UUID: 0`.
```
async for state in vehicle.core.connection_state():
        if state.is_connected:
            print(f"Vehicle discovered with UUID: {state.uuid}")
            break
```

Also I've always seen `state.uuid` as zero in almost all my codes whether vehicle is real or in HITL Mode.
Anyways that's not a problem. The problem is on this line `id_info = await vehicle.info.get_identification()` in `code_B.py`.
It gives an error on raspberry Pi but not on a normal Ubuntu PC.
Thanks.

julianoes, 2020-12-09:
This still sounds like a communication problem. Can you run `mavsdk_server` separately and get the debug output there?

You should find it somewhere in
`./.local/lib/python3.7/site-packages/mavsdk/bin/mavsdk_server` if you did `pip install --user mavsdk`
or `/usr/lib/python3.7/site-packages/mavsdk/bin/mavsdk_server` if you did `sudo pip install mavsdk`.

You can run it with `mavsdk_server udp://:13000` and then run the Python script without `vehicle.connect`.

JonasVautherin, 2020-12-10:
Just to be sure: you are aware that you can `try... except` that error, right?

Something like this (haven't tried it):

```
try:
    id_info = await vehicle.info.get_identification()
except InfoError as infoError:
    <do something with infoError>
```

@julianoes: could it be a timing error? Like the information has not been received yet? In which case the error could be caught, and the code could try again later... Or is the information supposed to be received very early, hinting for a communication problem?

hashimK, 2020-12-10:
Its working now by both methods on Raspberry Pi. (`code_B.py`). 

```
vehicle = System()
await vehicle.connect(system_address="udp://:14540")
```
and 

```
vehicle = System(mavsdk_server_address="localhost", port=50040)
await vehicle.connect()
```

Also the output is same for both these methods
```
Waiting for mavsdk_server to be ready...
Connected to mavsdk_server!
Vehicle discovered with UUID: 0
Identification: [hardware_uid: 00020000000035333837313851180046002f]
```
The mavsdk_server debug output is
```
[12:41:26|Info ] MAVSDK version: v0.34.0 (mavsdk_impl.cpp:20)
[12:41:26|Debug] New: System ID: 0 Comp ID: 0 (mavsdk_impl.cpp:417)
[12:41:26|Info ] Server started (grpc_server.cpp:44)
[12:41:26|Info ] Server set to listen on 0.0.0.0:50040 (grpc_server.cpp:45)
[12:41:26|Info ] Waiting to discover system on udp://:14540... (connection_initiator.h:21)
[12:45:29|Info ] New system on: 127.0.0.1:43242 (with sysid: 1) (udp_connection.cpp:190)
[12:45:29|Debug] Component Autopilot (1) added. (system_impl.cpp:314)
[12:45:29|Debug] Discovered 1 component(s) (UUID: 3546673870967879735) (system_impl.cpp:471)
[12:45:29|Warn ] command denied (512). (mavlink_commands.cpp:166)
[12:45:29|Warn ] command denied (512). (mavlink_commands.cpp:166)
[12:45:30|Debug] Falling back to Gimbal Version 1 (gimbal_impl.cpp:58)
[12:45:30|Debug] Falling back to gimbal protocol v1 (mission_impl.cpp:130)
[12:45:31|Debug] MAVLink: info: No airspeed sensor detected. Switch to non-airspee (system_impl.cpp:229)
[12:45:31|Debug] MAVLink: info: d mode. (system_impl.cpp:229)
[12:46:33|Info ] heartbeats timed out (system_impl.cpp:235)
[12:46:33|Debug] Lost 3546673870967879735 (mavsdk_impl.cpp:450)
[12:46:36|Debug] Discovered 1 component(s) (UUID: 3546673870967879735) (system_impl.cpp:471)
[12:46:36|Warn ] command denied (512). (mavlink_commands.cpp:166)
[12:46:36|Warn ] command denied (512). (mavlink_commands.cpp:166)
[12:46:37|Debug] Falling back to gimbal protocol v1 (mission_impl.cpp:130)
[12:46:37|Debug] Falling back to Gimbal Version 1 (gimbal_impl.cpp:58)
[12:46:38|Debug] MAVLink: info: No airspeed sensor detected. Switch to non-airspee (system_impl.cpp:229)
[12:46:38|Debug] MAVLink: info: d mode. (system_impl.cpp:229)
```

Seems weird as `code_B.py` wasn't working before, but under the same setup, hardware connections, etc `code_A.py` was working fine. 

Does this line `await vehicle.info.get_identification()` tries to send some acknowledgement back to px4, once it receives data?
Like any acknowledgement data passed from raspberry Pi to Pixhawk4?

JonasVautherin, 2020-12-10:
MAVSDK needs to receive the data from PX4. Until it does, if you call `get_identification()`, it will raise an error (that you should catch and act accordingly). My feeling is that for some reason, sometimes you call `get_identification()` before the information has reached MAVSDK from PX4 (either the message gets lost, or takes time to be sent, whatever).

So you may need to try multiple times, or just to wait a bit longer before you call `get_identification()`.

hashimK, 2020-12-10:
Ok thanks will implement error handling henceforth on every await calls.
Thanks for the explanation.

closed 2020-12-10 by hashimK