mavlink/MAVSDK-Python #290: Could not connect mavsdk to external PX4 SITL via ethernet

copterspace, 2020-12-17:
Dear colleagues, I have PX4 SITL Jmavsim running on Ubuntu computer, ip address is 192.168.1.17
I try to connect to this SITL from Raspberry PI onboard computer via ethernet.
Mavros connection with `roslaunch mavros px4.launch fcu_url:="udp://@192.168.1.17:14580"` works OK - I get valid mavlink heartbit with `rostopic echo /mavros/state`
<br>But this my attempt to connect to this SITL via mavsdk-python did not succeed:
```python
drone = System()
await drone.connect(system_address = 'udp://192.168.1.17:14580')
```
It starts mavsdk_server on Raspberry PI and then - nothing happens:
```
Waiting for mavsdk_server to be ready...
Connected to mavsdk_server!
Waiting for drone to connect...
```
I tried both 'udp://192.168.1.17:14580' and 'tcp://192.168.1.17:14580', different ports - result is the same - no connection.
<br>Although in the [API reference](http://mavsdk-python-docs.s3-website.eu-central-1.amazonaws.com/system.html) it's specified that it is possible to use external system UDP or TCP `system_address`.
<br>In [Getting Started presentation](https://auterion.com/getting-started-with-mavsdk-python/) said that is should work out of the box - but how to setup communication with external IP system properly?

Output from SITL mavlink status is the following:
```
instance #0:
	mavlink chan: #0
	type:		GENERIC LINK OR RADIO
	flow control: OFF
	rates:
	  tx: 13.865 kB/s
	  txerr: 0.000 kB/s
	  tx rate mult: 1.000
	  tx rate max: 4000000 B/s
	  rx: 0.000 kB/s
	FTP enabled: YES, TX enabled: YES
	mode: Normal
	MAVLink version: 2
	transport protocol: UDP (18570, remote port: 14550)
	partner IP: 192.168.1.39

instance #1:
	mavlink chan: #1
	type:		GENERIC LINK OR RADIO
	flow control: OFF
	rates:
	  tx: 14.448 kB/s
	  txerr: 0.000 kB/s
	  tx rate mult: 1.000
	  tx rate max: 4000000 B/s
	  rx: 0.000 kB/s
	FTP enabled: YES, TX enabled: YES
	mode: Onboard
	MAVLink version: 1
	transport protocol: UDP (14580, remote port: 14540)

instance #2:
	mavlink chan: #2
	type:		GENERIC LINK OR RADIO
	flow control: OFF
	rates:
	  tx: 2.894 kB/s
	  txerr: 0.000 kB/s
	  tx rate mult: 0.141
	  tx rate max: 4000 B/s
	  rx: 0.000 kB/s
	FTP enabled: YES, TX enabled: YES
	mode: Onboard
	MAVLink version: 1
	transport protocol: UDP (14280, remote port: 14030)
```
QGS from external workstation also connects to this SITL OK.

JonasVautherin, 2020-12-17:
MAVSDK-Python expects to receive the heartbeats from jMAVSim, and will not initiate the connection. Are you sure that your jMAVSim is sending heartbeats on 14580? By default it does it on 14550 and 14540.

Also, you want to use something like `udp://:14580` (i.e. not specify the URL there, because that's a filter).

Can you try with [px4-gazebo-headless](https://github.com/JonasVautherin/px4-gazebo-headless)?

1. Run SITL with `docker run --rm -it jonasvautherin/px4-gazebo-headless:1.11.0 <ip_of_rpi>`
2. On the RPi, run with `await drone.connect(system_address = 'udp://:14540')`

copterspace, 2020-12-18:
Thank you for fast reply, Jonas!
Yes, with your docker container with specified PRI IP works OK, Drone is detected on mavsdk, shows flightmode, etc.
I see you add provided IP to rcS Firmware file in https://github.com/JonasVautherin/px4-gazebo-headless/blob/master/edit_rcS.bash:
```
CONFIG_FILE=${FIRMWARE_DIR}/build/px4_sitl_default/etc/init.d-posix/rcS

sed -i "s/mavlink start \-x \-u \$udp_gcs_port_local -r 4000000/mavlink start -x -u \$udp_gcs_port_local -r 4000000 ${QGC_PARAM}/" ${CONFIG_FILE}
sed -i "s/mavlink start \-x \-u \$udp_offboard_port_local -r 4000000 -m onboard -o \$udp_offboard_port_remote/mavlink start -x -u \$udp_offboard_port_local -r 4000000 -o \$udp_offboard_port_remote ${API_PARAM}/" ${CONFIG_FILE}

echo 'param set MAV_BROADCAST 1' >> ${CONFIG_FILE}
```
And you also change default MAV_BROADCAST param to 1.
Should I manually make similar changes in my clone of PX4 firmware to make my setup work with jmavsim?
mavros works without MAV_BROADCAST = 1 and other Firmware changes, does it use different connection algorithm?...

JonasVautherin, 2020-12-18:
MAVSDK does not initiate the connection, as in it does not send heartbeats before it receives heartbeats. In C++, you can set it up to do that (with `add_udp_remote`), but that is not exposed in MAVSDK-Python. I believe that MAVROS allows for that, which is why it works for you.

One way is to edit the rcS file (no need to change MAV_BROADCAST in your case, I think). Another way would be to run mavlink-router on your RPi and have it send the messages to your laptop (which is somehow equivalent to changing the rcS).

julianoes, 2021-03-26:
No more follow up, closing.

closed 2021-03-26 by julianoes