mavlink/MAVSDK-Python #291: How to attach the Mocap pose data to PX4?

joker-csee, 2020-12-23:
Hi there,

Recently, I try to use mavsdk to pass the pose data captured by mocap(OptiTrack) to PX4, while two questions coming up.

1. I notice that async **set_attitude_position_mocap** needs a class named mavsdk.mocap.**Covariance**, which mocap doesn't provide. Is that necessary？If so, how can I get it (already have 'x y  z dot_x dot_y dot _z phi theta psi' data)?
2. What params should I set to ensure the mocap system has the same **origin** of coordinates with PX4, to make sense of the pose data?

Any help is grateful! Thanks!

JonasVautherin, 2020-12-24:
@irsdkv: Would you be able to help here? :blush:

irsdkv, 2020-12-24:
Hi all

1. Actually I didn't use MAVSDK-**Python** for a long time so I can be wrong. But now I see that `Covariance` class exists in `main` branch:
https://github.com/mavlink/MAVSDK-Python/blob/main/mavsdk/mocap.py#L369
I think that you shall try to re-setup package going thru documentation and correctly `import` all needed things.

2. About origin. After reboot, PX4 have no **Local** position by default. But when MoCap "server" starts to feed it with this position -  PX4 assume that it is its true position. So for example if with **Server** point of view PX4 have `x: 1., y: 2., z:0.` coordinates so PX4 estimator will accept them as the only true ones. 
If you ask about real autopilot pose related to, for example, Aruco marker used to determine drone pose, you should correctly adjust parameters [EKF2_EV_POS_X/Y/Z](https://dev.px4.io/v1.9.0/en/ros/external_position_estimation.html#ekf2-tuningconfiguration)
Please say if I misunderstood your question.

And additional potentially trouble with MoCap and PX4 for this time (at least PX4 `v0.11.1`):

At this time you should prefer to use [set_vision_position_estimate](https://github.com/mavlink/MAVSDK-Python/blob/27a7e6027e20c4aa85e3806048136ab1a585b18b/mavsdk/mocap.py#L1132) with default Autopilot firmware because of `EKF2` now [use only vehicle_visual_odometry](https://dev.px4.io/v1.9.0/en/ros/external_position_estimation.html#px4-mavlink-integration) uorb topic.

irsdkv, 2020-12-24:
And one more thing about Covariance.
You can skip its calculation but in this case you should set some array cells to `nan`:
https://github.com/PX4/PX4-Autopilot/blob/9fcbf18d82b84ac57ee13782b3e501fa077d4d97/msg/vehicle_odometry.msg#L61

joker-csee, 2020-12-28:
Probelms were perfectly solved with your suggestion.
Thanks a lot!

closed 2020-12-28 by joker-csee