mavlink/MAVSDK-Python #292: Can connect to MAVSK-PYTHON on Windows

Traveller30, 2020-12-23:
Hi
I have MAVSDK-PYTHON 0.14.0  installed on 2 machines , UBUNTU and WINDOWS on PYCharm in both machines.
The Ubuntu version connects and runs the following script with no problems but the Windows version won't connect

```
import mavsdk
import asyncio

 async def start():
    drone = mavsdk.System()
    await drone.connect(system_address="udp://:12002")

    async for state in drone.core.connection_state():
        if state.is_connected:
            print(f"Drone connected with UUID: {state.uuid}.")
            break
    async for position in drone.telemetry.position():
        print(f"Pos: {position.absolute_altitude_m}")
        break
    async for mission in drone.mission_raw.mission_changed():
        print(f"Mission state changed: {mission}")

 asyncio.ensure_future(start())

 asyncio.get_event_loop().run_forever()
```
I get the following response:

> Waiting for mavsdk_server to be ready...

The mavlink stream is transmitted on UDP:12002 and I checked that I can connect to it with QGC 

What am I missing?
thanks in advance

JonasVautherin, 2020-12-24:
Seems like `mavsdk_server` fails to start. Can you try to

* Download [mavsdk_server_win32.exe](https://github.com/mavlink/MAVSDK/releases/download/v0.35.0/mavsdk_server_win32.exe) (from the [latest release](https://github.com/mavlink/mavsdk/releases))
* Run it manually (setting the port to e.g. 50051 with the corresponding argument `-p 50051`
* Instantiate the `drone` with `drone = mavsdk.System(mavsdk_server_address='localhost', port=50051)` (such that MAVSDK-Python connects to this manually-started instance)

The `mavsdk_server` output should show you something (e.g. if it detects the drone or not).

Traveller30, 2020-12-25:
Thank you Jonas
Found the problem
When tried to run the file you sent: **mavsdk_server_win32.exe** manually got an error:"**VCRUNTIME140_1.dll is Missing**". 
after some googling found that its "Microsoft C++ Redistributable for Visual Studio" file.
Reinstalled 2015 version - didn't work
Reinstalled 2017 version - didn't work
Reinstalled 2019 version - **WORKED**!
Now the server is running from MAVSDK library or manually from running mavsdk_server_win32 and I can connect to the stream
I think its a good idea to add to the instructions for Windows version: "installing the suitable Microsoft C++ Redistributable"

hrnbot, 2021-01-04:
@Traveller30 Close the issue if it is solved.

JonasVautherin, 2021-01-04:
Nice! Thanks for reporting!

> I think its a good idea to add to the instructions for Windows version: "installing the suitable Microsoft C++ Redistributable"

@Traveller30: Would you mind contributing that to the docs?

closed 2021-01-04 by JonasVautherin

FrMohamedFaris, 2025-03-18:
As same, I got the error; I really don't know how to run it properly. And I even tried to open the file in command prompt to run with the.exe file. But it's not working.

julianoes, 2025-03-18:
@FrMohamedFaris please don't necrobump and create a new issue with enough info of what you did and what the error was.