mavlink/MAVSDK-Python #293: ServoPWM Output in Telemtry mssg

Traveller30, 2020-12-28:
Hi
Is it possible to connect to the "Servo_Output_Raw" mssg? 
I'm trying to see the pwm cmd's I sent to motors/servos and couldn't find anything in Telemetry beside "ActuatorOutputStatus"
,but it wont return anything

tnx

JonasVautherin, 2021-01-04:
What MAVLink message are you interested in? Should it be this `ActuatorOutputStatus`? Is it sent by PX4, or do you need to set the rate to have it send those messages?

Traveller30, 2021-01-07:
I need the "SERVO_OUTPUT_RAW" mssg , so I can see what pwm values are sent to the motors. 
Couldn't find in MAVSDK-PYTHON API
It is a regular Mavlink PX4 mssg . I can see the it in QGC .

JonasVautherin, 2021-01-07:
Apparently it's not supported indeed. Would you be willing to contribute it?

Traveller30, 2021-01-07:
I would love to
Just need to learn how :-)

JonasVautherin, 2021-01-07:
Maybe [this](https://youtu.be/T1orxSyqDzI) can help

NanyangYe, 2023-06-10:
Hi, any updates on this? thanks!

Traveller30, 2023-06-12:
couldn't find a solution
switched to PyMavlink , all mavlink mssgs supported there