mavlink/MAVSDK-Python #297: MissionRaw upload results in ' UNSUPPORTED'

DoppleGangster, 2021-01-12:
I'm currently attempting to use the orbit feature of MAVSDK with the mission_raw command. The result of the following code outputs the following error:

'mavsdk.mission_raw.MissionRawError: UNSUPPORTED: 'Unsupported''

I was hoping for some guidance on using the raw mission command as it would be beneficial for my work.

Thanks for your help!

`
import asyncio

from mavsdk import System
from mavsdk.mission_raw import MissionItem
from mavsdk import mission_raw

MAV_CMD_DO_ORBIT = 34
ORBIT_YAW_BEHAVIOUR = 0
MAV_MISSION_TYPE_MISSION = 0
MAV_FRAME_MISSION = 2

async def run():
    drone = System()
    await drone.connect()

    print("Waiting for drone to connect...")
    async for state in drone.core.connection_state():
        if state.is_connected:
            print(f"Drone discovered with UUID: {state.uuid}")
            break

    print_mission_progress_task = asyncio.ensure_future(print_mission_progress(drone))

    running_tasks = [print_mission_progress_task]
    termination_task = asyncio.ensure_future(observe_is_in_air(drone, running_tasks))

    mission_items = []
    mission_items.append(MissionItem(0,#Seq
                                     MAV_FRAME_MISSION,#frame
                                     MAV_CMD_DO_ORBIT,#command
                                     1,#current
                                     0,#autocontinue
                                     5.0,#param1
                                     float('nan'),#param2
                                     ORBIT_YAW_BEHAVIOUR,#param3
                                     0,#param4
                                     1,#x float('nan') Using the nan results in error.The documenation for orbit states:'NaN: Use current vehicle position or current center if already orbiting.'
									 1,#y float('nan') 
									 25.0,#z
									 MAV_MISSION_TYPE_MISSION))

    print("-- Uploading mission")
    await drone.mission_raw.upload_mission(mission_items)

    print("-- Arming")
    await drone.action.arm()

    print("-- Starting mission")
    await drone.mission_raw.start_mission()

    await termination_task

async def print_mission_progress(drone):
    async for mission_progress in drone.mission_raw.missionProgress():
        print(f"Mission progress: "
              f"{mission_progress.current}/"
              f"{mission_progress.total}")

async def observe_is_in_air(drone, running_tasks):
    """ Monitors whether the drone is flying or not and
    returns after landing """

    was_in_air = False

    async for is_in_air in drone.telemetry.in_air():
        if is_in_air:
            was_in_air = is_in_air

        if was_in_air and not is_in_air:
            for task in running_tasks:
                task.cancel()
                try:
                    await task
                except asyncio.CancelledError:
                    pass
            await asyncio.get_event_loop().shutdown_asyncgens()

            return

if __name__ == "__main__":
    loop = asyncio.get_event_loop()
    loop.run_until_complete(run())
`

dayjaby, 2021-01-12:
Orbit can't be part of a mission (yet).

DoppleGangster, 2021-01-12:
Is there any other way to execute the command currently?

Thanks!

JonasVautherin, 2021-01-13:
@dayjaby: thanks for the insights! Do you mean that `MAV_CMD_DO_ORBIT` is not supported by PX4 in mission mode, is that right? 

But then does the command work when not in mission mode? I have never used orbit, but whatever is supported by PX4, we should be able to add to MAVSDK :+1:.

DoppleGangster, 2021-01-13:
MISSIONRAW was the only location in the API reference that I could find for inputting a not already predefined action. So I have not tested the action in any other way besides the mission format.

dayjaby, 2021-01-13:
@JonasVautherin yes, in PX4 it is only supported as standalone command. PX4 goes into a special flight mode for orbit, which is is semi-automatic (depends quite a bit on the exact version of PX4 as there is ongoing development). Is the sending of a simple COMMAND_LONG documented somewhere for MAVSDK? I couldn't find any documents.

JonasVautherin, 2021-01-13:
Right, thanks @dayjaby for the insights! :blush: 

This means that it cannot go through the `mission_raw` plugin, and the `mavlink_passthrough` plugin is unfortunately only available in C++. I would suggest @DoppleGangster contributes that as a feature to MAVSDK-C++, such that it then automatically propagates to the language bindings (including MAVSDK-Python).

For instance, it could be `action.orbit(...)` in the `action` plugin, or if there is a more complex API around the orbit mode, it could even become an `orbit` plugin. Really open to the discussion here, and happy to give guidance if @DoppleGangster wants to try contributing it. Just let me know, I won't spend time describing the steps to follow if nobody is interested :see_no_evil: :smile:.

JonasVautherin, 2021-01-13:
(Note: I tried to show the process in a 20-minutes live session during the last devsummit, [that may be a starting point](https://www.youtube.com/watch?v=T1orxSyqDzI&list=PLYy2pGCdhu7x1BpFcPsLY72D6z3AUdZWf&index=28))

DoppleGangster, 2021-01-13:
Watched the video and seems straight forward enough. I'll give it a go and try and implement orbit as an action. Assuming I don't run into issues I'll contribute my changes.

Thanks for the responses.

DoppleGangster, 2021-01-18:
@JonasVautherin after following the video and attempting to compile I get the following error:

/home/drone1/Documents/MAVSDK/src/mavsdk_server/test/../src/plugins/action/action_service_impl.h:323:31: error: ‘class testing::NiceMock<mavsdk::testing::MockAction>’ has no member named ‘do_orbit’
  323 |         auto result = _action.do_orbit(request->radius_m(), request->velocity_ms(), request->yaw_behavior(), request->latitude_x(), request->longitude_y(), request->altitude_z());

Is there another step that I may have missed or perhaps I implemented one of the steps incorrectly?

Thanks for your help.

JonasVautherin, 2021-01-18:
Yes, it needs to be mocked [here](https://github.com/mavlink/MAVSDK/blob/develop/src/plugins/action/mocks/action_mock.h).

Something like (I just wrote types randomly, they have to correspond to your `do_orbit` prototype, but you get the idea):

```
MOCK_CONST_METHOD6(do_orbit, Action::Result(double, double, YawBehavior, double, double, double)){};
```

DoppleGangster, 2021-01-18:
Awesome that did the trick. Hopefully I get a chance to test the command today.

Thanks again!

DoppleGangster, 2021-01-18:
Alright next question. Is there a way to regenerate the python wrapper with my local build of MAVSDK?

JonasVautherin, 2021-01-19:
The link between both repositories (`mavlink/mavsdk` and `mavlink/mavsdk-python`) is the proto submodule (`mavlink/mavsdk-proto`). I gather you already modified it in `mavlink/mavsdk`. You need to propagate those changes into your clone of `mavlink/mavsdk-python`. Either by copying `mavsdk/proto` to `mavsdk-python/proto`, or by making your mavsdk-python clone point to your fork of mavsdk-proto.

Then, you can generate the python wrapper with `./other/tools/run_protoc.sh`.

Let me try to give an example, because I understand it may be a bit confusing :sweat_smile:.

### Example

Say you have `DoppleGangster/mavsdk` in /tmp/mavsdk and `DoppleGangster/mavsdk-python` in /tmp/mavsdk-python.

```
cd /tmp/mavsdk-python

# Synchronize proto/ between both clones
rm -rf proto
cp -r /tmp/mavsdk/proto .

# Generate the python bindings in a venv
python3 -m venv venv
pip install protoc-gen-mavsdk
./other/tools/run_protoc.sh
```

To go further, note that `cp` the proto changes is not exactly the nicest way. Actually, you will need to make a fork of `mavlink/mavsdk-proto` and make a PR with your changes. Your fork will be `DoppleGangster/mavsdk-proto`, and both your clones `DoppleGangster/mavsdk` and `DoppleGangster/mavsdk-python` could point to that fork, such that they can access your changes. But that's not needed, `cp` will work just fine.

DoppleGangster, 2021-01-19:
@JonasVautherin great makes sense! Hopefully I'll get some time later this week to continue testing. 

Really appreciate your responses.

DoppleGangster, 2021-01-24:
When attempting to run the action I get an error that the status is unimplemented. Is this indicating that the orbit command from MAVSDK is not currently a functional command?

Below is the error:

response = await self._stub.DoOrbit(request)
  File "/home/drone1/Documents/MAVSDK/venv/lib/python3.8/site-packages/aiogrpc/channel.py", line 40, in __call__
    return await fut
grpc._channel._MultiThreadedRendezvous: <_MultiThreadedRendezvous of RPC that terminated with:
	status = StatusCode.UNIMPLEMENTED
	details = ""
	debug_error_string = "{"created":"@1611455209.308076891","description":"Error received from peer ipv6:[::1]:50051","file":"src/core/lib/surface/call.cc","file_line":1067,"grpc_message":"","grpc_status":12}"

JonasVautherin, 2021-01-24:
You need to run your newly-built `mavsdk_server`, right? How are you connecting to the drone from python right now? Are you using your custom `mavsdk_server`?

DoppleGangster, 2021-01-24:
In the MAVSDK folder I performed the following:

cmake -DCMAKE_BUILD_TYPE=Debug -DBUILD_SHARED_LIBS=ON -Bbuild/default -H.
cmake --build build/default
sudo cmake --build build/default --target install
sudo ldconfig

For the Python MAVSDK folder I performed the steps you had described above.

I'm using the jmavsim to test my code.

JonasVautherin, 2021-01-24:
You should start `mavsdk_server` manually, it will be in `build/default/src/mavsdk_server/src/mavsdk_server`, I think.

Set the port when starting it, something like: `./mavsdk_server -p 50051 udp://:14540`. `50051` is what will be used by MAVSDK-Python (see below).

Then you have to tell python to connect to it, with:

```python
drone = System(mavsdk_server_address="localhost", port=50051)
await drone.connect()
```

DoppleGangster, 2021-01-25:
Making the changes in the code and starting the server manually from my local python build still ends in the same result:

> Waiting for mavsdk_server to be ready...
Connected to mavsdk_server!
Waiting for drone to connect...
Drone discovered with UUID: 0
Traceback (most recent call last):
  File "mission.py", line 91, in <module>
    loop.run_until_complete(run())
  File "/usr/lib/python3.8/asyncio/base_events.py", line 616, in run_until_complete
    return future.result()
  File "mission.py", line 31, in run
    await drone.action.do_orbit(5.0, float('nan'), ORBIT_YAW_BEHAVIOUR, float('nan'), float('nan'), float('nan'))
  File "/home/drone1/Downloads/MAVSDK-Python/mavsdk/action.py", line 522, in do_orbit
    response = await self._stub.DoOrbit(request)
  File "/home/drone1/Downloads/MAVSDK-Python/venv/lib/python3.8/site-packages/aiogrpc/channel.py", line 40, in __call__
    return await fut
grpc._channel._MultiThreadedRendezvous: <_MultiThreadedRendezvous of RPC that terminated with:
	status = StatusCode.UNIMPLEMENTED
	details = ""
	debug_error_string = "{"created":"@1611540952.740632776","description":"Error received from peer ipv6:[::1]:50051","file":"src/core/lib/surface/call.cc","file_line":1062,"grpc_message":"","grpc_status":12}"

JonasVautherin, 2021-01-25:
But you updated the proto submodule in mavsdk, ran `generate_from_protos.sh`, and then compiled `mavsdk_server` from that, didn't you? If the proto submodule contains the `do_orbit` definition, then it should be generated.

Does action_service_impl.h contain `do_orbit`?

DoppleGangster, 2021-01-25:
It does:

`grpc::Status DoOrbit(
        grpc::ServerContext* /* context */,
        const rpc::action::DoOrbitRequest* request,
        rpc::action::DoOrbitResponse* response) override
    {
        if (request == nullptr) {
            LogWarn() << "DoOrbit sent with a null request! Ignoring...";
            return grpc::Status::OK;
        }
            
        
            
        
            
        
            
        
            
        
            
        
        auto result = _action.do_orbit(request->radius_m(), request->velocity_ms(), request->yaw_behavior(), request->latitude_x(), request->longitude_y(), request->altitude_z());
        

        
        if (response != nullptr) {
            fillResponseWithResult(response, result);
        }
        

        return grpc::Status::OK;
    }`

JonasVautherin, 2021-01-25:
Right, so it should be there. Can you show me how you start `mavsdk_server` and how you connect from python?

DoppleGangster, 2021-01-25:
Sure, starting from inside my MAVSDK-Python folder:
Terminal 1:
`./mavsdk/bin/mavsdk_server -p 50051 udp://:14540
`
Terminal2:
'
python3 -m venv venv
source venv/bin/activate
python3 ../mission.py
`

JonasVautherin, 2021-01-25:
> Sure, starting from inside my MAVSDK-Python folder:

That's the issue. You should start the one you just built, which is in your MAVSDK folder, in `build/default/src/mavsdk_server/src/`. The one in your MAVSDK-Python folder is not the one you built, so it does not have your changes :wink:.

DoppleGangster, 2021-01-26:
Hell yeah. Thanks for keeping the help coming. 

One thing to note on the build for the MAVSDK with mav_server. I commented out the following lines in the CMakeLists.txt

`#if (BUILD_BACKEND)
#    message(FATAL_ERROR "The argument BUILD_BACKEND has been replaced by #BUILD_MAVSDK_SERVER. To build mavsdk_server, use -DBUILD_MAVSDK_SERVER=ON.")
#endif()
`
It threw the error even when I replaced the parameter:
`-DBUILD_BACKEND=ON
`
With:
`-DBUILD_MAVSDK_SERVER=ON.
`
Then after that the MAVSDK folder existed but the actual mavsdk_server was missing. I had to run the make command in the mavsdk_server folder to get the file. 

Thanks again for keeping the help coming. It is greatly appreciated. Should I upload the addition under my username for contribution? First time working on an open source project so not sure what the procedure is.

JonasVautherin, 2021-01-26:
About your error, `BUILD_BACKEND` has been replaced by `BUILD_MAVSDK_SERVER`. Maybe you had some remains of that old time, but a clean build should solve it.

About the do_orbit contribution, if that works on your side, of course please share it! It will go like this:

1. Open a PR with your changes in `action.proto` in MAVSDK-Proto
2. Once this is merged, update the proto submodule in your MAVSDK fork (to point to latest `main`, which will contain your changes from the PR in 1), and create a PR with your changes in MAVSDK
3. Optionally, when the MAVSDK PR is merged, you can open a PR in MAVSDK-Python with the proto submodule updated as well

Does that make sense? So the next step is a PR in MAVSDK-Proto, with your changes in `action.proto` :+1:.

DoppleGangster, 2021-01-26:
I do not appear to have permission to create a pull request with MAVSDK-Proto. I'm assuming this is controlled on your side?

JonasVautherin, 2021-01-26:
You have to fork the repo, so that it is under your control in `github.com/dopplegangster/mavsdk-proto`. You push your branch there, and from there you will be able to create a PR against `github.com/mavlink/mavsdk-proto` :+1:. Let me know how it goes :+1:!

DoppleGangster, 2021-01-26:
Got it thanks. The PR is in.

DoppleGangster, 2021-01-29:
I'm trying to get the changes made with the enum to compile and I'm running into an issue with the auto-generated file action_service_impl.h

It seems that the translations do not generate the correct names for the enum as it adds on "ORBIT_YAW_BEHAVIOUR_" in from of certain parts of the enum. If I manually correct the issue everything generates correctly. Did I do something incorrectly with my proto file?

Below if the troubled code:

`static rpc::action::OrbitYawBehavior translateToRpcOrbitYawBehavior(const mavsdk::Action::OrbitYawBehavior& orbit_yaw_behavior)
    {
        switch (orbit_yaw_behavior) {
            default:
                LogErr() << "Unknown orbit_yaw_behavior enum value: " << static_cast<int>(orbit_yaw_behavior);
            // FALLTHROUGH
            case mavsdk::Action::OrbitYawBehavior::ORBIT_YAW_BEHAVIOUR_HOLD_FRONT_TO_CIRCLE_CENTER:
                return rpc::action::ORBIT_YAW_BEHAVIOUR_HOLD_FRONT_TO_CIRCLE_CENTER;
            case mavsdk::Action::OrbitYawBehavior::ORBIT_YAW_BEHAVIOUR_HOLD_INITIAL_HEADING:
                return rpc::action::ORBIT_YAW_BEHAVIOUR_HOLD_INITIAL_HEADING;
            case mavsdk::Action::OrbitYawBehavior::ORBIT_YAW_BEHAVIOUR_UNCONTROLLED:
                return rpc::action::ORBIT_YAW_BEHAVIOUR_UNCONTROLLED;
            case mavsdk::Action::OrbitYawBehavior::ORBIT_YAW_BEHAVIOUR_HOLD_FRONT_TANGENT_TO_CIRCLE:
                return rpc::action::ORBIT_YAW_BEHAVIOUR_HOLD_FRONT_TANGENT_TO_CIRCLE;
            case mavsdk::Action::OrbitYawBehavior::ORBIT_YAW_BEHAVIOUR_RC_CONTROLLED:
                return rpc::action::ORBIT_YAW_BEHAVIOUR_RC_CONTROLLED;
        }
    }

    static mavsdk::Action::OrbitYawBehavior translateFromRpcOrbitYawBehavior(const rpc::action::OrbitYawBehavior orbit_yaw_behavior)
    {
        switch (orbit_yaw_behavior) {
            default:
                LogErr() << "Unknown orbit_yaw_behavior enum value: " << static_cast<int>(orbit_yaw_behavior);
            // FALLTHROUGH
            case rpc::action::ORBIT_YAW_BEHAVIOUR_HOLD_FRONT_TO_CIRCLE_CENTER:
                return mavsdk::Action::OrbitYawBehavior::ORBIT_YAW_BEHAVIOUR_HOLD_FRONT_TO_CIRCLE_CENTER;
            case rpc::action::ORBIT_YAW_BEHAVIOUR_HOLD_INITIAL_HEADING:
                return mavsdk::Action::OrbitYawBehavior::ORBIT_YAW_BEHAVIOUR_HOLD_INITIAL_HEADING;
            case rpc::action::ORBIT_YAW_BEHAVIOUR_UNCONTROLLED:
                return mavsdk::Action::OrbitYawBehavior::ORBIT_YAW_BEHAVIOUR_UNCONTROLLED;
            case rpc::action::ORBIT_YAW_BEHAVIOUR_HOLD_FRONT_TANGENT_TO_CIRCLE:
                return mavsdk::Action::OrbitYawBehavior::ORBIT_YAW_BEHAVIOUR_HOLD_FRONT_TANGENT_TO_CIRCLE;
            case rpc::action::ORBIT_YAW_BEHAVIOUR_RC_CONTROLLED:
                return mavsdk::Action::OrbitYawBehavior::ORBIT_YAW_BEHAVIOUR_RC_CONTROLLED;
        }
    }
`

Thanks again.

JonasVautherin, 2021-01-29:
Oh! There is a typo :see_no_evil:. The enum is called `OrbitYawBehavior`, but the values are spelled `BEHAVIOUR` (with a "U").

That will require a new PR in MAVSDK-Proto, I'm afraid :see_no_evil:.

DoppleGangster, 2021-01-29:
😌 PR is in. Thanks for looking into it.

DoppleGangster, 2021-02-01:
Thanks again for your help!

closed 2021-02-01 by DoppleGangster

aquaMatt900, 2021-02-09:
This is for anyone's reference coming in looking for answers to the error:
The argument BUILD_BACKEND has been replaced by #BUILD_MAVSDK_SERVER. To build mavsdk_server, use -DBUILD_MAVSDK_SERVER=ON.

Just set DBUILD_BACKEND=OFF the next time you call cmake:
sudo cmake -DBUILD_BACKEND=OFF -DBUILD_MAVSDK_SERVER=ON ...