mavlink/MAVSDK-Python #298: How to rotate drone during mission

coffeehacker1, 2021-01-13:
I am trying to create a mission where the drone flies to a waypoint, rotates a specific amount in order to fully face what it is photographing, and then takes a photo. I cannot figure out how to add "rotate 90 degrees" to the mission once the drone arrives at the waypoint.

JonasVautherin, 2021-01-13:
Doesn't it work if you set the following for this particular waypoint?

* The [fly through](https://github.com/mavlink/MAVSDK-Python/blob/main/mavsdk/mission.py#L32-L33) to `False` (such that it stops at the waypoint)
* The [gimbal_yaw](https://github.com/mavlink/MAVSDK-Python/blob/main/mavsdk/mission.py#L38-L39) to the desired yaw (I think that will be a bearing relative to north, but to be tested)
* The [camera_action](https://github.com/mavlink/MAVSDK-Python/blob/main/mavsdk/mission.py#L41-L42) to [TAKE_PHOTO](https://github.com/mavlink/MAVSDK-Python/blob/main/mavsdk/mission.py#L63-L64)

coffeehacker1, 2021-01-13:
Well I admit that I am right now running the simulation, but the heading value does not change irrespective of what value I specify for the gimbal_yaw. In the two items below, I specify that one is to face North and the other is to face South; neither occurs.

    mission_items.append(MissionItem(43.800859,
                                     -79.659737,
                                     25,
                                     10,
                                     False,
                                     float('nan'),
                                     0.0,
                                     MissionItem.CameraAction.TAKE_PHOTO,
                                     5.0,
                                     float('nan')))
    mission_items.append(MissionItem(43.800859,
                                     -79.659461,
                                     25,
                                     10,
                                     False,
                                     float('nan'),
                                     180.0,
                                     MissionItem.CameraAction.TAKE_PHOTO,
                                     5.0,
                                     float('nan')))

I am willing to believe this is my user error, but I don't see where the error is.

coffeehacker1, 2021-01-13:
Oh, I should point out that I am starting at   {  "latitude": "43.800814", "longitude": "-79.659829" }

JonasVautherin, 2021-01-14:
I am not completely convinced that this works in SITL... Do you have a way to confirm that using QGroundControl, i.e. by creating a mission that does exactly that, but from QGC, and check if it works in SITL? Would be helpful :+1:

coffeehacker1, 2021-01-14:
Do you mean to monitor the mission in QGC, or actually create the mission in QGC?

coffeehacker1, 2021-01-14:
Sorry; I thought that was a cancel button. I will recreate the mission in QGC and rerun to see what happens.

coffeehacker1, 2021-01-14:
I recreated the mission in QGC and saw no difference. I have copied the plan below.

{
    "fileType": "Plan",
    "geoFence": {
        "circles": [
        ],
        "polygons": [
        ],
        "version": 2
    },
    "groundStation": "QGroundControl",
    "mission": {
        "cruiseSpeed": 15,
        "firmwareType": 12,
        "hoverSpeed": 5,
        "items": [
            {
                "AMSLAltAboveTerrain": null,
                "Altitude": 20,
                "AltitudeMode": 1,
                "autoContinue": true,
                "command": 22,
                "doJumpId": 1,
                "frame": 3,
                "params": [
                    15,
                    0,
                    0,
                    null,
                    43.8011742,
                    -79.6602173,
                    20
                ],
                "type": "SimpleItem"
            },
            {
                "AMSLAltAboveTerrain": null,
                "Altitude": 20,
                "AltitudeMode": 1,
                "autoContinue": true,
                "command": 16,
                "doJumpId": 2,
                "frame": 3,
                "params": [
                    5,
                    0,
                    0,
                    null,
                    43.8009109,
                    -79.6597672,
                    20
                ],
                "type": "SimpleItem"
            },
            {
                "autoContinue": true,
                "command": 530,
                "doJumpId": 3,
                "frame": 2,
                "params": [
                    0,
                    0,
                    null,
                    null,
                    null,
                    null,
                    null
                ],
                "type": "SimpleItem"
[... 241 lines cut ...]
                    0,
                    0,
                    0,
                    0,
                    0,
                    0
                ],
                "type": "SimpleItem"
            }
        ],
        "plannedHomePosition": [
            43.8011741,
            -79.660219,
            206
        ],
        "vehicleType": 2,
        "version": 2
    },
    "rallyPoints": {
        "points": [
        ],
        "version": 2
    },
    "version": 1
}

coffeehacker1, 2021-01-14:
But I have to admit I am puzzled. Why do I specify the camera gimbal yaw when it is the drone itself that I am trying to rotate? Thanks for the clarification.

JonasVautherin, 2021-01-14:
> I recreated the mission in QGC and saw no difference.

This suggests that the feature does not work in SITL, is that correct? In that case, I'm afraid you will have to try on a real drone...

> But I have to admit I am puzzled. Why do I specify the camera gimbal yaw when it is the drone itself that I am trying to rotate?

Good question :+1:. But are you sure you are trying to rotate the drone? You said that you wanted to face what needs to be photographed. Hence you want the camera to move, not the drone, right? :slightly_smiling_face:

The way it is done will depend on the drone. Some drones can rotate the camera without rotating the vehicle's body, and they will do that (e.g. the Yuneec H520). Some drones can only change the pitch of the camera, and therefore the drone itself will have to rotate in order for the camera to rotate. But both are fine. What you want is for the camera to be in the right position at that waypoint.

Does that make sense?

coffeehacker1, 2021-01-14:
It does if the only reason I want to rotate the drone is to manipulate the camera (which it is in this case). But I can imagine an operator wishing to make that decision herself; a drone might have an excellent camera but an unreliable gimbal, or the camera is sometimes partially blocked at certain angles, so to eliminate either potential she always rotates the drone to face head-on. 

Nonetheless, I will close the question. I will reopen if my hardware drone fails to honor the gimbal yaw setting.

closed 2021-01-14 by coffeehacker1

JonasVautherin, 2021-01-14:
Well then that's a configuration of the drone. The protocol just says "point the camera there and take a picture", but then the drone can do whatever it is configured for. Either rotate the camera, or rotate the body, or rotate the camera a bit and the body a bit. It's not MAVSDK's problem, as long as the camera points to the correct direction for the picture :blush:.

julianoes, 2021-01-23:
Related to https://github.com/mavlink/MAVSDK/issues/909.