mavlink/MAVSDK-Python #300: Want to use Mavsdk but stucks with asyncio

a-altug, 2021-01-16:
Hi,

I want to land by image processing.
I was able to do this using Dronekit with using threading both for control and image process before, but I want to use Mavsdk.

I have one image processing class, one drone control (including async / await, Mavsdk) class and main script.

There is a while loop in the Main class where the process is executed.
There is a function in the image processing class that I call from the while loop in the main script .
There are async methods in the drone class that I call for different situations.

I want to run all of it as Dronekit but stuck on asyncio things which included by Mavsdk.

How can I run all of this together in main script and parallel?

Main:
   ```
initilize ImageProcess things
initilize Drone things

def method ():
   while:
      calling Image.Process

      if something:
          call Drone methods drive with image_things
      if something:
          call Drone methods set set_points

     imshow image

if __name__ == "__main__":

    loop = asyncio.get_event_loop()
    loop.run_until_complete(method())
```

Image:
```
   Process ():
       start
       calculate & assign self.
       end
```

Drone:
```
   check things

    async run():
       if "takeoff":
            async for i in self.drone.telemetry.position():
               if i== z:
               if i== z:
               await self.drone.offboard............

       elif "toward_to":
             check location
             await self.drone.offboard.set_position_ned()
      elif .... :
             .........disarm....
```

As an advice, an example that includes such sync and async things like this situation is very helpfull.

JonasVautherin, 2021-01-16:
Have you seen https://auterion.com/getting-started-with-mavsdk-python/ already? :blush:

a-altug, 2021-01-16:
Thank You for comment.
Yeah I did it already and also run all examples too. 😊

JonasVautherin, 2021-01-16:
Right. So what is not working? Your pseudo-code above seems reasonable... do you have actual code that is not working and that you could share?

a-altug, 2021-01-17:
yeah here it is:
Drone:
```

async def run(self):

        if self.phase == "TAKEOFF":
            async for i in self.drone.telemetry.position():
                if i.relative_altitude_m < 1:
                    self.north_spd, self.east_spd, self.down_spd, self.yaw_spd = 0.0, 0.0, -5.0, 0.0
                elif 1 <= i.relative_altitude_m < 10:
                    self.north_spd, self.east_spd, self.down_spd, self.yaw_spd = 25.0, 0.0, -2.0, 0.0
              ...
             ....
                await self.drone.offboard.set_velocity_body(
                    offboard.VelocityBodyYawspeed(self.north_spd, self.east_spd, self.down_spd, self.yaw_spd))

```
when run main as above the second conditional never works. always run as above 1 meter and also cv2.imshow(... )  statement never runs in main.... 
Thank you..

JonasVautherin, 2021-01-18:
One thing is that `async for ...` will run as an infinite loop. So you should run that in a separate task (so that it runs in parallel), and have it save the altitude in some global state somewhere. But that should not prevent your `if ... else` from working.

So you're saying that it loops and always goes into the `i.relative_altitude_m < 1`, right? Does the drone actually go above 1m in the simulator?

a-altug, 2021-01-19:
Yes drone continuously go above 1 meter but its speed goes 2 as in elif statement.
Then the "async for" loop is broken because the other self.phase is activated when code reaches a condition.

To being simple could you guide me about main function?
How should I call asynchronous drone.run() method (written with async and await) and synchronous image.process () method in main script. Which asyncio method is fits my problem to run both at same time or like multitasking. Because I need both classes to run at same time with main.

Previously when I used dronekit, I was making start a thread(which activate run() in drone class) before the main while loop(in main script) to push mavlink messages and was calling image processing method in main() loop. so completing of main loop was taking some time. In accordance with visual and coordinate statements, I was assingning drone's self variables so I was driving as I expected.

JonasVautherin, 2021-01-19:
Just for the context: is it correct that you are starting a new script from scratch (as opposed to adding MAVSDK to an existing script)?

JonasVautherin, 2021-01-19:
Ok. Let's start without OpenCV first. Can you write (and share here) code that listens to the position updates and saves them in a global state? Something like `observe_position` that would save the value in a global variable, for instance.

And then in your `async def run(self)`, you run a `while True` loop that prints the position saved in the state and calls `await asyncio.sleep(0)` (meaning that it won't wait, but it will release control so that another coroutine can run, as explained in my blog post above).

Once this works, we can see about adding OpenCV calls in that. How does that sound?

a-altug, 2021-01-21:
Thank you and sorry for late response.
I tried in this way but will try again with asyncio.sleep(0) on weekend.
For now exams take my time.
Will be informed with results.

a-altug, 2021-01-23:
Hi, there it is:

```
import asyncio
from mavsdk import System, telemetry, offboard

class Drone:
    def __init__(self):
        self.drone = None
        self.pos_alt = None

    async def connect_uav(self):
        """Main function to connect to the drone"""
        # Connect to the Simulation
        self.drone = System()
        await self.drone.connect(system_address="udp://:14540")

        # This waits till a mavlink based drone is connected
        async for state in self.drone.core.connection_state():
            if state.is_connected:
                print(f"-- Connected to drone with UUID: {state.uuid}")
                # break
                return
            else:
                print("Waiting for connection.")
        await asyncio.sleep(0)

    async def arm(self):
        async for gps_info in self.drone.telemetry.gps_info():
            if gps_info.fix_type == telemetry.FixType.FIX_3D:
                print(f"-- Got 3D fix")
                break

            else:
                print("Waiting for GPS 3D fix")

        async for is_armed in self.drone.telemetry.armed():
            if is_armed:
                print("Drone is armed!")
                break
            else:
                print("-- Arming")
                await self.drone.action.arm()
        await asyncio.sleep(0.001)

        print("-- Setting initial setpoint")
        await self.drone.offboard.set_attitude(offboard.Attitude(0.0, 0.0, 0.0, 0.0))
        return

    async def mode_offboard(self):
        try:
            print("-- Starting offboard mode")
            await self.drone.offboard.start()
        except offboard.OffboardError as error:
            print(f"Starting offboard mode failed with error code: \
            {error._result.result}")

            print("-- Disarming")
            await self.drone.action.disarm()
            # break
            return

    async def observe_position(self):

        async for i in self.drone.telemetry.position():
            self.pos_alt = i.relative_altitude_m
            await asyncio.sleep(0)
            #return

    async def run(self):
        while True:
            #await self.observe_position()
            print(self.pos_alt)
            await asyncio.sleep(0)

if __name__ == "__main__":
    #start_mission()
    drone = Drone()
    loop = asyncio.get_event_loop()
    
    loop.run_until_complete(drone.connect_uav())
    loop.run_until_complete(drone.arm())
    loop.run_until_complete(drone.mode_offboard())

    asyncio.ensure_future(drone.observe_position())

    loop = asyncio.get_event_loop()
    loop.run_until_complete(drone.run())
```

JonasVautherin, 2021-01-27:
```python
    loop.run_until_complete(drone.connect_uav())
    loop.run_until_complete(drone.arm())
    loop.run_until_complete(drone.mode_offboard())

    asyncio.ensure_future(drone.observe_position())
```

This is weird. Look at how the [mission example](https://github.com/mavlink/MAVSDK-Python/blob/main/examples/mission.py) does it: it runs only `loop.run_until_complete(run())` in the main, and `run()` itself starts the parallel tasks (here `observe_position()`). Then your other routines should just be called one after the other in `run()`, something like:

```python
await drone.connect_uav()
await drone.arm()
await drone.mode_offboard()
```

a-altug, 2021-01-28:
Sorry for misunderstanding.

Yeah they could be call in `run()` method right before `while` starts and I also call in my `main()` like you said. But 

   `And then in your async def run(self), you run a while True loop that prints the position saved in the state and calls await asyncio.sleep(0)`

part is changed it. And also I try to make `class Drone:` as fits in my case.

Then, for static issues and for one flowing scripts it is exactly usable that mission example's done. But for my case, which includes dynamically changeable issues, things are weird. Tasks are opening and closing dependently.

But thank you for your assistant because I got how to activate tasks together which dependent different status separately in separate times. And that was my point. :)

JonasVautherin, 2021-01-28:
Ok, glad to hear that it helped!

closed 2021-01-28 by JonasVautherin