mavlink/MAVSDK-Python #301: Multiple telemetry subsription breaks the telemetry.attitude_euler()

Troyhy, 2021-01-29:
I'm working with Holy Bro Durandal connected to embedded Arm computer with a serial connector to Telem1.

Problem is that telemetry.attitude_euler() will stop working if I subscribe to more than 6 telemetry streams. It is easy to see how it breaks from log pics below:
6 streams subscribed
[image]

7 streams subscribed
[image]

The most noticeable thing is the reduced output of Euler angle events, and that the angle is fixed near the initial angle. By "near" I mean that minor change to angle fraction is there, but it seems to be "random"

I did not found any particular stream that would cause this, my experience is that the number of subscriptions will break it.

Here is silmplifield test script

``` python
#!/usr/bin/env python3

import asyncio

from mavsdk import System

async def run():
    # Init the drone

    # mavsdk_server is started separately with command  to get the log output
    # mavsdk_server -p 50050 serial:///dev/ttyTHS1:115200

    drone = System(mavsdk_server_address="localhost", port=50050)  #
    await drone.connect(system_address="serial:///dev/ttyTHS1:115200")
    # system_address not used when mavsdk_address is set

    # Start the tasks
    # If more than 6 task are created euler angle stops "working", showing
    # only initial angle with minor changes that does not reflect the real angle
    asyncio.ensure_future(print_euler(drone))
    asyncio.ensure_future(print_battery(drone))
    asyncio.ensure_future(print_gps_info(drone))
    asyncio.ensure_future(print_in_air(drone))
    # asyncio.ensure_future(print_position(drone))
    asyncio.ensure_future(print_flight_mode(drone))
    asyncio.ensure_future(print_status_text(drone))
    # asyncio.ensure_future(print_health(drone))

async def print_battery(drone):
    async for battery in drone.telemetry.battery():
        print(f"Battery: {battery.remaining_percent}")

async def print_gps_info(drone):
    async for gps_info in drone.telemetry.gps_info():
        print(f"GPS info: {gps_info}")

async def print_in_air(drone):
    async for in_air in drone.telemetry.in_air():
        print(f"In air: {in_air}")

async def print_position(drone):
    async for position in drone.telemetry.position():
        print(position)

async def print_euler(drone):
    await drone.telemetry.set_rate_attitude(1)
    async for euler in drone.telemetry.attitude_euler():
        print(euler)

async def print_flight_mode(drone):
    async for flight_mode in drone.telemetry.flight_mode():
        print(flight_mode)

async def print_status_text(drone):
    async for status_text in drone.telemetry.status_text():
        print(status_text)

async def print_health(drone):
    async for health in drone.telemetry.health():
        print(health)

if __name__ == "__main__":
    # Start the main function
    asyncio.ensure_future(run())

    # Runs the event loop until the program is canceled with e.g. CTRL-C
    asyncio.get_event_loop().run_forever()
```

Troyhy, 2021-01-29:
I'm using
mavsdk==0.15.0
python 3.8.6

Safranek42, 2021-01-29:
I have the same problem but for me it happens at 8 streams, 7 it is still working fine. From the 8th stream the Euler angles are frozen with small changes in the last digits. And for me, angular velocity values behaves the same, too.

My setup is Durandal FC, NVidia Jetson Nano companion computer and serial connection between them.

JonasVautherin, 2021-01-29:
Thanks for the report. It's funny that it only impacts the euler angles :thinking:.

Would anybody feel like trying to reproduce this in C++? So that we would know if that needs to be debugged in Python or in C++ (MAVSDK-Python calls MAVSDK-C++) :see_no_evil:

Troyhy, 2021-02-01:
@Safranek42 mentioned that:

> And for me, angular velocity values behaves the same, too.

MuhammadBilal1, 2021-12-10:
@JonasVautherin @julianoes 
I am also getting the issue of **values freezing** when I try to use more than 5-6 subscriber methods, here is the list of methods I am using at the sametime:

telemetry.gps_info()
telemetry.battery()
telemetry.health()
telemetry.health_all_ok()
telemetry.status_text()
telemetry.gps_info()
telemetry.raw_gps()
telemetry.attitude_euler()
telemetry.velocity_ned()
telemetry.position()
telemetry.flight_mode()
telemetry.in_air()
telemetry.armed()
telemetry.landed_state()

If I decrease the number of subscribers to 4, then the values are coming fine and not freezing.