mavlink/MAVSDK-Python #303: TypeError: 'EnumTypeWrapper' object is not callable when trying to play a tune

hashimK, 2021-02-07:
Hi,
I'm trying to play a tone on my system but I'm getting this error (TypeError: 'EnumTypeWrapper').  As per the TuneDescription and SongElement documentation, I've made the below code. Can anybody let me know why the below code isn't working.

```
#!/usr/bin/env python3

import asyncio
from mavsdk import System
from mavsdk.tune import (TuneDescription,SongElement,TuneError)
import traceback

async def run():

    drone = System(mavsdk_server_address="localhost", port=50041)
    await drone.connect()

    print("Waiting for drone to connect...")
    async for state in drone.core.connection_state():
        if state.is_connected:
            print(f"Drone discovered with UUID: {state.uuid}")
            break

    id_info = await drone.info.get_identification()
    print(id_info)
    await asyncio.sleep(1)

    song_elem = []
    song_elem.append(SongElement.NOTE_G)
    try:
        tone1 = TuneDescription(song_elem,100)
        await drone.tune.play_tune(tone1)
    except Exception:
        print(traceback.print_exc())

    print("Tone played")

if __name__ == "__main__":
    loop = asyncio.get_event_loop()
    loop.run_until_complete(run())
```

Output:
```
Waiting for mavsdk_server to be ready...
Connected to mavsdk_server!
Waiting for drone to connect...
Drone discovered with UUID: 0
Identification: [hardware_uid: 000200000000353236353038510a001c002c]
Traceback (most recent call last):
  File "play_tone.py", line 28, in run
    await drone.tune.play_tune(tone1)
  File "/home/aifi/.local/lib/python3.8/site-packages/mavsdk/tune.py", line 460, in play_tune
    tune_description.translate_to_rpc(request.tune_description)
  File "/home/aifi/.local/lib/python3.8/site-packages/mavsdk/tune.py", line 261, in translate_to_rpc
    rpc_elem = tune_pb2.SongElement()
TypeError: 'EnumTypeWrapper' object is not callable
None
Tone played
```

JonasVautherin, 2021-02-07:
Sounds like a bug [here](https://github.com/mavlink/MAVSDK-Python/blob/main/mavsdk/tune.py#L261). Feel free to have a look if you can, otherwise I'll do it when I have time :blush:.

hashimK, 2021-02-07:
It would be very helpful if you can give some direction/solution. Been wrecking my head over this for quite some time now.
Thanks anyways for letting me know that its a bug at the sdk side.

JonasVautherin, 2021-03-16:
Getting fixed in #324. @hashimK would you have a way to test it? Either with your sample code above or with the example I added in `examples/tune.py` :blush:.

PR #324 (2021-03-18): Fix tune
Repeated enums were not translated correctly.

Took the opportunity to add an example (autumn leaves, again :notes:)

Resolves #303
closed 2021-03-18 by julianoes