mavlink/MAVSDK-Python #304: MAVSDK-Python goto_location function not working

nugustmd, 2021-02-09:
I'm using pixhawk 4 (firmware version: 1.11.3), and I'm running Hexarotor x expectation.

MAVSDK-Pyhon goto_location function does not work on real drone.

But the PX4 for gazebo simulation works fine.

Is there any way to solve it?

julianoes, 2021-02-09:
Do you see any error message, maybe in QGroundControl? Can you try goto in QGroundControl by pressing somewhere in the map?

Do you have a logfile of a flight where you try it?

nugustmd, 2021-02-10:
I don't get any errors in QGroundControl.

Goto works fine in QGroundContol (click anywhere on the map).

But it doesn't work for code written by MAVSDK-Python.

We upload log files and code files.

Thank you. 

[MAVSDK-PYTHON.zip](https://github.com/mavlink/MAVSDK-Python/files/5955221/MAVSDK-PYTHON.zip)

julianoes, 2021-02-10:
@nugustmd would you mind uploading the log to logs.px4.io and then share the link here? That's easiest.

JonasVautherin, 2021-02-10:
FYI: Just tried from MAVSDK-Python using `mavsdk_server` v0.36.0 against SITL, and `goto_location()` works

nugustmd, 2021-02-11:
@julianoes  Log file when running goto in QGroundControl: https://logs.px4.io/plot_app?log=7838d87d-9a2e-46ec-a6e4-08af387005cc

nugustmd, 2021-02-11:
@JonasVautherin  In SITL(GAZEBO) goto_location works fine. 

But even sending commands(goto_location) to the real UAV(Hexarotor x) doesn't work. 

There were no specific errors. 

I am using MAVSDK VERSION 0.35.1.

Thank you.

julianoes, 2021-02-11:
Do you actually get a connection to the real drone? Do you see telemetry information, etc.?

JonasVautherin, 2021-02-11:
> If the time is okay, can you check my MAVSDK_python code once?

If it works with SITL, then the code should be fine. The only thing that could be wrong would be the setup, i.e. the lines that look like:

```python
drone = System(something)
await drone.connect(something)
```

Could you paste here those lines? Would be more convenient than having to download an archive :see_no_evil:

nugustmd, 2021-02-12:
@julianoes  

It uses radio telemetry to connect with real drones.

The mavsdk server is running by writing the following command.

cmd command: ~/MAVSDK-Python/mavsdk/bin/./mavsdk_server -p 50050 serial:///dev/ttyUSB0:57600

If you use it as above, will it be a problem when using MAVSDK?

nugustmd, 2021-02-12:
@JonasVautherin 

MAVSDK server operation is operated by code 1 and server connection is operated by code 2. 

1 = ~/MAVSDK-Python/mavsdk/bin/./mavsdk_server -p 50050 serial:///dev/ttyUSB0:57600
2 =     drone_1 = System(mavsdk_server_address="localhost", port=50050)
          await drone_1.connect()

The simulation runs by connecting via UDP, but the real drone receives serial and makes it into a server before connecting. Will this be a problem?

JonasVautherin, 2021-02-12:
Ok, that seems fine. Now what's the `mavsdk_server` output? Does it say that it detects the drone? Can you copy the log here?

nugustmd, 2021-02-16:
@JonasVautherin 

lhs@lhs-ThinkPad-T480s:~$ sudo ~/MAVSDK-Python/mavsdk/bin/./mavsdk_server -p 50050 serial:///dev/ttyUSB-drone1:57600
[09:43:50|Info ] MAVSDK version: v0.35.1 (mavsdk_impl.cpp:21)
[09:43:50|Debug] New: System ID: 0 Comp ID: 0 (mavsdk_impl.cpp:418)
[09:43:50|Info ] Server started (grpc_server.cpp:45)
[09:43:50|Info ] Server set to listen on 0.0.0.0:50050 (grpc_server.cpp:46)
[09:43:50|Info ] Waiting to discover system on serial:///dev/ttyUSB-drone1:57600... (connection_initiator.h:21)
[09:43:50|Debug] Component Autopilot (1) added. (system_impl.cpp:324)
[09:43:51|Debug] Discovered 1 component(s) (UUID: 3762846554330968882) (system_impl.cpp:481)
[09:43:51|Info ] System discovered (connection_initiator.h:57)
[09:43:51|Debug] New: System ID: 51 Comp ID: 68 (mavsdk_impl.cpp:418)
[09:43:51|Debug] Component Unsupported component (68) added. (system_impl.cpp:324)
[09:43:51|Warn ] command denied (512). (mavlink_commands.cpp:166)
[09:43:52|Debug] Falling back to gimbal protocol v1 (mission_impl.cpp:130)
[09:43:52|Debug] Falling back to Gimbal Version 1 (gimbal_impl.cpp:58)
[09:43:52|Warn ] command denied (512). (mavlink_commands.cpp:166)
[09:44:45|Debug] MAVLink: critical: CAUTION: Avionics Power low:   4.60 Volt (system_impl.cpp:230)
[09:44:46|Debug] MAVLink: warning: Preflight: GPS Horizontal Pos Drift too high (system_impl.cpp:230)
[09:44:46|Debug] MAVLink: info: Armed by external command (system_impl.cpp:230)
[09:44:46|Debug] MAVLink: info: [logger] /fs/microsd/log/2021-02-16/00_44_45.ulg (system_impl.cpp:230)
[09:44:48|Debug] MAVLink: info: Using minimum takeoff altitude: 30.00 m (system_impl.cpp:230)
[09:44:48|Debug] MAVLink: info: Takeoff detected (system_impl.cpp:230)
---------------------------------------------------------------------------------------------------------------

command denied (512). (mavlink_commands.cpp:166) <- Could this warning be a problem? Is there a solution?

julianoes, 2021-02-16:
> command denied (512)

This just means that some messages could not be requested. It should not concern you. What should concern you is:
> [09:44:45|Debug] MAVLink: critical: CAUTION: Avionics Power low:   4.60 Volt (system_impl.cpp:230)
[09:44:46|Debug] MAVLink: warning: Preflight: GPS Horizontal Pos Drift too high (system_impl.cpp:230)

nugustmd, 2021-02-18:
After changing to mavlink-router and changing the connection command as follows, it worked normally. 

`drone_1 = System()
 await drone_1.connect(system_address="udp://127.0.0.1:14540")`

-------------- Add function (before take off)  ---------------------
    print("Fetching amsl altitude at home location....")
    async for terrain_info in drone_1.telemetry.home():
    absolute_altitude = terrain_info.absolute_altitude_m
    break

Thanks for your help.

closed 2021-02-18 by julianoes