mavlink/MAVSDK-Python #305: Unable to remotely connect to jMAVSim

DoppleGangster, 2021-02-09:
I'm attempting to simulate my offboard program by connecting to a remote jMAVSim on another machine. I set MAV_BROADCAST=1 and can see the packets being broadcast on the network. On the remote machine I want to connect with I started the MAVSDK server using:

`./mavsdk_server -p 50051 udp://14540`

In my python program on the remote machine I am attempting to connect to the simulation with the following:

`drone = System(mavsdk_server_address='localhost', port=50051) `
`await drone.connect(system_address="udp://:14540")`

I also tried using 14550 as the port for connecting as it list it when starting:

`INFO  [mavlink] mode: Normal, data rate: 4000000 B/s on udp port 18570 remote port 14550`
`INFO  [mavlink] mode: Onboard, data rate: 4000000 B/s on udp port 14580 remote port 14540`
`INFO  [mavlink] mode: Onboard, data rate: 4000 B/s on udp port 14280 remote port 14030`

I set the address on the host of the simulation to 192.168.142.50. The remote machine was set to 192.168.142.100. Below if what the console reports during the build.

`pxh> INFO  [mavlink] using network interface {19FB4A01-3177-4D76-8744-616F0AA4C99D}, IP: 192.168.142.50`
`INFO  [mavlink] with netmask: 255.255.255.0`
`INFO  [mavlink] and broadcast IP: 192.168.142.255`

I also disabled firewalls on both machine to eliminate any issues with the ports being blocked. Any guidance would be greatly appreciated.

julianoes, 2021-02-09:
Can you see the broadcast heartbeats arriving on the other machine? You can check that with wireshark.

JonasVautherin, 2021-02-09:
> ./mavsdk_server -p 50051 udp://14540

Is that a typo, or are you missing a `:`? It should be `./mavsdk_server -p 50051 udp://:14540`

DoppleGangster, 2021-02-09:
Sorry that was a typo. The mavsdk_server is running as you show. Also using wireshark I am able to see the udp packets from the remote machine. I did notice that the packets show in wireshark with the source port being 18570, not sure if this is a problem or not.

julianoes, 2021-02-10:
> with the source port being 18570

As long as the destination is correct this should work:
```
./mavsdk_server -p 50051 udp://:14550
```

JonasVautherin, 2021-02-10:
Would you mind trying with [px4-gazebo-headless](https://github.com/jonasvautherin/px4-gazebo-headless)?

If MAVSDK is running on the machine with address `192.168.142.100`, and you run this from the machine `192.168.142.50`, it should work:

```
docker run --rm -it jonasvautherin/px4-gazebo-headless:1.11.0 192.168.142.100
```

It will send MAVLink to 192.168.142.100 on port 14540, and to 192.168.142.50 on port 14550.

DoppleGangster, 2021-02-10:
Changing my server to udp://:14550 allowed did the trick.

Just for anyone else who ends up searching this issue:

1. Start the simulation on the host machine
2. Start MAVSDK Server (Offboard Machine): `./mavsdk_server -p 50051 udp://:14550`
3. Start Python Program with the follow connection:
    `drone = System(mavsdk_server_address='localhost', port=50051)`
    `await drone.connect(system_address="udp://:14550")`

Thanks again for the assistance!

closed 2021-02-10 by DoppleGangster

julianoes, 2021-02-10:
Following up here for anyone reading this later. The port 14540 does not work with the option `MAV_BROADCAST` due to:
https://github.com/PX4/PX4-Autopilot/blob/82d6cc3dba3bbc8d6fbd0a3449e7d67560d4a67d/src/modules/mavlink/mavlink_main.cpp#L833
https://github.com/PX4/PX4-Autopilot/blob/82d6cc3dba3bbc8d6fbd0a3449e7d67560d4a67d/ROMFS/px4fmu_common/init.d-posix/rcS#L259

So either the link needs to be set to "not onboard" by removing `-m onboard` or by removing the check in the if clause.