mavlink/MAVSDK-Python #309: Terminating Orbit Command

DoppleGangster, 2021-02-15:
I cannot seem to find a way to terminate the orbit command once started. I'm currently attempting to switch from having an active orbit to manual control:

`manual_control.set_manual_control_input`

The manual control input works correctly prior to giving the orbit command but when switching back after orbiting the drone now accepts the manual commands and still continues to attempt an orbit.

I looked the the mavlink messages and I couldn't find a command that would cause some kind of a drone halt so not sure what the correct action would be for stopping the orbit.

Thanks!

julianoes, 2021-02-16:
Have you tried switching to manual with `manual_control.start_altitude_control()`?

DoppleGangster, 2021-02-16:
Makes sense I attempted the below commands and they do terminate the orbit however each causes an undesirable outcome. 

With the following command:

`manual_control.start_altitude_control()`

End result:

Drone allows for manual commands however immediately after executing the command the drone seems to either shoot vertical or tangential to the orbit.

and with the following command:

`manual_control.start_position_control()`

End result:

`manual_control.set_manual_control_input(0.0, 0.0, 0.0, 0.0)`

Must be given immediately before and not immediately after otherwise the drone with fail safe land. When I do give this command prior to starting position control the drone appears to operate normally briefly. However shortly after flying, the drone stops taking the manual inputs and only responds to the land command.

dayjaby, 2021-02-16:
Do you send the manual control inputs continuously? I think 10Hz is a good rate.

DoppleGangster, 2021-02-16:
10Hz is my target rate at sending the commands currently. The manual commands work normally until I switch out of an orbit. I am using jmavsim and have not attempted on a physical drone so I not sure if perhaps it is an issue with the simulator.

DoppleGangster, 2021-02-16:
It looks like the action that causes the drone to lock up when returning manual control is when the r value is set:

` set_manual_control_input(x, y, z, r)`

So asking the drone to rotate after orbiting leads to this condition.

DoppleGangster, 2021-02-16:
Actually I take that back it happens with x and y and z inputs as well.

julianoes, 2021-02-16:
If you just want to switch to Hold mode you could try `mission->pause_mission()`. That's not super intuitive thought...

DoppleGangster, 2021-02-16:
Nice thanks! Looks like that does the trick.

So when terminating orbit I perform the following commands:

`await drone.mission.pause_mission()`
`await drone.manual_control.set_manual_control_input(0.0, 0.0, 0.0, 0.0)`
`await drone.manual_control.start_position_control()`

closed 2021-02-16 by DoppleGangster

julianoes, 2021-02-17:
@DoppleGangster makes sense. It looks a little awkward but does the trick :).