mavlink/MAVSDK-Python #311: Shell Command not working

nugustmd, 2021-02-22:
I want to send "camera_trigger test" with On the PX4 console command, but it doesn't work.

code: `drone.shell.send("camera_trigger test")`

Is there another way to utilize MAVSDK-Python to operate the Trigger Camera?

JonasVautherin, 2021-02-22:
Are you running against SITL?

Not sure what this does. Isn't it equivalent to `drone.camera.take_photo()`?

nugustmd, 2021-02-23:
@JonasVautherin 

It is actually being used in drones.

The drone.camera.take_photo() command does not work.

Do I need to set the camera separately for the drone.camera.take_photo() command to work?

If the camera_trigger test is performed in QGC after setting the current Trigger GPIO LOW Mode, it works normally.

julianoes, 2021-02-23:
Right, `take_photo()` will likely not work as expected. It's a bit tricky because there are multiple interfaces to trigger photos with various support.

Do you know which message/command QGC uses to test it?

nugustmd, 2021-02-24:
@julianoes 

No I don't know. Tried only take_photo().

The current drone camera settings are shown below.

How do I set it to use the take_photo() function?
[image]

I want to use it as a GPIO trigger interface.

JonasVautherin, 2021-02-24:
> How do I set it to use the take_photo() function?

`take_photo()` is a MAVSDK concept. Internally it sends `MAV_CMD_IMAGE_START_CAPTURE`, see [here](https://github.com/mavlink/MAVSDK/blob/develop/src/plugins/camera/camera_impl.cpp#L210-L222):

```
MavlinkCommandSender::CommandLong CameraImpl::make_command_take_photo(float interval_s, float no_of_photos)
{
    MavlinkCommandSender::CommandLong cmd_take_photo{};

    cmd_take_photo.command = MAV_CMD_IMAGE_START_CAPTURE;
    cmd_take_photo.params.param1 = 0.0f; // Reserved, set to 0
    cmd_take_photo.params.param2 = interval_s;
    cmd_take_photo.params.param3 = no_of_photos;
    cmd_take_photo.params.param4 = float(_capture.sequence++);
    cmd_take_photo.target_component_id = _camera_id + MAV_COMP_ID_CAMERA;

    return cmd_take_photo;
}
```

nugustmd, 2021-02-24:
Are there any commands that can perform Trigger Camera (Camera Test) in MAVSDK-Python?

I want to trigger the camera by sending a specific command, not distance or time.

julianoes, 2021-02-24:
@nugustmd no that's currently not implemented. It's a bit unfortunate as there are several interfaces that could potentially be used.

closed 2021-03-02 by nugustmd