mavlink/MAVSDK-Python #312: LogFile Download: No documentation available / does not work as expected

tommy3001, 2021-02-23:
Hi. 

I am not able to find any information / documentation / examples about log file download using serial / usb connection from  a pixhawk flight controller. 
When I try to do it in this way, I get no response:
  
    import threading
    import serial
    import struct
    import asyncio
    from mavsdk import System
    async def print_flight_mode():
        drone = System()
        await drone.connect(system_address="serial:///dev/ttyUSB0:115200")
        print("Waiting for drone to connect...")
        async for state in drone.core.connection_state():
            if state.is_connected:
                print(f"Drone discovered with UUID: {state.uuid}")
                break
        drone.log_files.download_log_file(5,"/home/tommy/")
    if __name__ == "__main__":
        loop = asyncio.get_event_loop()
        loop.run_until_complete(print_flight_mode())

JonasVautherin, 2021-02-23:
> drone.log_files.download_log_file(5,"/home/tommy/")

This gets you a coroutine and never executes it. You have to do it this way:

```
await drone.log_files.download_log_file(5,"/home/tommy/")
```

If you're interested in the implementation, [here is the code that is being executed](https://github.com/mavlink/MAVSDK/blob/develop/src/plugins/log_files/log_files_impl.cpp#L204). If you're new to asyncio, you can [have a look at this tutorial](https://auterion.com/getting-started-with-mavsdk-python/) :blush:.

Hope this helps!

tommy3001, 2021-02-24:
THX. 

But adding _await_ does not help. 
I get the following issue:

Waiting for mavsdk_server to be ready...
Connected to mavsdk_server!
Traceback (most recent call last):
  File "getLogFiles.py", line 19, in <module>
    loop.run_until_complete(print_flight_mode())
  File "/usr/lib/python3.7/asyncio/base_events.py", line 584, in run_until_complete
    return future.result()
  File "getLogFiles.py", line 14, in print_flight_mode
    await drone.log_files.download_log_file(50,"/home/pi/")
TypeError: object async_generator can't be used in 'await' expression
Exception ignored in: <function System.__del__ at 0xb53a8fa8>
Traceback (most recent call last):
  File "/home/pi/.local/lib/python3.7/site-packages/mavsdk/system.py", line 54, in __del__
  File "/home/pi/.local/lib/python3.7/site-packages/mavsdk/system.py", line 90, in _stop_mavsdk_server
ImportError: sys.meta_path is None, Python is likely shutting down

the other examples are working as expected.

JonasVautherin, 2021-02-24:
> TypeError: object async_generator can't be used in 'await' expression

Oh, it's an async generator, sorry. Because it sends you the progress [see here](https://github.com/mavlink/MAVSDK/blob/develop/src/plugins/log_files/log_files_impl.cpp#L264).

So it has to be called (like the calibration, for instance [here](https://github.com/mavlink/MAVSDK-Python/blob/main/examples/calibration.py#L13-L14)) with something like this:

```
async for progress in drone.log_files.download_log_file(5,"/home/tommy/"):
    print(progress)
print("done")
```

Would you mind trying that? Async generators are explained in my tutorial linked above when talking about the telemetry :+1:.

tommy3001, 2021-02-27:
Sorry,t hat does not work either...

    Waiting for mavsdk_server to be ready...
    Connected to mavsdk_server!
    Waiting for drone...
    Drone discovered with UUID: 0
    Traceback (most recent call last):
      File "./getLogFiles.py", line 18, in <module>
        loop.run_until_complete(run())
      File "/usr/lib/python3.7/asyncio/base_events.py", line 584, in run_until_complete
        return future.result()
      File "./getLogFiles.py", line 12, in run
        async for progress in drone.log_files.download_log_file(5,"/home/tommy/"):
      File "/usr/local/lib/python3.7/dist-packages/mavsdk/log_files.py", line 411, in download_log_file
        raise LogFilesError(result, "download_log_file()", id, path)
    mavsdk.log_files.LogFilesError: INVALID_ARGUMENT: 'Invalid Argument'; origin: download_log_file(); params: (5, '/home/pi/')
    Exception ignored in: <function System.__del__ at 0xb541a660>
    Traceback (most recent call last):
      File "/usr/local/lib/python3.7/dist-packages/mavsdk/system.py", line 54, in __del__
      File "/usr/local/lib/python3.7/dist-packages/mavsdk/system.py", line 90, in _stop_mavsdk_server
    ImportError: sys.meta_path is None, Python is likely shutting down

JonasVautherin, 2021-03-01:
It complains about an "invalid argument", did you have a look at the code [here](https://github.com/mavlink/MAVSDK/blob/develop/src/plugins/log_files/log_files_impl.cpp#L219)?

It seems to me that the log file with id "5" does not exist. Could it be?

tommy3001, 2021-03-01:
No, there are around 40 logfiles on the flight controller available, it is running on a pi.

Assaf-Friedman, 2021-03-02:
Hi,
I struggled with the same issue today. I found out that you have to specify also the file name not just the path.
for example try: 
        async for progress in drone.log_files.download_log_file(5, "/home/tommy/log5.ulg "):
            print(progress)
        print("done")

or:

log_id =5    
        async for progress in drone.log_files.download_log_file( log_id , f '/home/tommy/log{log_id}.ulg ):
            print(progress)
        print("done")`

hope that it helps

JonasVautherin, 2021-03-02:
aha, I did not know that. Maybe the C++ code should write to a default file name if the path is a directory and the default file does not exist? @julianoes an opinion on that?

@Assaf-Friedman: would you be willing to contribute that (in the C++ code)?

JonasVautherin, 2021-03-03:
That's the link I pasted in my comments above: https://github.com/mavlink/MAVSDK/blob/develop/src/plugins/log_files/log_files_impl.cpp#L204-L268

tommy3001, 2021-03-09:
I'm still getting random errors here.
With the px4_sitl jmavsim simulator, the latest firmware and the latest mavsdk python from repo, it occasionally works. 
After a few re-runs of the script, it occasionally downloads the log files. 
On a RASPi with freshly built from source MAVSK and MAVSK-Python, it doesn't work at all.  Even the log_files.get_entries is not working. Battery telemetry is working in both cases. 
Here is the code i used, I changed only the system_address for the pi: 

```python
import asyncio
from mavsdk import System
import random
import string
def get_random_string(length):
    # choose from all lowercase letter
    letters = string.ascii_lowercase
    result_str = ''.join(random.choice(letters) for i in range(length))
    return result_str
    
async def run():
    drone = System()
    await drone.connect(system_address="udp://:14540")
    # Start the tasks
    asyncio.ensure_future(print_battery(drone))
    asyncio.ensure_future(print_logs(drone))
    asyncio.ensure_future(get_log(drone))

async def get_log(drone):
    async for progress in drone.log_files.download_log_file(1,"/tmp/"+get_random_string(6) + ".ulg"):
        print(progress)
    print("done")
    
async def print_battery(drone):
    async for battery in drone.telemetry.battery():
        print(f"Battery: {battery.remaining_percent}")
        
async def print_logs(drone):
    logs = await drone.log_files.get_entries()
    for log in logs: 
    	print(f"Log {logs.index(log)}: {log}")
    
if __name__ == "__main__":
    # Start the main function
    asyncio.ensure_future(run())
    # Runs the event loop until the program is canceled with e.g. CTRL-C
    asyncio.get_event_loop().run_forever()
```
On both systems the logs are available by using Qgroundcontrol.

I am not yet familiar with the MAVSDK CPP async routines in detail.

JonasVautherin, 2021-03-09:
Again, `InvalidArgument` is thrown from [here](https://github.com/mavlink/MAVSDK/blob/develop/src/plugins/log_files/log_files_impl.cpp#L211-L223), which is when the id cannot be found. Are you sure that `1` is valid in your case?

tommy3001, 2021-03-09:
Yes. Log Id 0 ...  8 is also listed in Qgroundcontrol on the drone and local on the simulator. It is working local on the simulator, despite the fact, that i have here to start the script several times in a row on the same simulator state conditions to get a downloaded log file.

tommy3001, 2021-03-09:
Ähm, got the point._entries used by download_log_file_async() in your linked source is filled by get_entries_async(). So, you can only download the logs, when get_entries_async() is completed. I assumed, that download_log_file_async() is sufficient. The docs should be updated, that get_entries_async() with a complete flag is mandatory.  From my point, you can close it.

closed 2021-03-09 by tommy3001

JonasVautherin, 2021-03-10:
Good find! Would you mind opening a PR to update the docs?