mavlink/MAVSDK-Python #313: Getting Controllers Output as Telemetry

mzahana, 2021-02-24:
Hi @julianoes 
Is there a way to get the controllers outputs (e.g. rates controller output of multicopters, output of the attitude controller,  velocity controller, position controller) from PX4 by MAVSDK-Python?
This is useful, for example, for r2eal-time system identification and control tuning.

Thanks.

JonasVautherin, 2021-02-24:
What MAVLink messages are you interested in?

mzahana, 2021-02-25:
@JonasVautherin thanks for your reply.

I am not sure about the MAVLink messages, but I guess the uORB topics are similar to the ones [here](https://github.com/PX4/flight_review/blob/9e36168f55ca9edd2bc9beee4372263bdc88ce44/plot_app/helper.py#L297-L311)
Namely, `actuator_controls_0, actuator_controls_1 ,vehicle_angular_velocity ,  vehicle_attitude , vehicle_attitude_setpoint , vehicle_rates_setpoint , position_setpoint_triplet , vehicle_local_position , vehicle_local_position_setpoint `, but I am not sure if they all have corresponding MAVLink messages and streams.

Thank you.

JonasVautherin, 2021-02-25:
I think you need to check which MAVLink messages you are interested in. MAVSDK exclusively deals with MAVLink messages, so if you don't find the message, you won't be able to add it to MAVSDK.

mzahana, 2021-02-25:
@JonasVautherin thank you very much for your reply.
Do you know where I can find the list of MAVLink messages being streamed from PX4?

JonasVautherin, 2021-02-25:
I would start from here: http://mavlink.io/en/messages/common.html

mzahana, 2021-02-26:
@JonasVautherin 
Here are the MAVLink messages I am interested to receive from PX4.
* `ACTUATOR_CONTROL_TARGET`
* `ATTITUDE_TARGET`
* `POSITION_TARGET_LOCAL_NED`

JonasVautherin, 2021-02-26:
The first one seems to be [here](https://github.com/mavlink/MAVSDK-Python/blob/main/mavsdk/telemetry.py#L3487), but the other two are not implemented as far as I can tell. Open to PRs if you want to contribute them :slightly_smiling_face:. The first one would add the interface into MAVSDK-Proto, a bit like [here](https://github.com/mavlink/MAVSDK-Proto/blob/main/protos/telemetry/telemetry.proto#L50).

mzahana, 2021-02-27:
@JonasVautherin thanks for your reply.
I would be happy to contribute if I know where to implement the new functions. So, if there is documentation ( or someone gives me pointers) on how to proceed, I will give it a try.

Thanks.

JonasVautherin, 2021-03-01:
Did you have a look at MAVSDK-Proto? Also maybe [this](https://www.youtube.com/watch?v=T1orxSyqDzI) can help you understand how features are added.

julianoes, 2021-03-01:
And this: https://mavsdk.mavlink.io/develop/en/contributing/plugins.html#create-a-plugin