mavlink/MAVSDK-Python #315: "mavsdk.action.ActionError: COMMAND_DENIED:" when using takeoff_and_land.py

mommotti, 2021-02-27:
I set up an environment on my Raspberry Pi 4, mavproxy + mavsdk + pixhawk

**The mavproxy command (which I run from the Raspberry, _works as intended_):**
> 
mavproxy.py --master=/dev/ttyACM0 --out=udp:127.0.0.1:14540 --out=udp:127.0.0.1:14550 --out=udp:127.0.0.1:14560 --out=udp:192.168.1.107:14540 --out=udp:192.168.1.107:14550 --out=udp:192.168.1.107:14560 --out=udp:192.168.1.107:57600 --aircraft=ABZERO --daemon &

**The connection to Mission Planner:**
[Mission Planner Screenshot](https://prnt.sc/108gt2d)

**takeoff_and_land.py actual code:**
```
#!/usr/bin/env python3

import asyncio
from mavsdk import System

async def run():

    drone = System()
    await drone.connect(system_address="serial:///dev/ttyACM0:57600")

    print("Waiting for drone to connect...")
    async for state in drone.core.connection_state():
        if state.is_connected:
            print(f"Drone discovered with UUID: {state.uuid}")
            break

    print("Waiting for drone to have a global position estimate...")
    async for health in drone.telemetry.health():
        if health.is_global_position_ok:
            print("Global position estimate ok")
            break

    print("-- Arming")
    await drone.action.arm()

    print("-- Taking off")
    await drone.action.takeoff()

    await asyncio.sleep(5)

    print("-- Landing")
    await drone.action.land()

if __name__ == "__main__":
    loop = asyncio.get_event_loop()
    loop.run_until_complete(run())

```

**The actual error when doing takeoff_and_land.py:**
```
pi@raspberrypi:~/drone $ python3 takeoff_and_land.py
Waiting for mavsdk_server to be ready...
Connected to mavsdk_server!
Waiting for drone to connect...
Drone discovered with UUID: 0
Waiting for drone to have a global position estimate...
Global position estimate ok
-- Arming
-- Taking off
Traceback (most recent call last):
  File "takeoff_and_land.py", line 38, in <module>
    loop.run_until_complete(run())
  File "/usr/lib/python3.7/asyncio/base_events.py", line 584, in run_until_complete
    return future.result()
  File "takeoff_and_land.py", line 28, in run
    await drone.action.takeoff()
  File "/home/pi/.local/lib/python3.7/site-packages/mavsdk/action.py", line 360, in takeoff
    raise ActionError(result, "takeoff()")
mavsdk.action.ActionError: COMMAND_DENIED: 'Command Denied'; origin: takeoff(); params: ()
Exception ignored in: <function System.__del__ at 0xb5387660>
Traceback (most recent call last):
  File "/home/pi/.local/lib/python3.7/site-packages/mavsdk/system.py", line 54, in __del__
  File "/home/pi/.local/lib/python3.7/site-packages/mavsdk/system.py", line 90, in _stop_mavsdk_server
ImportError: sys.meta_path is None, Python is likely shutting down
```
**The mavproxy output when I do takeoff_and_land.py:**
```
ARMED
Arming checks disabled
APM: finished motor test
DISARMED
```
Any help is much appreciated! ( I'm quite noob, I may be missing something! )

mommotti, 2021-03-01:
If you take a closer look, "python3 takeoff_and_land.py" also outputs

> Drone discovered with UUID: 0

Is that a valid UUID?

JonasVautherin, 2021-03-01:
`COMMAND_DENIED` says that PX4 received the command and denied it. Do you have a way to know why PX4 denied it?

julianoes, 2021-03-01:
Duplicate of https://github.com/mavlink/MAVSDK-docs/issues/230.

closed 2021-03-01 by julianoes