mavlink/MAVSDK-Python #316: Higher baud rate then 57600 over TELEM2 port

Assaf-Friedman, 2021-03-02:
Hi,
I am connecting a Raspberry pi serial pins to Pixhawk TELEM2 port,  when I use:
`await drone.connect(system_address="serial:///dev/serial0:57600")` 
everything works well, but if I try to use higher baud rate, first setting SER_TEL2_BAUD to 115200 in the pixhawk for example and then runing:
`await drone.connect(system_address="serial:///dev/serial0:115200")` 
 it it does not work and I get:

`Waiting for mavsdk_server to be ready...
Connected to mavsdk_server!
Waiting for drone to connect...
__anext__ grpc exception
Traceback (most recent call last):
  File "/home/pi/.local/lib/python3.7/site-packages/aiogrpc/utils.py", line 145, in cb
    fut.result()
  File "/usr/lib/python3.7/concurrent/futures/thread.py", line 57, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/home/pi/.local/lib/python3.7/site-packages/aiogrpc/utils.py", line 131, in _next
    return next(self._iterator)
  File "/home/pi/.local/lib/python3.7/site-packages/grpc/_channel.py", line 416, in __next__
    return self._next()
  File "/home/pi/.local/lib/python3.7/site-packages/grpc/_channel.py", line 803, in _next
    raise self
grpc._channel._MultiThreadedRendezvous: <_MultiThreadedRendezvous of RPC that terminated with:
	status = StatusCode.UNAVAILABLE
	details = "Socket closed"
	debug_error_string = "{"created":"@1614693830.411919106","description":"Error received from peer ipv6:[::1]:50051","file":"src/core/lib/surface/call.cc","file_line":1068,"grpc_message":"Socket closed","grpc_status":14}"
>
Traceback (most recent call last):
  File "read_sd.py", line 41, in <module>
    loop.run_until_complete(run())
  File "/usr/lib/python3.7/asyncio/base_events.py", line 584, in run_until_complete
    return future.result()
  File "read_sd.py", line 14, in run
    async for state in drone.core.connection_state():
  File "/home/pi/.local/lib/python3.7/site-packages/mavsdk/core.py", line 207, in connection_state
    async for response in connection_state_stream:
  File "/home/pi/.local/lib/python3.7/site-packages/aiogrpc/utils.py", line 154, in __anext__
    return await asyncio.shield(self._next_future)
  File "/home/pi/.local/lib/python3.7/site-packages/aiogrpc/utils.py", line 145, in cb
    fut.result()
  File "/usr/lib/python3.7/concurrent/futures/thread.py", line 57, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/home/pi/.local/lib/python3.7/site-packages/aiogrpc/utils.py", line 131, in _next
    return next(self._iterator)
  File "/home/pi/.local/lib/python3.7/site-packages/grpc/_channel.py", line 416, in __next__
    return self._next()
  File "/home/pi/.local/lib/python3.7/site-packages/grpc/_channel.py", line 803, in _next
    raise self
grpc._channel._MultiThreadedRendezvous: <_MultiThreadedRendezvous of RPC that terminated with:
	status = StatusCode.UNAVAILABLE
	details = "Socket closed"
	debug_error_string = "{"created":"@1614693830.411919106","description":"Error received from peer ipv6:[::1]:50051","file":"src/core/lib/surface/call.cc","file_line":1068,"grpc_message":"Socket closed","grpc_status":14}"
>
Exception ignored in: <function System.__del__ at 0xb543fc48>
Traceback (most recent call last):
  File "/home/pi/.local/lib/python3.7/site-packages/mavsdk/system.py", line 54, in __del__
  File "/home/pi/.local/lib/python3.7/site-packages/mavsdk/system.py", line 90, in _stop_mavsdk_server
ImportError: sys.meta_path is None, Python is likely shutting down
`

how can I use higher baud rates? 57600 is very slow rate especially when I transfer log files

JonasVautherin, 2021-03-02:
Which RPi is it? The error sounds like either `mavsdk_server` is not running, or it crashed. Do you know how to start it manually and get its log output?

Assaf-Friedman, 2021-03-02:
I use RPI 4 model B.
I Do not know how to start the mavsdk_server manually and get its log output.
However I found out now that if I insert a sleep time like in here:
`   drone = System()
    await drone.connect(system_address="serial:///dev/serial0:1500000")
    print("Waiting for drone to connect...")
    time.sleep(1)
    async for state in drone.core.connection_state():
        if state.is_connected:
            print(f"Drone discovered with UUID: {state.uuid}")
            break`
I am able to communicate with baud rates up to 1500000.
any Idea why? any suggestion how to do it more elegantly without brute force sleep time?

julianoes, 2021-03-03:
Interesting.

You can run the `mavsdk_server` manually: `.local/lib/python3.9/site-packages/mavsdk/bin/mavsdk_server -p 50051 serial:///dev/serial0:1500000` (maybe you need to change the Python version, or maybe your pip installation is somewhere else).

Then you change the script to:
```
drone = System(mavsdk_server_address='localhost')
   await drone.connect(system_address="serial:///dev/serial0:1500000")
   print("Waiting for drone to connect...")
   async for state in drone.core.connection_state():
       if state.is_connected:
           print(f"Drone discovered with UUID: {state.uuid}")
           break
```