mavlink/MAVSDK-Python #319: pip install mavsdk on Mac Big Sur M1: "ERROR: No matching distribution found for mavsdk==0.16.1 (from -r requirements.txt (line 9))"

ajmontoya, 2021-03-06:
I'm attempting to install mavsdk via pip and keep getting "ERROR: No matching distribution found..."
Here's the relevant portions of the output from verbose pip install:

```
Skipping link: none of the wheel's tags match: py3-none-macosx_10_9_x86_64: https://files.pythonhosted.org/packages/c6/bf/a47e093605f33be33b56dccdcdf509a080c1330fed89ba95bdf79d384aa6/mavsdk-0.16.1-py3-none-macosx_10_9_x86_64.whl#sha256=8298b31cdfb86a7699dc3a0cd74f5700fc87447f703c1267026fe122493cc8c3 (from https://pypi.org/simple/mavsdk/) (requires-python:>=3.6)
    Skipping link: none of the wheel's tags match: py3-none-manylinux1_x86_64: https://files.pythonhosted.org/packages/3c/84/1984042247e617e1f01fc31ca9b666c97b1f002310a1055eda054b0ca7c1/mavsdk-0.16.1-py3-none-manylinux1_x86_64.whl#sha256=ab2338a3e80618cd93cb15eb7cf6cf289a4f8eac982cdf13a973e61cdfa52c92 (from https://pypi.org/simple/mavsdk/) (requires-python:>=3.6)
    Skipping link: none of the wheel's tags match: py3-none-manylinux2014_aarch64: https://files.pythonhosted.org/packages/78/4d/eb060ccda4e1f1e83f28e07bdaaf2fedd9f84aa12d1ebb8de5a5cd1a068d/mavsdk-0.16.1-py3-none-manylinux2014_aarch64.whl#sha256=9ff69141057cfd488bfd8b71697a471ba3d1e5af75f25cf5574429aeb28e8762 (from https://pypi.org/simple/mavsdk/) (requires-python:>=3.6)
    Skipping link: none of the wheel's tags match: py3-none-manylinux2014_x86_64: https://files.pythonhosted.org/packages/44/6e/3df56975c624a78c95eaf54fdc3e20a0c09ea9395e14e7d52e62e924930c/mavsdk-0.16.1-py3-none-manylinux2014_x86_64.whl#sha256=c186d2eff3cc19e4b1aa128a6e0e98f6810ea3a5599d69f30c82fbbe361b4b9c (from https://pypi.org/simple/mavsdk/) (requires-python:>=3.6)
    Skipping link: none of the wheel's tags match: py3-none-win32: https://files.pythonhosted.org/packages/f3/46/8a0235f9b3a278f726756356feb47d158601f88492bb45d5e97eebd4bea0/mavsdk-0.16.1-py3-none-win32.whl#sha256=91e45e74889c993cb83a0e6c6075e88faf13de3405dbbcdc0f5f645215c20d43 (from https://pypi.org/simple/mavsdk/) (requires-python:>=3.6)
    Skipping link: none of the wheel's tags match: py3-none-win_amd64: https://files.pythonhosted.org/packages/39/52/3c1fb9b7ca0f7f692588847f2698b474225e3197c689352bdad117c7cb02/mavsdk-0.16.1-py3-none-win_amd64.whl#sha256=f12f8b7a5e952889aeb5500479a3cd40c31eb25093878feb4f6a64f4ba8e1da6 (from https://pypi.org/simple/mavsdk/) (requires-python:>=3.6)
  Given no hashes to check 0 links for project 'mavsdk': discarding no candidates
  ERROR: Could not find a version that satisfies the requirement mavsdk==0.16.1 (from -r requirements.txt (line 9)) (from versions: 0.5.2, 0.7.0, 0.8.0, 0.8.1, 0.8.2, 0.9.0, 0.9.1, 0.10.0, 0.11.0, 0.12.0)
```

OS: MacOS v 11.2.2 Big Sur
Arch: M1 ARM
Python version: 3.8.2
PIP version: 19.2.3
Notes: Running Terminal in Rosetta mode and using a virtual environment via venv

I duplicated Terminal and am running it in Rosetta (x86) mode but doesn't seem to avoid this issue. Not sure if I'm interpreting the output correctly but seems like the only available versions PIP is looking to install is 0.5.2 to 0.12.0 tho current version is 0.16.1

Any help would be greatly appreciated.

julianoes, 2021-03-06:
Thanks for the information. I suppose it's time to start building mavsdk_server for M1 and push it to PyPi like all the other binaries that we already ship for x86_64, armv7, armv8/aarch64, win32, win_amd64, etc.

I think the first step would to have an M1 macOS machine in GitHub action but that doesn't seem very easy, just reading through:
https://github.com/actions/virtual-environments/issues/2187

ajmontoya, 2021-03-06:
Thanks for the quick reply! I was able to install ver 0.12.0 but haven't had a chance to run it yet. Probably will do so later this weekend.

JonasVautherin, 2021-03-07:
Does pip support the M1 already?

ben-xD, 2022-05-19:
There is no package for M1 macs (darwin arm64) available on PyPI https://pypi.org/project/mavsdk/#files, even for the latest release this month (1.3.0).

That doesn't mean you can compile it yourself and try though. I personally didn't do that. I ran it in a Python docker image - which implicitly uses `linux arm64` which is available on PyPI.

JonasVautherin, 2022-05-19:
One issue we have is that the Github CI does not provide any M1 workers, so... :sweat_smile:

julianoes, 2022-05-19:
Looks like this has been added (or is about to?): https://github.com/github/roadmap/issues/507

JonasVautherin, 2022-05-19:
Oh it's really really new then. I had checked very recently and it was not there :grin:. Well it would for sure be nice to try it!

JonasVautherin, 2022-05-23:
I tried running one of those new workers, but they never start: https://github.com/mavlink/MAVSDK/pull/1783

Either I'm doing it wrong, or it's not ready for production yet

Sitin, 2022-06-17:
Hi! What is the status of this issue?

@ben-xD, I've tested local build on my M1-Pro. Everything looks fine for now.

JonasVautherin, 2022-06-17:
I think it's not there: https://github.com/actions/virtual-environments/issues/2187

Sitin, 2022-07-16:
Do we really need to build this on Mac? Can't we use a wheel for ARM64?

Sitin, 2022-07-16:
@JonasVautherin, what if I donate an M1 cloud instance until this issue will be resolved? Say, provide an SSH key pair which you can store in build environment.

julianoes, 2022-07-18:
However, this tutorial would suggest that it is possible:
https://betterprogramming.pub/run-github-actions-self-hosted-macos-runners-on-apple-m1-mac-b559acd6d783

julianoes, 2024-06-02:
This should be fixed with [2.7.0](https://github.com/mavlink/MAVSDK-Python/releases/tag/2.7.0).

[image]

closed 2024-06-02 by julianoes