mavlink/MAVSDK-Python #320: Input external gps through mavsdk

matthewk-rydalch, 2021-03-08:
Is there any way to update the position estimate with gps data coming from an onboard computer?  What I could use is something similar to set_vision_position_estimate(), but will lla instead of pose data.
I conduct research where I need to work with the gps data on an onboard computer, so my gps module is connected to my computer.  I would rather update the position from my computer than have to hook up the module to my pixhawk.

JonasVautherin, 2021-03-09:
I would say that this is a PX4 question. Once it is possible and there exists a MAVLink message for it, we can add it to MAVSDK :+1:

julianoes, 2021-03-09:
From a mavlink point of view we could use [GPS_INPUT](https://mavlink.io/en/messages/common.html#GPS_INPUT) but we would have to add support to PX4 first.