mavlink/MAVSDK-Python #322: do_orbit example

kwanyongcho7, 2021-03-15:
Hi, I want to use do_orbit action and need what to put in ??? part.

await drone.action.do_orbit(15, 2, ???, xstart, ystart, 30)

Is there  any examples of using do_orbit?

JonasVautherin, 2021-03-15:
What the documentation you pointed to says is that this is the `OrbitYawBehavior`, "Yaw behaviour during orbit flight". It means that it tells the drone what to do with the yaw while orbiting. You can ask the drone to "look at the center of the orbit" (`HOLD_FRONT_TO_CIRCLE_CENTER`), keep the heading it had when it started the orbit (`HOLD_INITIAL_HEADING`), etc.

The way it works is that it is an enum you pass as a parameter. Probably something like:

```python
from mavsdk.action import OrbitYawBehavior

...

await drone.action.do_orbit(15, 2, OrbitYawBehavior.HOLD_FRONT_TO_CIRCLE_CENTER, xstart, ystart, 30)
```

kwanyongcho7, 2021-03-16:
Thanks a lot.

closed 2021-03-16 by kwanyongcho7