mavlink/MAVSDK-Python #323: download_log_file() not working

kwanyongcho7, 2021-03-16:
Hi, 

I want to download log files in my px4, and I can see from 0 to 465 log files in QGroundControl.

#!/usr/bin/env python3

import asyncio
from mavsdk import System
from mavsdk.action import OrbitYawBehavior
from mavsdk.log_files import Entry
from mavsdk.log_files import LogFiles

async def run():
    drone = System()
    await drone.connect(system_address="udp://:14540")
    
    print("Waiting for drone to connect...")
    async for state in drone.core.connection_state():
        if state.is_connected:
            print(f"Drone discovered with UUID: {state.uuid}")
            break

    print("Waiting for drone to have a global position estimate...")
    async for health in drone.telemetry.health():
        if health.is_global_position_ok:
            print("Global position estimate ok")
            break

    print("Fetching amsl altitude at home location....")
    async for terrain_info in drone.telemetry.home():
        absolute_altitude = terrain_info.absolute_altitude_m
        break

    
    print("Log Download")
    drone.log_files.download_log_file(412, "C:\\Users\\user\\Desktop\\Logs\\")
    print("Done")
  

if __name__ == "__main__":
    loop = asyncio.get_event_loop()
    loop.run_until_complete(run())

I use this code to get No. 412 log file, but It nothing happened.

Is there any way I can get log file from px4?

JonasVautherin, 2021-03-16:
I think it's the same issue as described [here](https://github.com/mavlink/MAVSDK-Python/issues/312#issuecomment-788894010), which says that the filename should be explicit. 

Can you try with something like `"C:\\Users\\user\\Desktop\\Logs\\log_412.ulg"`?

kwanyongcho7, 2021-03-19:
It said "INVALID_ARGUMENT" 

```
import asyncio
from mavsdk import System
import random
import string
def get_random_string(length):
    # choose from all lowercase letter
    letters = string.ascii_lowercase
    result_str = ''.join(random.choice(letters) for i in range(length))
    return result_str
    
async def run():
    drone = System()
    await drone.connect(system_address="udp://:14540")
    # Start the tasks
    asyncio.ensure_future(print_battery(drone))
    asyncio.ensure_future(print_logs(drone))
    asyncio.ensure_future(get_log(drone))

async def get_log(drone):
    async for progress in drone.log_files.download_log_file(412, "C:\\Users\\user\\Desktop\\log_412.ulg"):
        print(progress)
    print("done")
    
async def print_battery(drone):
    async for battery in drone.telemetry.battery():
        print(f"Battery: {battery.remaining_percent}")
        
async def print_logs(drone):
    logs = await drone.log_files.get_entries()
    for log in logs: 
    	print(f"Log {logs.index(log)}: {log}")
    
if __name__ == "__main__":
    # Start the main function
    asyncio.ensure_future(run())
    # Runs the event loop until the program is canceled with e.g. CTRL-C
    asyncio.get_event_loop().run_forever()
```

JonasVautherin, 2021-03-19:
I supposed that you are not yet using https://github.com/mavlink/MAVSDK/pull/1358, so that error message comes from [here](https://github.com/mavlink/MAVSDK/blob/develop/src/plugins/log_files/log_files_impl.cpp#L223-L234).

I think that you need to `get_entries()` first, so that MAVSDK populates its list.

kwanyongcho7, 2021-03-25:
Can I get py examples for log download?

julianoes, 2021-03-25:
Will do Python release, hopefully that will fix it.

julianoes, 2021-03-26:
@kwanyongcho7 I have done a release, please test with 0.17.0

kwanyongcho7, 2021-03-29:
[image]

async def run():
    drone = System()
    await drone.connect(system_address="udp://:14540")
   
    drone.log_files.download_log_file(41, "C:\\Users\\user\\Desktop\\log_41.ulg")
    await drone.log_files.get_entries()

Task exception was never retrieved
future: <Task finished name='Task-1' coro=<run() done, defined at C:\Users\user\Desktop\MAVSDK-Python\examples\3. log.py:11> exception=LogFilesError(<mavsdk.log_files.LogFilesResult object at 0x000001B996C4CDC0>, 'get_entries()')>
Traceback (most recent call last):
  File "C:\Users\user\Desktop\MAVSDK-Python\examples\3. log.py", line 15, in run
    await drone.log_files.get_entries()
  File "C:\Users\user\AppData\Local\Programs\Python\Python39\lib\site-packages\mavsdk\log_files.py", line 363, in get_entries
    raise LogFilesError(result, "get_entries()")
mavsdk.log_files.LogFilesError: TIMEOUT: 'Timeout'; origin: get_entries(); params: ()

Thanks for help and I tested with the code. 
The Invalid Argument Error is not shown but It said Time out error.

julianoes, 2021-03-29:
@kwanyongcho7 you are using the API wrong now!

```
async for progress in drone.log_files.download_log_file(1, "C:\\Users\\user\\Desktop\\log_4.ulg"):
```

You need to pass the Entry that you receive in `get_entries` as the first result, so instead of the `1`.

julianoes, 2021-03-29:
@kwanyongcho7 try this exampe which now works for me: https://github.com/mavlink/MAVSDK-Python/pull/328

kwanyongcho7, 2021-03-30:
Oh, I read Old document, so is it impossible that I choose log file to download using Log ID?

julianoes, 2021-03-30:
I'm closing this. Just comment again if this is not resolved for you yet @kwanyongcho7, thanks.

closed 2021-03-30 by julianoes

kwanyongcho7, 2021-03-30:
I have speed problem, but I can download it. Thanks for your help!!

kwanyongcho7, 2021-05-06:
Is get_entries() working now?

julianoes, 2021-05-06:
@kwanyongcho7 it should be.

kwanyongcho7, 2021-05-06:
I could download my log on 30 Mar, but now, get_entries() function is not working.. 

Any advice for me??

julianoes, 2021-05-07:
@kwanyongcho7 please create an issue and describe your problem with some detail.