mavlink/MAVSDK-Python #327: async set_position_velocity_ned(position_ned_yaw, velocity_ned_yaw) surprising behavior

amaalo2, 2021-03-26:
Hello,

I am using the async `set_position_velocity_ned(position_ned_yaw, velocity_ned_yaw)` API for a collision avoidance project where I have two quadrotors (ownship and intruder) meeting at a point. I have a problem with the design of the API and with its implementation.

_Design_
I find it strange that `set_position_velocity_ned(position_ned_yaw, velocity_ned_yaw)` requires both a destination point a velocity vector. This could potentially lead to a contradictory scenario. Consider the following : the ownship is at (0,0,-1)  and wants to go to (0,0,-10). With the final point specified then the direction of the velocity vector is constrained. Hence, only the speed should be fed into the API. Furthermore, a user could give a wrong velocity vector to the API and that should not happen. Here, one could feed in v = (1,0,0) which should not be allowed.

_Implementation error_

Even with the appropriate velocity given to the API, the API ignores the feed-forward velocity command.  I have written below a simple code which reproduces the issue. 

`#!/usr/bin/env python3

"""
Caveat when attempting to run the examples in non-gps environments:

`drone.offboard.stop()` will return a `COMMAND_DENIED` result because it
requires a mode switch to HOLD, something that is currently not supported in a
non-gps environment.
"""

import asyncio

from mavsdk import System
from mavsdk.offboard import (OffboardError, PositionNedYaw, VelocityNedYaw)

async def run():
    """ Does Offboard control using position NED coordinates. """

    drone = System(mavsdk_server_address="localhost", port=50040)
    await drone.connect()

    print("-- Arming")
    await drone.action.arm()

    print("-- Setting initial setpoint")
    await drone.offboard.set_position_ned(PositionNedYaw(0.0, 0.0, 0.0, 0.0))

    print("-- Starting offboard")
    try:
        await drone.offboard.start()
    except OffboardError as error:
        print(f"Starting offboard mode failed with error code: {error._result.result}")
        print("-- Disarming")
        await drone.action.disarm()
        return

    async def print_z_velocity(drone):
        async for odom in ownship.telemetry.odometry():
            print(odom.velocity_body.z_m_s)

    asyncio.ensure_future(print_z_velocity(drone))

    print("-- Go 0m North, 0m East, -10m Down within local coordinate system")
    await drone.offboard.set_position_velocity_ned(PositionNedYaw(0.0, 0.0, -10.0, 0.0),VelocityNedYaw(0.0,0.0,-1.0,0.0))
    await asyncio.sleep(10)

    await drone.action.land()

    print("-- Stopping offboard")
    try:
        await drone.offboard.stop()
    except OffboardError as error:
        print(f"Stopping offboard mode failed with error code: {error._result.result}")

if __name__ == "__main__":
    loop = asyncio.get_event_loop()
    loop.run_until_complete(run())
`

Here are the z velocity results where it clear that the quadrotor goes faster than the desired 1m/s. 

0.0106473658233881
-0.008168228901922703
-0.02428366057574749
-0.045656610280275345
-0.0769643783569336
-0.1020079031586647
-0.13413767516613007
-0.16563808917999268
-0.2160492092370987
-0.27253517508506775
-0.30292394757270813
-0.36661872267723083
-0.42053401470184326
-0.49221932888031006
-0.5449873208999634
-0.6234992146492004
-0.6891863942146301
-0.7496235966682434
-0.8233122825622559
-0.8922696709632874
-0.9995257258415222
-1.0931577682495117
-1.219588279724121
-1.313400387763977
-1.426629662513733
-1.5306116342544556
-1.686812162399292
-1.850787878036499
-1.9868569374084473
-2.1087396144866943
-2.287050724029541
-2.377728223800659
-2.5615577697753906
-2.7105743885040283
-2.841900587081909
-3.0307421684265137
-3.1824533939361572
-3.315337657928467
-3.5051984786987305
-3.6568148136138916
-3.7890307903289795
-3.977020502090454
-4.1264262199401855
-4.31162691116333
-4.440070152282715
-4.585522174835205
-4.711474418640137
-4.890402317047119
-5.032202243804932
-5.206510543823242
-5.326474189758301
-5.461108684539795
-5.626007080078125
-5.739096164703369
-5.865636825561523
-5.9742255210876465
-6.125515937805176
-6.272313594818115
-6.386413097381592
-6.484564304351807
-6.594022750854492
-6.687442779541016
-6.816668510437012
-6.916486740112305
-7.037040710449219
-7.118645191192627
-7.2089338302612305
-7.285554885864258
-7.39096736907959
-7.471928596496582
-7.540432453155518
-7.635262489318848
-7.708396911621094
-7.770180702209473
-7.854506492614746

@julianoes

julianoes, 2021-03-29:
Ok, I can reproduce it. Something is definitely off. I'm having a look.

julianoes, 2021-03-29:
@amaalo2 it looks like this is a PX4 problem.

If you want you can try my fix: https://github.com/PX4/PX4-Autopilot/pull/17261

amaalo2, 2021-03-29:
Hi @julianoes , I took a look at the pull request, and it seems to have failed the checks. I am not too familiar with PX4 code, so I can't really help there but many thanks for helping out!

julianoes, 2021-03-30:
@amaalo2 you can already checkout that branch for PX4 and test it if you want.

It is however likely going to take a few days for reviews and fixes, until it's merged into PX4 master.

julianoes, 2021-03-30:
@amaalo2 I was educated that the current behavior is actually correct. It allows to add a feed-forward term to the controller, that's also what the documentation of this method says.

It is not, however, a limit. If you want to limit the velocity you need to either use velocity setpoints only, and close the position loop outside. Or you need to use the velocity control parameters and the maximum speed: https://docs.px4.io/master/en/advanced_config/parameter_reference.html#MPC_XY_VEL_MAX

closed 2021-03-30 by julianoes

daxaxelrod, 2021-08-02:
Ah, thanks so much @julianoes! I was hoping to set the velocity limit for each action as sometimes our drone has further to go and we can open up the throttle while still moving towards and positionNED. Seems like the param setting is going to work.

Curious what you think of mavlink message type [MAV_CMD_DO_CHANGE_SPEED](https://mavlink.io/en/messages/common.html#MAV_CMD_DO_CHANGE_SPEED)