mavlink/MAVSDK-Python #329: Orbit Command Stops Telemetry Updates

DoppleGangster, 2021-03-29:
When engaging the orbit mode telemetry updates that are subscribed cease to update. Once the orbit command is stopped and manual mode reengaged the telemetry data resumes updating.

JonasVautherin, 2021-03-29:
Did you try with QGC? Does it keep receiving updates?

DoppleGangster, 2021-03-29:
I tried to find the option to engage an orbit with QGC and I must be missing how this would be implemented. Is there a way to give commands that are not listed in the drop down menu?

DoppleGangster, 2021-03-29:
I'll try running the orbit outside the simulator to see if maybe the issue still persists.

JonasVautherin, 2021-03-29:
You can start the orbit from MAVSDK and check if QGC still receives the telemetry or not. The question is whether it is still sent by PX4 or not :+1:.

DoppleGangster, 2021-03-30:
It did continue to update the values. After digging into it I realize I wasn't calling asyncio.sleep while orbiting. My messaging runs in separate thread so I could still stop the orbit but the subscribed threads that were running inside the asyncio.ensure_future did not receive any operational time as they are part of the thread executing the orbit.

Thanks for you help.

closed 2021-03-30 by DoppleGangster