mavlink/MAVSDK-Python #330: Reset GPS

kwanyongcho7, 2021-04-02:
Can I reset GPS status of my multirotor? 

I am using Airsim HITL and when collision happens, I want to reset my drone and re-fly.

I use Airsim API reset to set original but the compass in QGroundcontrol still pointing wrong way, I hope it set to North as I first start it. Also, Altitude is keep decreasing that wants to be set Zero. 

Do i have to change Airsim code? or can I simply using MAVSDK PYTHON code?

julianoes, 2021-04-05:
Unfortunately, a reset in SITL is not supported by PX4.

kwanyongcho7, 2021-04-06:
@julianoes 
I used Airsim Reset. I reset my drone when it is upside down, I can see the drone correctly but QGC's gyroscope(next to the compass) kept upside down which makes me cannot takeoff.

julianoes, 2021-04-06:
@kwanyongcho7 after a while the ekf should detect the problem and reset the attitude but that might take a while sometimes.

kwanyongcho7, 2021-04-07:
I guess that Airsim automatically fix upside down like a teleport, but drone & QGC thought it is still upside down..

julianoes, 2021-04-08:
Right. So the first step would be to support something like reset in PX4.