mavlink/MAVSDK-Python #332: takeoff_and_land.py error

mommotti, 2021-04-08:
**My setup:** Pixhawk 2 connected to a Raspberry Pi 3b

**The error:**

> pi@raspberrypi:~/drone $ python3 takeoff_and_land.py
> Waiting for mavsdk_server to be ready...
> Connected to mavsdk_server!
> Waiting for drone to connect...
> Drone discovered with UUID: 0
> Waiting for drone to have a global position estimate...
> Global position estimate ok
> -- Arming
> -- Taking off
> Traceback (most recent call last):
>   File "takeoff_and_land.py", line 40, in <module>
>     loop.run_until_complete(run())
>   File "/usr/lib/python3.7/asyncio/base_events.py", line 584, in run_until_complete
>     return future.result()
>   File "takeoff_and_land.py", line 30, in run
>     await drone.action.takeoff()
>   File "/home/pi/.local/lib/python3.7/site-packages/mavsdk/action.py", line 360, in takeoff
>     raise ActionError(result, "takeoff()")
> mavsdk.action.ActionError: COMMAND_DENIED: 'Command Denied'; origin: takeoff(); params: ()
> Exception ignored in: <function System.__del__ at 0x75201540>
> Traceback (most recent call last):
>   File "/home/pi/.local/lib/python3.7/site-packages/mavsdk/system.py", line 55, in __del__
>   File "/home/pi/.local/lib/python3.7/site-packages/mavsdk/system.py", line 91, in _stop_mavsdk_server
> ImportError: sys.meta_path is None, Python is likely shutting down

Am I missing something?
Thank you for your help

julianoes, 2021-04-08:
>  mavsdk.action.ActionError: COMMAND_DENIED: 'Command Denied'; origin: takeoff(); params: ()

Something is missing to arm. You can print out statustexts to get a hint what is missing, see: http://mavsdk-python-docs.s3-website.eu-central-1.amazonaws.com/plugins/telemetry.html#mavsdk.telemetry.StatusText

JonasVautherin, 2021-04-08:
Just to add my two cents: `COMMAND_DENIED` means that PX4 refused the command (and answered with a "denied" result), meaning that your drone is not ready to arm. So it's not a bug, it's really just an exception that you may catch with `try... except`.

mommotti, 2021-04-08:
Thank you guys!
I'm currently running MAVProxy on the Pi, and QGroundControl on Windows.
Here is a screenshot from QGC: 
[image]
And sometimes I get a GPS glitch:
[image]
I managed to output the battery telemetry with this:
    async for voltage_v in drone.telemetry.battery():
        print(voltage_v)
        break

and got output:
`Battery: [voltage_v: 0.6650000214576721, remaining_percent: 0.0]`

If these errors cannot be resolved, there is a way to bypass them?
I'm currently working from home and I can put the drone outside my window but apparently it's giving some problems...
Thank you in advance!

JonasVautherin, 2021-04-08:
Are you running PX4 or Ardupilot?

mommotti, 2021-04-08:
> 
> 
> Are you running PX4 or Ardupilot?
Apparently Ardupilot! :/

I'm currently using a PixHawk 2.1:
[image]
I just checked some QGroundControl images on Google, and on the top right corner there is the PX4 logo. Mine has an ArduPilot logo.
And just read that ArduPilot is not quite compatible with MAVSDK... https://github.com/mavlink/MAVSDK/issues/728
Is there a way to switch from ArduPilot to PX4? I'm very new to this.
Thanks in advance

JonasVautherin, 2021-04-08:
In QGC, when you flash a firmware, you can choose if you want PX4 or Ardupilot. I think that would be the easy way :slightly_smiling_face:.

mommotti, 2021-04-11:
> 
> 
> In QGC, when you flash a firmware, you can choose if you want PX4 or Ardupilot. I think that would be the easy way 🙂.

Thank you very much! Once I switch to PX4 I get an issue about the power (battery) setup, this issue is not present with Ardupilot... I opened an issue on the PX4 page for that: https://github.com/PX4/PX4-Autopilot/issues/17366
A colleague managed to use takeoff_and_land.py with Ardupilot:
The code:
[image]
Running the program:
[image]

We were wondering what kind of issues we will face by using Ardupilot in the future!

By the way thank you guys for the amazing work! Been learning a lot from your talks on YouTube!

JonasVautherin, 2021-04-11:
Seems like it's solved, feel free to reopen if it's not!

closed 2021-04-11 by JonasVautherin

julianoes, 2021-04-12:
Thanks for the feedback @mommotti.