mavlink/MAVSDK-Python #333: goto_location makes drone height zero, parameters

kwanyongcho7, 2021-04-09:
I am using goto_location action to fly drone to specific location

await drone.action.goto_location(x, y, 31, degree)

but, sometimes It goes forward + down to zero height.

and about the yaw degree, is there any way i just make drone goes forward? like heading to?

julianoes, 2021-04-09:
According to the [docs](http://mavsdk-python-docs.s3-website.eu-central-1.amazonaws.com/plugins/action.html?highlight=goto#mavsdk.action.Action.goto_location) `goto_location` uses altitude in AMSL (above mean sea level). Are you sure you are setting the correct altitude?

To prevent yawing you should be able to set heading to NAN.

kwanyongcho7, 2021-04-12:
So, I used drone.telemetry.home() terrain_info.absolute_altitude_m, and It works. However, Sometimes, drone goes too high.

julianoes, 2021-04-12:
Is this outdoor with GPS or indoor?

kwanyongcho7, 2021-04-12:
I think I solved it with refreshing terrain_info.absolute_altitude value when it starts each round.

closed 2021-04-12 by kwanyongcho7