mavlink/MAVSDK-Python #334: Logging for multiple vehicles - surprising behavior

amaalo2, 2021-04-09:
Hello, I was running this script with two UAVs `logfile_download.py`. I connected to the second drone and checked the output log. 
It seems that the output log is always the same although the UAVs stay both on the ground and do not move. Moreover, the output log always corresponds to a mission that I ran 2 or 3 days ago and did the values are always the same. Lastly, output log gets copied over 3 or 4 times.  However, the logging function for drone 1 works properly (i.e. by changing to `port=50040`)

Here is the log that I keep on getting : https://review.px4.io/plot_app?log=52e70e86-98fd-48d6-984c-f72a6e200aed 

@julianoes  @JonasVautherin  what do you think could cause this?

[image]

```
#!/usr/bin/env python3

import asyncio
from mavsdk import System
import sys

async def run():
    # Init the drone
    drone = System(mavsdk_server_address="localhost", port=50041)
    await drone.connect()

    print("Waiting for drone to connect...")
    async for state in drone.core.connection_state():
        if state.is_connected:
            print(f"Drone discovered")
            break

    entries = await get_entries(drone)
    for entry in entries:
        await download_log(drone, entry)

async def download_log(drone, entry):
    date_without_colon = entry.date.replace(":", "-")
    filename = f"/home/anthony/Documents/GitRepos/TonySimulinkCode/Linux_Testing/src/logs/{date_without_colon}-drone_id{id}.ulog"
    print(f"Downloading: log {entry.id} from {entry.date} to {filename}")
    previous_progress = -1
    async for progress in drone.log_files.download_log_file(entry, filename):
        new_progress = round(progress.progress*100)
        if new_progress != previous_progress:
            sys.stdout.write(f"\r{new_progress} %")
            sys.stdout.flush()
            previous_progress = new_progress

async def get_entries(drone):
    entries = await drone.log_files.get_entries()
    for entry in entries:
        print(f"Log {entry.id} from {entry.date}")
    return entries

if __name__ == "__main__":
    asyncio.get_event_loop().run_until_complete(run())
```

[image]

JonasVautherin, 2021-04-09:
```
    filename = f"/home/anthony/Documents/GitRepos/TonySimulinkCode/Linux_Testing/src/logs/{date_without_colon}-drone_id{id}.ulog"
```

Shouldn't this be using `{entry.id}` instead? What is `{id}`?

```
    filename = f"/home/anthony/Documents/GitRepos/TonySimulinkCode/Linux_Testing/src/logs/{date_without_colon}-drone_id{entry.id}.ulog"
```

amaalo2, 2021-04-09:
@JonasVautherin  Yes, you're right  it should be `{entry.id}` but nevertheless, the problem still persists. 
https://review.px4.io/plot_app?log=98a3733c-89a4-426c-8899-a6494fda02a3

JonasVautherin, 2021-04-09:
So you're saying that it always downloads the oldest log file, whatever entry you select?

What does `print(f"Log {entry.id} from {entry.date}")` in `get_entries` print?

Note that the implementation is [here](https://github.com/mavlink/MAVSDK/blob/main/src/plugins/log_files/log_files_impl.cpp), if you want to try debugging :innocent:.