mavlink/MAVSDK-Python #335: failure inject

kwanyongcho7, 2021-04-14:
`await drone.failure.inject(mavsdk.failure.FailureUnit.SENSOR_MAG, mavsdk.failure.FailureType.OFF, 0)
`
I try to inject failure to my drone in Airsim, HITL.

`mavsdk.failure.FailureError: UNSUPPORTED: 'Unsupported'; origin: inject(); params: (<FailureUnit.SENSOR_MAG: 2>, <FailureType.OFF: 1>, 0)`

But, I got error message.

Is this my code problem or does my drone not support this failure?

JonasVautherin, 2021-04-14:
> Is this my code problem or does my drone not support this failure?

Seems like the drone answered "UNSUPPORTED" :+1:

julianoes, 2021-04-15:
The failures are only supported for SITL right now. This needs to be added to PX4 first.

JonasVautherin, 2021-04-19:
Feels like this can be closed, feel free to reopen if necessary!

closed 2021-04-19 by JonasVautherin