mavlink/MAVSDK-Python #336: Failure Injection Connection Error

haiyu-zat, 2021-04-19:
Hi there,

I am trying to do failure injection in SITL, I am running the following command:

`await drone.failure.inject(FailureUnit.SYSTEM_BATTERY, FailureType.OFF, 0) `

But I am getting the following error:

`CONNECTION_ERROR: 'Connection Error'; origin: inject(); params: (<FailureUnit.SYSTEM_BATTERY: 9>, <FailureType.OFF: 1>, 0)`

Could you help me identify the issue? It works fine when I try with FailureUnit.SENSOR_GPS

Thanks!

JonasVautherin, 2021-04-19:
Can you share some code that we could use to reproduce the bug?

haiyu-zat, 2021-04-19:
```
import asyncio

from mavsdk import *
from mavsdk.failure import FailureUnit, FailureType

async def run():
    drone = System()
    await drone.connect(system_address="udp://:14540")

    #await drone.failure.inject(FailureUnit.SENSOR_GPS, FailureType.OK, 0)

    print("Waiting for drone to connect...")
    async for state in drone.core.connection_state():
        if state.is_connected:
            print(f"Drone discovered with UUID: {state.uuid}")
            break

    print("Waiting for drone to have a global position estimate...")
    async for health in drone.telemetry.health():
        if health.is_global_position_ok:
            print("Global position estimate ok")
            break
    
    print("-- Arming")
    await drone.action.arm()

    print("-- Taking off")
    await drone.action.takeoff()

    await asyncio.sleep(5)

    print("Injecting Failure")
    #await drone.failure.inject(FailureUnit.GPS_SENSOR, FailureType.OFF, 0)
    await drone.failure.inject(FailureUnit.SYSTEM_BATTERY, FailureType.OFF, 0)

    await asyncio.sleep(10)

    print("-- Landing")
    await drone.action.land()

if __name__ == "__main__":
    loop = asyncio.get_event_loop()
    loop.run_until_complete(run())
```

julianoes, 2021-04-20:
You need to do this call:
```
    print("Waiting for drone to connect...")
    async for state in drone.core.connection_state():
        if state.is_connected:
            print(f"Drone discovered with UUID: {state.uuid}")
            break
```

before:

```
await drone.failure.inject(FailureUnit.SENSOR_GPS, FailureType.OK, 0)
```

haiyu-zat, 2021-04-20:
Hi,

Sorry for being unclear, in my example I commented out the sensor_gps failure injections and am just calling
`await drone.failure.inject(FailureUnit.SYSTEM_BATTERY, FailureType.OFF, 0)`

julianoes, 2021-04-20:
Oh, right. Is that against SITL or a real Pixhawk?

haiyu-zat, 2021-04-20:
SITL