mavlink/MAVSDK-Python #337: Mission: start_mission() fails with param error

DoppleGangster, 2021-04-20:
I'm attempting to upload and start a mission. Using a fresh install (pip3 install mavsdk) with the example mission.py results in the following error:

> mavsdk.mission.MissionError: ERROR: 'Error'; origin: start_mission(); params: ()

I tried casting all the hard coded parameters to make sure it wasn't an issue of python passing an invalid type but that did not correct the error:

`mission_items.append(MissionItem(float(47.398039859999997),
                                     float(8.5455725400000002),
                                     float(25.0),
                                     float(10.0),
                                     bool(True),
                                     float('nan'),
                                     float('nan'),
                                     MissionItem.CameraAction.NONE,
                                     float('nan'),
                                     float('nan')))`

Could use a pro-tip to help me get this sorted out.

Thanks!

JonasVautherin, 2021-04-20:
Does the [example mission](https://github.com/mavlink/MAVSDK-Python/blob/main/examples/mission.py) work? Did you try your code against SITL? Do the other functions work (e.g. can you arm and takeoff?)?

DoppleGangster, 2021-04-20:
That is the example I am attempting to run. The I am using the SITL to for the test. It does appear to arm. Below is the full output when executing the file:

> Waiting for mavsdk_server to be ready...
Connected to mavsdk_server!
Waiting for drone to connect...
Drone discovered with UUID: 0
-- Uploading mission
Mission progress: -1/0
-- Arming
-- Starting mission
Mission progress: 0/3
Traceback (most recent call last):
  File "mission.py", line 103, in <module>
    loop.run_until_complete(run())
  File "/usr/lib/python3.8/asyncio/base_events.py", line 616, in run_until_complete
    return future.result()
  File "mission.py", line 67, in run
    await drone.mission.start_mission()
  File "/home/richard/.local/lib/python3.8/site-packages/mavsdk/mission.py", line 760, in start_mission
    raise MissionError(result, "start_mission()")
mavsdk.mission.MissionError: ERROR: 'Error'; origin: start_mission(); params: ()

julianoes, 2021-04-20:
Can you show the whole script?

This doesn't look like it's actually connected properly:
```
Drone discovered with UUID: 0
```

DoppleGangster, 2021-04-20:
That is the output I get using [mission.py](https://github.com/mavlink/MAVSDK-Python/blob/main/examples/mission.py) from the example folder.

I'm running jmavsim:

> make px4_sitl_default jmavsim

JonasVautherin, 2021-04-20:
Does PX4 show an output? E.g. saying that the mission is too far away? What you can try is upload the mission from MAVSDK, then download it from QGC, and check why it is refused...

DoppleGangster, 2021-04-21:
I couldn't find any issues from the ulg file when I uploaded them for viewing. I then deleted all the log files and then attempted to run the mission.py again. This time it executed. Does the clear_mission() command need to be run before uploading a new mission? I know I had initially uploaded invalid data with my own program before attempting to use the mission.py file.

DoppleGangster, 2021-04-21:
Alright I know how to recreate the issue:

1. Delete PX4-Autopilot and then go grab a fresh copy follow the directions to [build](https://docs.px4.io/master/en/dev_setup/dev_env_linux_ubuntu.html).
2. Start the simulation using 'make px4_sitl_default jmavsim'
3. Run mission.py

The error I had above should happen. If then you start QGC and again run mission.py it should execute. Even after closing QGC it will continue to run the python script from there on out.

DoppleGangster, 2021-04-21:
I'm going to close this as this does not seem related to MAVSDK-Python.

Thanks for your responses.

closed 2021-04-21 by DoppleGangster