mavlink/MAVSDK-Python #338: Question: ServerUtility Functionality

DoppleGangster, 2021-04-22:
I am trying to find a way to access the mavsdk_server console output string. Is the purpose of the ServerUtility plugin to access the server output strings?

If not then is there another plugin that offers this capabilty?

Thanks!

JonasVautherin, 2021-04-22:
The best way right now is to run `mavsdk_server` in command line manually (e.g. `./mavsdk_server udp://:14540`) and then to connect MAVSDK-Python to it with:

```python
from mavsdk import System
drone = System(mavsdk_server_address="localhost")
```

Setting `mavsdk_server_address` means that MAVSDK-Python will not start the embedded `mavsdk_server` and will instead just connect to the one you started manually. The output of the one you start manually will appear in the terminal.

DoppleGangster, 2021-04-22:
Is there a way to subscribe to the servers output that can be accessed from within the python script or is that what you are referring to when mention adding an API or file output?

JonasVautherin, 2021-04-22:
That's what I am referring

DoppleGangster, 2021-04-22:
Sounds good. Thanks for the response.