mavlink/MAVSDK-Python #340: Problem with setup.py on Raspberry

DonatelloX, 2021-05-03:
The command:
python3 setup.py build
does not work properly on Raspberry beacuse the downloaded server is _mavsdk_server_musl_x86_64_ instead of _mavsdk_server_linux-armv7l-musl_
I think the problem is in line 55, because MAVSDK_SERVER_ARCH is null.

JonasVautherin, 2021-05-03:
Does `pip install mavsdk` work?

DonatelloX, 2021-05-03:
> pip install mavsdk
Yes, "pip3" works.
But if you want to build and install the package locally (with the latest repository changes) then setup.py won't work on Raspberry.

julianoes, 2021-05-03:
Aha, can you try by setting it manually:
```
export MAVSDK_SERVER_ARCH=linux_armv7l
```

DonatelloX, 2021-05-03:
OK @julianoes it works with:
`export MAVSDK_SERVER_ARCH=armv7l`
before setup.py

julianoes, 2021-05-04:
Ok, so is the issue resolved. Or do you have a suggestion what we should improve?

DonatelloX, 2021-05-04:
Yes, the issue is resolved. Thank you for help!

In my opinion, it would be better to update the repository readme.md:
_The protoc-gen-dcsdk plugin should be updated, because it has been renamed to protoc-gen-mavsdk._
And I would add these rows:

After generating the wrapper and only in ARM architectures with linux, defines a variable:
`export MAVSDK_SERVER_ARCH=<ARM embedded architecture>`
Supported architectures: armv6l, armv7l and aarch64. For example in raspberry it is armv7l.

Then you can install a development (editable) bersion of the package using:
```
python3 setup.py build
pip3 install -e .
```

julianoes, 2021-05-04:
That sounds great. Do you want to do a pull request :wink:?

sandhyavineethasanam, 2021-05-24:
i have the same problem
bash: scripts/kconfig/merge_config.sh: No such file or directory
i am doing a similar project as yours.
porting android 10 onto raspberry pi 3. 
i have tried this "export MAVSDK_SERVER_ARCH=linux_armv7l" 
the ubuntu version i am using is 20.04
pls can u help

DonatelloX, 2021-05-24:
In raspberry OS (32bit), I use these instructions: 
```
git clone https://github.com/mavlink/MAVSDK-Python --recursive
cd MAVSDK-Python
cd proto/pb_plugins
pip3 install -r requirements.txt
which protoc-gen-mavsdk
cd ../..
pip3 install -r requirements.txt -r requirements-dev.txt
./other/tools/run_protoc.sh
export MAVSDK_SERVER_ARCH=armv7l
python3 setup.py build
pip3 install -e .
```
No problem on both Raspberry 3+ and Raspberry CM4. 
Maybe your OS is 64bit so you should be using `export MAVSDK_SERVER_ARCH=aarch64`, but I'm not sure if this can work though.

sandhyavineethasanam, 2021-05-24:
tq for the help.
i will try it and let you know

sandhyavineethasanam, 2021-05-25:
It's not working. My project is porting android onto Raspberry pi. I have
the latest version of Ubuntu.  I have tried to download  every version of
Android but I kept getting errors.

Pls can u assist me with a doubt I have
1) for suppose I download the compiled version of raspand directly from the
internet and then perform the SD card partitions and putting the files into
the respective partitions and managed to boot the Raspberry pi.
How can I see from my terminal if I have actually been successful...from
terminal.
My mentors what's to see the terminal to determine so.
I mean are there any set of commands that we type in the terminal to see so.

Please advise

closed 2021-06-04 by DonatelloX