mavlink/MAVSDK-Python #345: Raspberry Pi : from mavsdk : Illegal Instruction

TheAsuraSlayer, 2021-05-12:
Im starting off with mavsdk on my raspberry pi connected to a pixhawk via the UART.

the following is the code.

when i run it with   "python3 simple_arm.py"

the terminal shows the output as "Illegal Instruction"

i think ive narrowed down the issue to 
`from mavsdk import System`

```python
import asyncio
from mavsdk import System

async def run():
    print("In run")
    drone =System()
    await drone.connect(system_address="udp://localhost:14550")
    print("Waiting to connect to drone.....")
    async for state in drone.core.connection_state():
        if state.is_connected:
            print("Conected to drone: %s" % state.uuid)
            break

    print("Waiting for drone to have a global position estimate...")
    async for health in drone.telemetry.health():
        if health.is_global_position_ok:
            print("Global position estimate ok")
            break    
if __name__ == "__main__":
    print("In main loop")
    loop = asyncio.get_event_loop()
    loop.run_until_complete(run())
```

julianoes, 2021-05-12:
@TheAsuraSlayer how did you install mavsdk? Did you do `pip3 install --user mavsdk`?

TheAsuraSlayer, 2021-05-12:
> @TheAsuraSlayer how did you install mavsdk? Did you do `pip3 install --user mavsdk`?

Yeah that's how I did it.
I did an update after that too just to be sure

julianoes, 2021-05-12:
Ok, that's odd. Let me try to reproduce that.

Can you run the file mavsdk_server manually. You should find it by looking for it:

```
find ~ -name mavsdk_server
```

E.g. for me, it's `~/.local/lib/python3.9/site-packages/mavsdk/bin/mavsdk_server`

And then run that manually and copy the output here.

TheAsuraSlayer, 2021-05-12:
[image]

 the above is the result if i run it manually

julianoes, 2021-05-12:
Interesting, so that just works fine.

Can you take a screenshot of when the illegal instruction happens?

TheAsuraSlayer, 2021-05-12:
> Interesting, so that just works fine.
> 
> Can you take a screenshot of when the illegal instruction happens?

[image]

julianoes, 2021-05-12:
Ok, thank you. I'll try to reproduce it.

TheAsuraSlayer, 2021-05-12:
> Ok, thank you. I'll try to reproduce it.

Thank you. The issue definitely is with accesing(importing the mavsdk library)

`from mavsdk import System`

even a simple `import mavsdk` in an otherwise working script is throwing up the same 'Illegal Instruction' error

julianoes, 2021-05-12:
It works for me:
```
./test.py 
In main loop
In run
Waiting for mavsdk_server to be ready...
Connected to mavsdk_server!
Waiting to connect to drone.....
^CTraceback (most recent call last):
  File "./test.py", line 23, in <module>
    loop.run_until_complete(run())
  File "/usr/lib/python3.7/asyncio/base_events.py", line 571, in run_until_complete
    self.run_forever()
  File "/usr/lib/python3.7/asyncio/base_events.py", line 539, in run_forever
    self._run_once()
  File "/usr/lib/python3.7/asyncio/base_events.py", line 1739, in _run_once
    event_list = self._selector.select(timeout)
  File "/usr/lib/python3.7/selectors.py", line 468, in select
    fd_event_list = self._selector.poll(timeout, max_ev)
KeyboardInterrupt
```

Are you sure you don't have another `mavsdk_server` that is used somewhere?

You can look for it:
```
sudo find / -name mavsdk_server
```

TheAsuraSlayer, 2021-05-12:
> It works for me:
> 
> ```
> ./test.py 
> In main loop
> In run
> Waiting for mavsdk_server to be ready...
> Connected to mavsdk_server!
> Waiting to connect to drone.....
> ^CTraceback (most recent call last):
>   File "./test.py", line 23, in <module>
>     loop.run_until_complete(run())
>   File "/usr/lib/python3.7/asyncio/base_events.py", line 571, in run_until_complete
>     self.run_forever()
>   File "/usr/lib/python3.7/asyncio/base_events.py", line 539, in run_forever
>     self._run_once()
>   File "/usr/lib/python3.7/asyncio/base_events.py", line 1739, in _run_once
>     event_list = self._selector.select(timeout)
>   File "/usr/lib/python3.7/selectors.py", line 468, in select
>     fd_event_list = self._selector.poll(timeout, max_ev)
> KeyboardInterrupt
> ```
> 
> That's on:
> 
> ```
> Linux rpi1b-test 5.10.17+ #1403 Mon Feb 22 11:26:13 GMT 2021 armv6l GNU/Linux
> ```
> 
> With:
> 
> ```
> pip3 install --upgrade mavsdk
> Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
> Requirement already up-to-date: mavsdk in ./.local/lib/python3.7/site-packages (0.17.0)
> Requirement already satisfied, skipping upgrade: aiogrpc>=1.5 in ./.local/lib/python3.7/site-packages (from mavsdk) (1.8)
> Requirement already satisfied, skipping upgrade: protobuf>=3.13.0 in ./.local/lib/python3.7/site-packages (from mavsdk) (3.13.0)
> Requirement already satisfied, skipping upgrade: grpcio>=1.11.0 in /usr/lib/python3/dist-packages (from mavsdk) (1.16.1)
> Requirement already satisfied, skipping upgrade: setuptools in ./.local/lib/python3.7/site-packages (from protobuf>=3.13.0->mavsdk) (50.3.2)
> Requirement already satisfied, skipping upgrade: six>=1.9 in /usr/lib/python3/dist-packages (from protobuf>
> ```
> 
> Are you sure you don't have another `mavsdk_server` that is used somewhere?
> 
> You can look for it:
> 
> ```
> sudo find / -name mavsdk_server
> ```
> 
> Also, I installed grpcio like this:
> 
> ```
> sudo apt install python3-grpcio
> ```

with regards to  find mavsdk_server this is the output

[image]

julianoes, 2021-05-12:
Aha! Might it be that you have installed mavsdk_server previously with root? If so, try to remove the root one:
```
sudo rm /usr/local/lib/python3.7/dist-packages/mavsdk*
```

TheAsuraSlayer, 2021-05-12:
> Aha! Might it be that you have installed mavsdk_server previously with root? If so, try to remove the root one:
> 
> ```
> sudo rm -rf /usr/local/lib/python3.7/dist-packages/mavsdk*
> ```
> 
> Or try with
> `sudo pip3 remove mavsdk`

removed the root mavsdk, but still the same problem

[image]

[image]

```
print("check 1")
import asyncio
print("check 2")
from mavsdk import System

print("Check 3")

async def run():
    print("In run")
    drone =System(mavsdk_server_address="localhost", port=50051)
    await drone.connect(system_address="udp://localhost:14550")
    print("Waiting to connect to drone.....")
    async for state in drone.core.connection_state():
        if state.is_connected:
            print("Conected to drone: %s" % state.uuid)
            break

    print("Waiting for drone to have a global position estimate...")
    async for health in drone.telemetry.health():
        if health.is_global_position_ok:
            print("Global position estimate ok")
            break
if __name__ == "__main__":
    print("In main loop")
    loop = asyncio.get_event_loop()
    loop.run_until_complete(run())

```

julianoes, 2021-05-12:
Interesting, can you check how grpcio is installed?

TheAsuraSlayer, 2021-05-12:
> Interesting, can you check how grpcio is installed?

I'm actually doing a complete reinstall and checking. Will update once it's done

TheAsuraSlayer, 2021-05-12:
> > Interesting, can you check how grpcio is installed?
> 
> I'm actually doing a complete reinstall and checking. Will update once it's done

Unfortunately even after a completely clean install.... THE SAME RESULT

what does ILLEGAL INSTRUCTION even mean

julianoes, 2021-05-13:
Odd! My assumption is that some binary is not armv6 compatible and contains an "illegal instruction" for that architecture.

julianoes, 2021-05-13:
Can you try to run it in gdb to see when/where it happens?

TheAsuraSlayer, 2021-05-14:
> Can you try to run it in gdb to see when/where it happens?

how do i do that? can you walk me through the steps?

julianoes, 2021-05-17:
Something like this: `gdb --args python3 simple_arm.py` and then you enter `run` and enter.

beenjamminnallday, 2021-05-17:
Hello Julian, 

Me and a friend of mine are having the same issue. The difference with the above inquiry is were using a Pi Zero W. I don't think the HW is the issue as ive used the same Pi to run MAVSDK Python (different SD card). On the PI 4 model B, I am not unable to get this error. 

Thanks for your time. Attached is my output from gdp. 

[image]

TheAsuraSlayer, 2021-05-17:
> Something like this: `gdb --args python3 simple_arm.py` and then you enter `run` and enter.

[image]

so its basically the same issue with benjamin too.

beenjamminnallday, 2021-05-17:
When trying a manual build for contributors, I got this.
[image]

julianoes, 2021-05-17:
Oh, the hint is `libcrypto` so this might have something to do with openssl.

julianoes, 2021-05-17:
Check this: https://raspberrypi.stackexchange.com/questions/108985/raspian-buster-lite-grep-fails-with-illegal-instruction

TheAsuraSlayer, 2021-05-28:
> Check this: https://raspberrypi.stackexchange.com/questions/108985/raspian-buster-lite-grep-fails-with-illegal-instruction

tried a different powersupply. reinstalled everything from scratch. still no luck

daxaxelrod, 2021-06-05:
Getting the same thing. I'm able to connect to the vehicle using mavproxy.py

daxaxelrod, 2021-06-05:
Another note, I got illegal instructions when I was running on a Pi zero w 1.1 2017. I switched back to my normal RPi 3 and the error went away. Chipset architecture related maybe?

julianoes, 2021-06-07:
Maybe. I could not see it with my RPi1 which should be the same arch as RPi Zero W. Can you tell me which OS and exactly which version you are using? That way I can try to reproduce it again.

liuuuk311, 2021-06-10:
Hello @julianoes I have the same problem describe above and I'm using this version
```
pi@raspberrypi:~ $ hostnamectl
   Static hostname: raspberrypi
         Icon name: computer
        Machine ID: 6d7ad4a334794ff79b3684014c62b213
           Boot ID: fa5c334e80fe4b5983a95098e9889e39
  Operating System: Raspbian GNU/Linux 10 (buster)
            Kernel: Linux 5.10.17+
      Architecture: arm
 ```

julianoes, 2021-06-10:
And which steps did you do to install mavsdk and the pip dependencies?

liuuuk311, 2021-06-10:
Initially I tried just `pip3 install mavsdk`. Then the error came up and I tought was because of the mavsdk_server so I downloaded the compiled version from [here](https://github.com/mavlink/MAVSDK/releases). However the error is still present.

By reading all the comments on this issue, it seems like it's not a problem with the mavsdk_server, is it?

julianoes, 2021-06-10:
I'm suspecting one of the libraries used, mostly because it works for me. I'll try fresh to reproduce it.

daxaxelrod, 2021-06-10:
@julianoes Just to follow up

Pi Zero - Raspbian GNU/Linux Version 9
Python v3.9.5

mavsdk version - 0.18.0 
also tried version 0.17.0

julianoes, 2021-06-14:
Ok, I can reproduce the problem on my RPi 1:

> Operating System: Raspbian GNU/Linux 10 (buster)
> Kernel: Linux 5.10.17+
> Architecture: arm

I believe the problem stems from grpcio. When I install grpcio using `sudo apt-get install python3-grpcio` it works. However, when I install it with pip as part of `pip3 install --user mavsdk` it doesn't work and fails with `Illegal Instruction`.

I've followed up in the issues, see:
https://github.com/grpc/grpc/issues/26474
https://github.com/piwheels/packages/issues/217

For now, I suggest this workaround:
```
pip3 uninstall grpcio
sudo apt-get install python3-grpcio
```

liuuuk311, 2021-06-15:
I'll try your workaround, thank you @julianoes!

julianoes, 2021-07-02:
This should be fixed according to https://github.com/piwheels/packages/issues/217#issuecomment-872874116.

closed 2021-07-02 by julianoes