mavlink/MAVSDK-Python #346: Debugging mission_raw upload

petergerten, 2021-05-13:
What is the best way to debug mission_raw upload errors?
For example I get:

```
mavsdk.mission_raw.MissionRawError: UNSUPPORTED: 'Unsupported'; origin: upload_mission(); params: ([<mavsdk.mission_raw.MissionItem object at 0x7fb574687310>, <mavsdk.mission_raw.MissionItem object
at 0x7fb5762f0210>, <mavsdk.mission_raw.MissionItem object at 0x7fb575710bd0>, <mavsdk.mission_raw.MissionItem object at 0x7fb575710990>, <mavsdk.mission_raw.MissionItem object at 0x7fb575710a10>, <mavsdk.mission_raw.MissionItem object at 0x7fb575710850>, <mavsdk.mission_raw.MissionItem object at 0x7fb575710890>],)
```
The error message does not tell me much. Neither what the error is ('unsupported' is very generic - for example I had a wrong coordinate frame in this case but nothing would have indicated it) - nor which mission item triggered it.

PX4 console isn't very helpful either:

`WARN  [mavlink] IGN MISSION_ITEM: Busy`

julianoes, 2021-05-13:
I think you have to debug it on the PX4 side where the mission is parsed and check where/why it responds with unsupported. No doubt the mission_raw interface is not so easy to use, as you have the full API and it's hard to know what is correct and/or supported. Guess what, that's exactly why I wrote the simplified mission class in the first place :smirk:.

JonasVautherin, 2021-05-13:
BTW I think that the `IGN MISSION_ITEM: Busy` is a wrong message, I mentioned it here: https://github.com/PX4/PX4-Autopilot/pull/17562

petergerten, 2021-05-15:
@julianoes thanks. I am aware of the simplified API. I am using mission_raw because I am importing .plan files (not created by QGC but a third party GCS).

petergerten, 2021-05-15:
@JonasVautherin thanks for the heads up!

fire-code-d, 2021-06-25:
I think this is because the pymavlink version and the mavsdk version do not match.

I got this error when I use pymavlink==2.4.15 and mavsdk==0.18.0

and then, I changed the pymavlink to version 2.4.10, the problem solved

tourist888, 2023-05-09:
Now I am trying to use pymavlink==2.4.15 and mavsdk==0.18.0. But can not install. What is you python and pip version at that time.

JonasVautherin, 2023-05-09:
@tourist888: how is that related to the original question of this issue? :thinking:

julianoes, 2023-05-09:
This issue is old and there was confusing follow up. Closing.

closed 2023-05-09 by julianoes