mavlink/MAVSDK-Python #349: Heading to the north degree

a-altug, 2021-05-19:
Is there any telemetry object to get heading to the north?

```
    async for raw in drone.telemetry.raw_gps():
         print(f"Heading: {raw.yaw_deg}")
```

did not work. Always printed 0.0 for `offboard_velocity_body.py` example

@JonasVautherin

julianoes, 2021-05-20:
Would attitude yaw work?
http://mavsdk-python-docs.s3-website.eu-central-1.amazonaws.com/plugins/telemetry.html#mavsdk.telemetry.EulerAngle

a-altug, 2021-05-20:
It worked as "Yaw in earth frame from north.".
Thank you.

[http://mavsdk-python-docs.s3-website.eu-central-1.amazonaws.com/plugins/telemetry.html#mavsdk.telemetry.EulerAngle
](url)
It may be useful to change the `yaw_deg` description.

closed 2021-05-20 by a-altug

julianoes, 2021-05-21:
@a-altug feel free to make a PR to improve the docs, it would go here:
https://github.com/mavlink/MAVSDK-Proto/blob/5fd5962e83ecb0b616788351df63335287e66d3e/protos/telemetry/telemetry.proto#L25-L27