mavlink/MAVSDK-Python #350: Dynamically changing inputs

a-altug, 2021-05-20:
Is there a way to change `offboard` classes' inputs dynamically after called for once for dynamic-offboard drive like landing on a marker or flying on a path by making obstacle avoidance.

Or, is calling them continuously by using a while loop (that different `ofboard.class()`es are called for different situations. Means, driving with velocity setpoint or driving with position setpoint) only solution?

@julianoes 
@JonasVautherin

JonasVautherin, 2021-05-20:
I'm afraid I don't really get what you mean. Could you show a pseudo-code of what you would want to do?

a-altug, 2021-05-20:
`while` part of #351

JonasVautherin, 2021-05-20:
Ok I still don't get this issue. Did you play with the offboard examples? https://github.com/mavlink/MAVSDK-Python/tree/main/examples

julianoes, 2021-05-21:
@a-altug you can update the setpoint you need dynamically at the rate you want. Is that what you meant?