mavlink/MAVSDK-Python #351: Lagging when reading .telemetry.position()

a-altug, 2021-05-20:
```
class AirUnit:

      def __init__(self):
          self.drone = None
  
          ...

      # activating from main script  
      async def connect_uav(self):
          """Main function to connect to the drone"""
          # Connect to the Simulation
          self.drone = System()
          await self.drone.connect(system_address="udp://:14540")
          await asyncio.sleep(1)
  
          async for state in self.drone.core.connection_state():
  
              if state.is_connected:
                  print(f"-- Connected to drone with UUID: {state.uuid}")
                  self.is_conn = True
  
                  asyncio.ensure_future(self.observe_position())
                  asyncio.ensure_future(self.observe_is_in_air())
                  asyncio.ensure_future(self.observe_gimbal_euler())
                  asyncio.ensure_future(self.observe_attitude())
  
                  return
  
              else:
                  self.is_conn = False
                  print("Waiting for connection.")
  
      # activating from main script
      async def arm(self):
          async for gps_info in self.drone.telemetry.gps_info():
              if gps_info.fix_type == telemetry.FixType.FIX_3D:
                  print(f"-- Got 3D fix")
                  break
              else:
                  print("Waiting for GPS 3D fix")
  
          async for is_armed in self.drone.telemetry.armed():
              if is_armed:
                  print("Air Unit is armed!")
                  break
              else:
                  print("-- Arming")
                  await self.drone.action.arm()
          await asyncio.sleep(0)
  
          print("-- Setting initial setpoint")
          await self.drone.offboard.set_attitude(offboard.Attitude(0.0, 0.0, 0.0, 0.0))
          return
  
      # activating from main script
      async def mode_offboard(self):
          try:
              print("-- Starting offboard mode")
              await self.drone.offboard.start()
              return
  
          except offboard.OffboardError as error:
              print(f"Starting offboard mode failed with error code: \
              {error._result.result}")
  
              print("-- Disarming")
              await self.drone.action.disarm()
              return
  
      async def observe_gimbal_euler(self):
          async for gimbal_euler in self.drone.telemetry.camera_attitude_euler():
              self.gimbal_yaw = gimbal_euler.yaw_deg
              self.gimbal_pitch = gimbal_euler.pitch_deg
              self.gimbal_roll = gimbal_euler.roll_deg
              await asyncio.sleep(0.005)
  
      async def observe_is_in_air(self):
          async for is_in_air in self.drone.telemetry.in_air():
              self.is_in_air = is_in_air
  
              if not is_in_air:
                  if self.phase == "LANDING_ON_PATH":
                      print("-- Disarming!")
                      await self.drone.action.disarm()
                      print("-- Done!")
                      print("Air Unit landed!")
              print("in_air--------", is_in_air)
              await asyncio.sleep(0.005)
  
      async def observe_attitude(self):
          async for attitude_euler in self.drone.telemetry.attitude_euler():
              tt = time.perf_counter()
              self.attitude_yaw = attitude_euler.yaw_deg
              self.attitude_pitch = attitude_euler.pitch_deg
              self.attitude_roll = attitude_euler.roll_deg
              await asyncio.sleep(0.005)
  
      async def observe_position(self):
          async for position in self.drone.telemetry.position():
              tt = time.perf_counter()
              self.pos_alt = position.relative_altitude_m
              self.pos_lat = position.latitude_deg
              self.pos_lon = position.longitude_deg
              self.pos_abs_alt = position.absolute_altitude_m
              await asyncio.sleep(0.005)
              print("observe_position: ", tt - time.perf_counter())
  
              #return
  
      async def run(self):
          while True:
  
              tttt = time.perf_counter()
              #await self.observe_position()
              #print("observe: ", time.perf_counter() - tttt)
              
              ...
  
              if self.phase == "TAKEOFF":
                  if self.pos_alt < 1:
                      self.north_spd, self.east_spd, self.down_spd, self.yaw_spd = 0.0, 0.0, -5.0, 0.0
  
                  elif 1 <= self.pos_alt < 5:
                      self.north_spd, self.east_spd, self.down_spd, self.yaw_spd = 0.0, 0.0, -2.0, 0.0
  
                  elif self.pos_alt > 5:
                      self.north_spd, self.east_spd, self.down_spd, self.yaw_spd = 0.0, 0.0, 0.0, 0.0
                      self.phase = "LANDING_ON_PATH"
  
                  await self.drone.offboard.set_velocity_body(offboard.VelocityBodyYawspeed(self.north_spd, self.east_spd, self.down_spd, self.yaw_spd))
                  await asyncio.sleep(0)
  
                  #await self.drone.gimbal.set_mode(GimbalMode.YAW_LOCK)
                  #await asyncio.sleep(0)    
  
              ...
  
              elif self.phase == "LANDING_ON_PATH":
                  await self.drone.offboard.set_velocity_body(offboard.VelocityBodyYawspeed(self.north_spd, self.east_spd, self.down_spd, self.yaw_spd))
                  await asyncio.sleep(0)
        
              ...
  
              print("drone_time: ", time.perf_counter() - tttt)
```

The readings from ` async def observe_position(self):` come with great delay. Because of this, the statements inside the While loop are not active correctly. The `.phase`, which should be active at 5 meters in gazebo simulation, is active around 15 meters.
(while loop -> 20Hz)

If I call the method in the while loop instead of using `asyncio.ensure_future (self.observe_position ())` and break the async loop with `return`, I can read values real time but the while loop time gets longer. (nearly 1Hz).
```

    ...
                  #asyncio.ensure_future(self.observe_position())
    ...

    async def observe_position (self):
             async for position in self.drone.telemetry.position():
                 self.pos_alt = position.relative_altitude_m
                 self.pos_lat = position.latitude_deg
                 self.pos_lon = position.longitude_deg
                 self.pos_abs_alt = position.absolute_altitude_m
                 return
    
         async def run (self):
             while True:
             ...
                 await self.observe_position ()
             ...
```

What can be done about `.telemetry.position()` lagging?

JonasVautherin, 2021-05-20:
Why do you `await asyncio.sleep(0.005)` in all the `async for`? :thinking:

JonasVautherin, 2021-05-20:
> 200Hz is highly enough for me and by using ram to much can cause the system unbalanced.
And the problem is not about sleeps. Checked.

Got it for the problem, but still, that's not how it works. The messages that are sent by the drone arrive anyway, and you consume them one after the other. If you sleep, you just add an artificial delay in your code, but the "queue of messages" still growth the same way. In other words, you don't want to sleep here, you just want to consume the messages as fast as possible (i.e. don't do time-consuming operations there).

Now if you want to reduce the rate of some messages, there are `set_rate` functions, e.g. in the telemetry plugin. And when you `set_rate()`, then you effectively ask the drone to send fewer messages.

--------------------

Back to your problem, where do you call `connect_uav`, which starts `observe_position`?

a-altug, 2021-05-21:
The Drone's `run` method(above) loops 20Hz, main script loops 100Hz, gimbal serial loops 100Hz, image process 100Hz.
200Hz is really enough to consume the messages :)
But for now.
If the whole system could be stable and the classes parameters and the coroutines let it be, everyone wants consume as much as he/she can :)

But thank you for suggestion.

______________________________________________________________________________________________________

An object is created in main script .
When the time comes up `connect_uav` is called in main script by the help of created object.

```
    from drone import Drone
    ...
    drone = Drone()
    await drone.connect_uav()
    ....
    await drone.arm()
    ....
    await drone.mode_offboard()
```

All depend conditions.

julianoes, 2021-05-21:
@a-altug 

> Now if you want to reduce the rate of some messages, there are set_rate functions, e.g. in the telemetry plugin. And when you set_rate(), then you effectively ask the drone to send fewer messages.

This is key.

a-altug, 2021-05-24:
> @a-altug
> 
> > Now if you want to reduce the rate of some messages, there are set_rate functions, e.g. in the telemetry plugin. And when you set_rate(), then you effectively ask the drone to send fewer messages.
> 
> This is key.

I have just had time to try it. It is definitely the key.
Thank you both.

JonasVautherin, 2021-05-24:
Great! Closing then, feel free to re-open (or open a new issue) if necessary!

closed 2021-05-24 by JonasVautherin