mavlink/MAVSDK-Python #355: Returing telemetry position data from async loop

MJ-1007, 2021-05-31:
Hello,

Is there a way to get the position telemetry data after using **async for position in drone.telemetry.position():**? The purpose is to use the real time position data from this async loop and return it so that I can use the data for some other purpose.

Is there any way to do this in MAVSDK-Python? Any suggestion will be very helpful.

Thanks in advance.

JonasVautherin, 2021-05-31:
Not sure I understand. This is what `drone.telemetry.position()` does: it gives you the position telemetry data:

```python
async for position in drone.telemetry.position():
    <do something with position>
```

MJ-1007, 2021-06-03:
Hello,

Thank you for your response.

I would like to know if there is a possibility to return this position information to another user-defined function where I will be using this continuous telemetry data for performing some tasks.

Like for example:

"""
Telemetry information from drone
"""
`async def telemetry(drone):` # This is a function specific for telemetry
`  async for position in drone.telemetry.position():`
        `lat = position.latitude_deg`
        `lon = position.longitude_deg`

Now I would like to use this **lat** and **lon** information in some other function, say functionA to perform some tasks.

Is it possible to continuously return these values to another function?

Thanks in advance,

JonasVautherin, 2021-06-03:
Why not something like:

```python
async def handle_position(drone):
    async for position in drone.telemetry.position():
        await some_function(position)

async def some_function(position):
    <do something with position>
```

MJ-1007, 2021-06-03:
Hello,

I would like to hold or make the drone hold, when it reaches a particular position and then continue its path to another location. Something like this:

async def handle_position(drone):
    async for position in drone.telemetry.position():
        await some_function(position)

async def some_function(position):
    <do something with position>
    await asyncio.sleep(<wait for sometime>)
    <proceed for the next location>
     await asyncio.sleep(<wait again for sometime>)

I tried using the method you proposed however if I add sleep within the function, then the reception of telemetry is also delayed by the wait time. So in order to proceed with the next location, it waits until all the telemetry is received, which is ofcourse something that asyncio.sleep will do.

I would like to get this telemetry data continuously as I am comparing the actual drone location from telemetry with the intended location where the drone has to reach.

Thanks again for the prompt response.

JonasVautherin, 2021-06-03:
Just check the position and do what's needed when it is reached, e.g.

```python
async def handle_position(drone):
    async for position in drone.telemetry.position():
        if (should_hold(position)):
            await drone.mission.pause_mission()
```

No need to sleep :thinking:

MJ-1007, 2021-06-03:
Thank you for your response.

Is it possible to use `await drone.mission.pause_mission()` even if I am using `goto_location` command instead of using mission? I am using `goto_location` to have a control over orientation of the drone while moving from one location to another. Is there a way that I can make changes in orientation by using a mission function?

Thanks again.

JonasVautherin, 2021-06-03:
Oh, so you `goto_location`, and sometimes you want to interrupt the goto and hold? I mean, there is a [hold](https://github.com/mavlink/MAVSDK-Proto/blob/main/protos/action/action.proto#L91-L100) function in the action plugin, too...

MJ-1007, 2021-06-07:
Oh okay. Thank you so much for your suggestions. I will have a look into it.

Thanks.

Hamidraei23, 2021-08-16:
Hello I think I am having a similar problem, I need attitude and distance sensor data to use it for another function to achieve 
the horizontal distance between drone and the target and so that I can use it for controlling the drone in a no GPS environment.
the problem is when I want to get attitude telemetry data, if I write and start it in another loop I don't know how to enter the value to main loop, so I have used await distance sensor and await attitude and it makes the program very slow that the camera is able to be run in 13 ,14 FPS. So it would be great if U could let me know how to get a return value from async distance sensor function into the main loop