mavlink/MAVSDK-Python #356: RC status telemetry always shows available

7dnaiel492, 2021-06-01:
After initially connecting the RC to the drone, mavsdk.telemetry.RcStatus.is_available is always True, even with the RC turned off. QGC recognizes and announces the RC as disconnected, but the SDK does not update the RC telemetry.

The RC loss failsafe (NAV_RCL_ACT) behaves as expected in position, altitude, etc modes but does not override offboard control. I need some way to monitor RC disconnect via MAVSDK to stop offboard command. I set the Futaba offboard switch failsafe value to switch out of offboard and confirmed the switch is being toggled on RC disconnect, but the drone still remains in offboard mode. I've read all the information I can find and cannot find a way to monitor the raw RC switch state or come up with a way to monitor RC status. Any ideas how I can accomplish this?

I am using a Freefly Alta X and Futaba T14SG. PX4 version 1.3.0.0 and AltaQGC v1.3.8. I believe I am confined to these versions from Freefly.

JonasVautherin, 2021-06-01:
What do the other two values say? `was_available_once` and `signal_strength_percent`? Do they get updates?

7dnaiel492, 2021-06-01:
`RcStatus: [was_available_once: True, is_available: True, signal_strength_percent: 42.0]` no matter the state of the RC. 

I am getting RcStatus telemetry updates at 0.6-0.8 Hz. `signal_strength_percent` changed from 41.0 to 42.0 just now, but that's the only status change I've seen. When the drone boots without the RC off, RcStatus is None. After the RC connects to the vehicle, the status is as above even when the SDK reconnected.

JonasVautherin, 2021-06-03:
It is done here: https://github.com/mavlink/MAVSDK/blob/main/src/plugins/telemetry/telemetry_impl.cpp#L1141-L1157

```c++
void TelemetryImpl::process_rc_channels(const mavlink_message_t& message)
{
    mavlink_rc_channels_t rc_channels;
    mavlink_msg_rc_channels_decode(&message, &rc_channels);

    bool rc_ok = (rc_channels.chancount > 0);
    set_rc_status(rc_ok, rc_channels.rssi);

    std::lock_guard<std::mutex> lock(_subscription_mutex);
    if (_rc_status_subscription) {
        auto callback = _rc_status_subscription;
        auto arg = rc_status();
        _parent->call_user_callback([callback, arg]() { callback(arg); });
    }

    _parent->refresh_timeout_handler(_rc_channels_timeout_cookie);
}
```

where the `is_available` comes from `bool rc_ok = (rc_channels.chancount > 0);`. I guess that the chancount drops to 0 when the RC is disconnected (at least that's what I seem to observe on QGC)...

7dnaiel492, 2021-06-04:
While monitoring in QGC Widgets>MAVLink Inspector, `chancount` remains at 18 when the transmitter is turned off. `chancount` drops to zero when the receiver is unplugged from the flight controller, but that doesn't really help me. Am I missing a Futaba configuration that emits a failsafe rather than holding existing RC values?

Alternatively, I am able to grab the latest MAVLink StatusText which is set to `"Manual control lost"` when the RC is disconnected. Is there a value I am able to access through the sdk that sets that status text when the RC is lost but doesn't update `RcStatus.is_available`?

JonasVautherin, 2021-06-04:
I don't know :confused:. But if you do find a MAVLink message that gives this information (other than the StatusText, because that would be a hack), then we can use it in MAVSDK instead of `chancount` :+1:

julianoes, 2021-06-07:
Sounds like we need a timeout to update that flag.Thanks for the note @7dnaiel492.

julianoes, 2021-06-09:
My comment above is wrong as @JonasVautherin has already found :facepalm:.

However, I think we ought to actually check the SYS_STATUS message to determine the status with this bit:
https://mavlink.io/en/messages/common.html#MAV_SYS_STATUS_SENSOR_RC_RECEIVER

julianoes, 2021-06-09:
@7dnaiel492 how have you set-up RC lost? I assume according to this? https://docs.px4.io/master/en/config/radio.html#rc-loss-detection

And if so, did you use the low throttle detection? Because that would explain why 18 channels are still sent but throttle is very low which means it's "lost".

7dnaiel492, 2021-06-09:
Yes, setup with low throttle detection.

julianoes, 2021-06-10:
Got it. Yes, I guess in that case we should really look at the SYS_STATUS instead. Because it makes somewhat sense to keep sending the RC channels because that's feedback for a user how to set it up. E.g. for you it allowed you too see the throttle low value and set the low throttle detection accordingly.

closed 2021-06-14 by julianoes