mavlink/MAVSDK-Python #357: Using MAVSDK-Python to control one drone in HITL and the other drone in SITL

amaalo2, 2021-06-04:
Hello,

I am currently working on a collision avoidance project. I am using MAVSDK-Python to send commands to control the main quadrotor (ownship). When I first started the project, I had two quads both running in SITL and I would instantiate both of them using `./gazebo_sitl_multiple_run.sh -n 2 -t px4_sitl_default`. I was able to control the ownship and achieve collision avoidance in the scenario I was focusing on. 

I wanted to know if it is possible to have the ownship in HITL, but keep the intruder in SITL? I followed the instructions from https://docs.px4.io/master/en/simulation/hitl.html, and for only one drone, the ownship, I was able to successfully control it with MAVSDK-Python. @JonasVautherin  @julianoes Can you label this issue as question?

JonasVautherin, 2021-06-04:
How did you start `mavsdk_server` for the situation with two SITL that was working?

julianoes, 2021-06-07:
@amaalo2 you need to set up the ports correctly for when you have multiple drones connected.

SITL will - by default - use UDP port 14550 and UDP port 14540.
HITL will - by default - connect via serial USB (e.g. `/dev/ttyACM0`) and then the HITL simulator (jMAVSim or Gazebo) forwards it to 14550.

You probably want to change one of them to use a different port to disambiguate, and you also need to make sure the two systems don't have the same MAVLink sysid which you can set with the [MAV_SYS_ID](https://docs.px4.io/master/en/advanced_config/parameter_reference.html#MAV_SYS_ID) param.

amaalo2, 2021-06-07:
@JonasVautherin  I would start both mavsdk_server with the following commands : 
```
gnome-terminal -- ./mavsdk_server udp://:14540 -p 50040 # mavsdk_server1
gnome-terminal -- ./mavsdk_server udp://:14541 -p 50041 # mavsdk_server2
```

Inside the python script, I would have two lines like so : 

```
ownship = System(mavsdk_server_address="localhost", port=50040)
intruder = System(mavsdk_server_address="localhost", port=50041)
```

@julianoes  - I am not too sure where I can change the MAV_SYS_ID param. In any case, it is not too clear to me how/when and in which order I should start the simulation of HITL+ SITL? Is it HITL first, then SITL, otherway around? Do I need to create a new script? 

Really appreciate the help!

JonasVautherin, 2021-06-07:
The way you start mavsdk, you don't need to change `MAV_SYS_ID` as far as MAVSDK is concerned, because they are on different connections :+1:.

Now given what you said above:

> SITL will - by default - use UDP port 14550 and UDP port 14540.
> HITL will - by default - connect via serial USB (e.g. /dev/ttyACM0) and then the HITL simulator (jMAVSim or Gazebo) forwards it to 14550.

This means that `ownship` will connect to SITL (udp://:14540), and `intruder` will not connect to anything because it listens on udp://:14541 and nobody sends mavlink there.

Now the problem is that both SITL and HITL broadcast on 14550 (still given what you said above), which will be an issue. Can you change HITL such that it broadcasts on 14541 instead of 14551? Then your setup above should work.

amaalo2, 2021-06-07:
So I think I understand what you are saying. Gazebo should forward the 14541 instead of 14540. I looked into ``  Tools/sitl_gazebo/models/iris/iris.sdf.jinja`` and changed the ``<sdk_udp_port>14540</sdk_udp_port>`` to ``<sdk_udp_port>14541</sdk_udp_port>`` and ``<qgc_udp_port>14550</qgc_udp_port>`` to ``<qgc_udp_port>14551</qgc_udp_port>`` that still didn't work. 

Also to start the simulation (Gazebo, MAVSDK_Server, QGcontrol), it's weird, it's as if I am starting two simulations. 

``` 
#Start QGC
cd /home/anthony
gnome-terminal -- ./QGroundControl.AppImage 

# Connect Pixhawk 4 before starting the script
cd /home/anthony/PX4-Autopilot
eval 'DONT_RUN=0 make px4_sitl_default gazebo'
eval 'source Tools/setup_gazebo.bash $(pwd) $(pwd)/build/px4_sitl_default'
eval 'gazebo Tools/sitl_gazebo/worlds/hitl_iris.world'

```

Then I open another terminal and run this script

```
#Start MAVSDK Servers
cd /home/anthony/Documents/GitRepos/MAVSDK-Python/mavsdk/bin

gnome-terminal -- ./mavsdk_server udp://:14540 -p 50040 # mavsdk_server1
gnome-terminal -- ./mavsdk_server udp://:14541 -p 50041 # mavsdk_server2

#Start Gazebo
cd /home/anthony/PX4-Autopilot/Tools
gnome-terminal -- ./gazebo_sitl_multiple_run2.sh -n 1 -t px4_sitl_default

```

So at this point, I have two instances of Gazebo running and I don't think this is working as intended

julianoes, 2021-06-10:
So now you are starting two drones via SITL again and none via HITL, right? I'm confused.

amaalo2, 2021-06-10:
To my knowledge, and maybe I am wrong, the first code block starts HITL according to https://docs.px4.io/master/en/simulation/hitl.html 

> ```
#Start QGC
cd /home/anthony
gnome-terminal -- ./QGroundControl.AppImage 

# Connect Pixhawk 4 before starting the script
cd /home/anthony/PX4-Autopilot
eval 'DONT_RUN=0 make px4_sitl_default gazebo'
eval 'source Tools/setup_gazebo.bash $(pwd) $(pwd)/build/px4_sitl_default'
eval 'gazebo Tools/sitl_gazebo/worlds/hitl_iris.world'
```

Then to start the SITL simulation, I  am not sure if this is correct as well but I used 

```
#Start Gazebo
cd /home/anthony/PX4-Autopilot/Tools
gnome-terminal -- ./gazebo_sitl_multiple_run2.sh -n 1 -t px4_sitl_default
```

With the idea that ``-n 1 `` is the argument to start only 1 drone in SITL. 

What would you do differently @julianoes ?

julianoes, 2021-06-11:
Ok, got it, makes sense.

> So I think I understand what you are saying. Gazebo should forward the 14541 instead of 14540. I looked into Tools/sitl_gazebo/models/iris/iris.sdf.jinja and changed the [...]

That sounds correct. Did you make sure the .sdf file is re-generated correctly after your change/