mavlink/MAVSDK-Python #358: Waiting for drone to connect...

offjangir, 2021-06-08:
I was trying to run some simple examples with gazebo.
I launch iris in gazebo  with px4 autopilot it works perfectly but when I run any example / my code following is seen:
Waiting for mavsdk_server to be ready...
Connected to mavsdk_server!
Waiting for drone to connect...

I wanted to use mavsdk for multi drone simulation but for some reason My connection is not being set up.
Help please!

JonasVautherin, 2021-06-08:
By default, the examples listens on `udp://:14540`. I think that SITL broadcasts on 14540 by default, so if both run on the same machine, it should work.

Does QGC connect to your SITL instance? QGC uses `udp://:14550` by default.

offjangir, 2021-06-09:
I am not using QGC .
when I run SITL instance  directly
-- make px4_sitl_default gazebo_iris
everything works fine and I am able to control things..

what I want to do is to launch multiple bots .. when I do that with 
roslaunch px4 multi_uav_mavros_sitl.launch

mavsdk server connects but Drones dont get connected.
For ex.
I launched multiple drones but tried to connect to udp 14540  with drone id 0 and mavsdk isnt able to find drone.

I am new at this so  maybe doing some trivial mistake.

JonasVautherin, 2021-06-10:
When you run `make px4_sitl_default gazebo_iris`, it will broadcast mavlink messages to udp 14540 (that's the default for MAVSDK) and udp 14550 (that's the default for QGC).

Now when you run multiple drones with that ROS command, I don't know what it does. Did you read what `multi_uav_mavros_sitl.launch` does? Where is that file?

offjangir, 2021-06-10:
The file is located  /PX4-Autopilot/launch 
here are the contents of the file

<?xml version="1.0"?>
<launch>
    
    
    
    <arg name="est" default="ekf2"/>
    <arg name="vehicle" default="iris"/>
    <arg name="world" default="$(find mavlink_sitl_gazebo)/worlds/empty.world"/>
    
    <arg name="gui" default="true"/>
    <arg name="debug" default="false"/>
    <arg name="verbose" default="false"/>
    <arg name="paused" default="false"/>
    
    <include file="$(find gazebo_ros)/launch/empty_world.launch">
        <arg name="gui" value="$(arg gui)"/>
        <arg name="world_name" value="$(arg world)"/>
        <arg name="debug" value="$(arg debug)"/>
        <arg name="verbose" value="$(arg verbose)"/>
        <arg name="paused" value="$(arg paused)"/>
    </include>
    
    <group ns="uav0">
        
        <arg name="ID" value="0"/>
        <arg name="fcu_url" default="udp://:14540@localhost:14580"/>
        
        <include file="$(find px4)/launch/single_vehicle_spawn.launch">
            <arg name="x" value="0"/>
            <arg name="y" value="0"/>
            <arg name="z" value="0"/>
            <arg name="R" value="0"/>
            <arg name="P" value="0"/>
            <arg name="Y" value="0"/>
            <arg name="vehicle" value="$(arg vehicle)"/>
            <arg name="mavlink_udp_port" value="14560"/>
            <arg name="mavlink_tcp_port" value="4560"/>
            <arg name="ID" value="$(arg ID)"/>
            <arg name="gst_udp_port" value="$(eval 5600 + arg('ID'))"/>
            <arg name="video_uri" value="$(eval 5600 + arg('ID'))"/>
            <arg name="mavlink_cam_udp_port" value="$(eval 14530 + arg('ID'))"/>
        </include>
        
        <include file="$(find mavros)/launch/px4.launch">
            <arg name="fcu_url" value="$(arg fcu_url)"/>
            <arg name="gcs_url" value=""/>
            <arg name="tgt_system" value="$(eval 1 + arg('ID'))"/>
            <arg name="tgt_component" value="1"/>
        </include>
    </group>
    
    <group ns="uav1">
        
        <arg name="ID" value="1"/>
        <arg name="fcu_url" default="udp://:14541@localhost:14581"/>
        
        <include file="$(find px4)/launch/single_vehicle_spawn.launch">
            <arg name="x" value="1"/>
            <arg name="y" value="0"/>
            <arg name="z" value="0"/>
            <arg name="R" value="0"/>
            <arg name="P" value="0"/>
            <arg name="Y" value="0"/>
            <arg name="vehicle" value="$(arg vehicle)"/>
            <arg name="mavlink_udp_port" value="14561"/>
            <arg name="mavlink_tcp_port" value="4561"/>
            <arg name="ID" value="$(arg ID)"/>
            <arg name="gst_udp_port" value="$(eval 5600 + arg('ID'))"/>
            <arg name="video_uri" value="$(eval 5600 + arg('ID'))"/>
            <arg name="mavlink_cam_udp_port" value="$(eval 14530 + arg('ID'))"/>
        </include>
        
        <include file="$(find mavros)/launch/px4.launch">
            <arg name="fcu_url" value="$(arg fcu_url)"/>
            <arg name="gcs_url" value=""/>
            <arg name="tgt_system" value="$(eval 1 + arg('ID'))"/>
            <arg name="tgt_component" value="1"/>
        </include>
    </group>
    
    <group ns="uav2">
        
        <arg name="ID" value="2"/>
        <arg name="fcu_url" default="udp://:14542@localhost:14582"/>
        
        <include file="$(find px4)/launch/single_vehicle_spawn.launch">
            <arg name="x" value="0"/>
            <arg name="y" value="1"/>
            <arg name="z" value="0"/>
            <arg name="R" value="0"/>
            <arg name="P" value="0"/>
            <arg name="Y" value="0"/>
            <arg name="vehicle" value="$(arg vehicle)"/>
            <arg name="mavlink_udp_port" value="14562"/>
            <arg name="mavlink_tcp_port" value="4562"/>
            <arg name="ID" value="$(arg ID)"/>
            <arg name="gst_udp_port" value="$(eval 5600 + arg('ID'))"/>
            <arg name="video_uri" value="$(eval 5600 + arg('ID'))"/>
            <arg name="mavlink_cam_udp_port" value="$(eval 14530 + arg('ID'))"/>
        </include>
        
        <include file="$(find mavros)/launch/px4.launch">
            <arg name="fcu_url" value="$(arg fcu_url)"/>
            <arg name="gcs_url" value=""/>
            <arg name="tgt_system" value="$(eval 1 + arg('ID'))"/>
            <arg name="tgt_component" value="1"/>
        </include>
    </group>
</launch>

offjangir, 2021-06-10:
https://github.com/PX4/PX4-Autopilot/blob/8c4b900f9ac6aa48c60def3d1cda04ee22e59777/launch/multi_uav_mavros_sitl.launch#L1

offjangir, 2021-06-10:
Yeah I took a look at it but didn't think of a reason it should not work.
udp is clearly defined for each drone.

Launch file for your Reference :
https://github.com/PX4/PX4-Autopilot/blob/8c4b900f9ac6aa48c60def3d1cda04ee22e59777/launch/multi_uav_mavros_sitl.launch#L1

JonasVautherin, 2021-06-10:
But do you run mavsdk to listen on 14560, 14561 and 14562? Isn't that where your launch file says that mavlink will be broadcasted?

offjangir, 2021-06-10:
drone = System()
    await drone.connect(system_address="udp://:14560")

yes I am ,
for starters I just want to connect to one drone in the whole swarm. but when I run 

Waiting for mavsdk_server to be ready...
Connected to mavsdk_server!
Waiting for drone to connect...

I am stuck here as drone is not being discovered.

offjangir, 2021-06-10:
The example that I am running is like this :

async def run():

    drone = System()
    await drone.connect(system_address="mavlink_udp_port://:14560")

    print("Waiting for drone to connect...")
    async for state in drone.core.connection_state():
        if state.is_connected:
            print(f"Drone discovered!")
            break

    print("Waiting for drone to have a global position estimate...")
    async for health in drone.telemetry.health():
        if health.is_global_position_ok:
            print("Global position estimate ok")
            break

    print("-- Arming")
    await drone.action.arm()

    print("-- Taking off")
    await drone.action.takeoff()

    await asyncio.sleep(5)

    print("-- Landing")
    await drone.action.land()

if __name__ == "__main__":
    loop = asyncio.get_event_loop()
    loop.run_until_complete(run())

JonasVautherin, 2021-06-10:
You need to make sure that your drone (SITL or not) sends mavlink packets that reach MAVSDK. If it is your first time using MAVSDK, a quick way to try with a headless gazebo is [documented here](https://github.com/jonasvautherin/px4-gazebo-headless):

```
docker run --rm -it jonasvautherin/px4-gazebo-headless:1.11.0
```

And if the docker container does not run on the same host as mavsdk, you can specify the IP where mavsdk runs, for instance:

```
docker run --rm -it jonasvautherin/px4-gazebo-headless:1.11.0 192.168.1.12
```

If this works, then it means that you need to look into what your SITL does, and make sure it can reach mavsdk.

offjangir, 2021-06-12:
@JonasVautherin  Thanks for your help. Actually I found out the problems through one of your answer in [#931](https://github.com/mavlink/MAVSDK/issues/931)
The PX4 launch files launch Mavros in the file itself and hence Mavsdk  is not able to connect to the ports.
For  Connecting multiple vehicles I did was this : 
1) Run  multiple vehicles with no ROS  in Gazebo ([follow this ](https://docs.px4.io/master/en/simulation/multi_vehicle_simulation_gazebo.html))
2) Launch mavsdk_severs : 
	cd  /Path_to_mavsdk/bin
	./mavsdk_server -p 50040 udp://:14540
	./mavsdk_server -p 50041 udp://:14541
	./mavsdk_server -p 50042 udp://:14542

Everything works fine after that.

closed 2021-06-12 by offjangir